framps/raspiBackup #533: raspiBackup klappt nicht

DarthWeber, 2022-08-25:
Hi framp, beim Üben der restore kam ich eigtl. immer gut klar, aber nun ist es gerade Ernst - und ich kriege den restore einfach nicht hin :-(

`root@r2d2:~# raspiBackup --unsupportedEnvironment -g -d /dev/sdd /backup/.....
!!! RBK0269W: @@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
!!! RBK0268W: @@@ Nicht unterstützte Umgebung @@@
!!! RBK0269W: @@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
--- RBK0009I: r2d2: raspiBackup.sh V0.6.7-hotfix-508 ($/) Do 25. Aug 13:58:18 CEST 2022 gestartet.
!!! RBK0173W: =========> HINWEIS <=========
!!! RBK0173W: Dieses ist ein temporärer Hotfix der auf die nächste Version upgraded werden muss sobald eine verfügbar ist.
!!! RBK0173W: =========> HINWEIS <=========
!!! RBK0018W: Ziel /dev/sdd mit 29.81 GiB ist größer als die Backupquelle mit 14.83 GiB. Die root Partition wird entsprechend vergrößert um den ganzen Pl
!!! RBK0065W: Gerät /dev/sdd wird repartitioniert und die gesamten Daten werden gelöscht.
--- RBK0146I: Keine Partitionstabelle auf /dev/sdd gefunden.
--- RBK0066I: Gerät /dev/sdd wird überschrieben mit der gesicherten Boot- und Rootpartition.
--- RBK0069I: Bootpartition /dev/sdd1 wird formatiert und erhält die zurückgespielte Bootpartition.
--- RBK0070I: Rootpartition /dev/sdd2 wird formatiert und erhält die zurückgespielte Rootpartition.
--- RBK0038I: Bist Du sicher? j/N
j
--- RBK0050I: Backup wird von /backup/Chekow/Chekow-rsync-backup-20220825-010001 zurückgespielt.
`

das Problem ist: Bei der letzten Meldung bleibt es Stunden stehen, es kommt nix mehr. Die logr kriege ich erst, wenn ich mit ctrl-c abbreche. Müsste ich mit "-g" nicht eigtl. einen Fortschritt sehen ?

Naja, ich alsss das jetzt mal einfach weiter laufen....

framps, 2022-08-25:
> Bei der letzten Meldung bleibt es Stunden stehen, es kommt nix mehr.

Stunden? Da ist irgendwo der Wurm drin :cry: 

> Müsste ich mit "-g" nicht eigtl. einen Fortschritt sehen ?

Ja. Aber da ist leider ein Bug in raspiBackup der schon gefixed ist aber erst in der naechsten Release drin sein wird.

> Die logr kriege ich erst, wenn ich mit ctrl-c abbreche.

raspiBackup schreibt das Debuglog waehrend es aktiv ist in /tmp. D.h. Du kannst dort mal in das Log reinsehen ob Du dort eine Indikation findest warum es so lange dauert.

DarthWeber, 2022-08-25:
ja, in der logr in tmp steht exakt das, was ich auch in der Konsole habe, also das von oben. Nach "RBK0050I: Backup wird von /backup/Chekow/Chekow-rsync-backup-20220825-010001 zurückgespielt." kommt eben nichts mehr.

Hm, dann warte ich mal auf ad neue release, oder ? Aber s ist komsich, dass die Lampe im Adapter nicht flackert. Ich habe eben ein älteres ddz restored - das ging mit win32diskimage auf Anhieb und die LED blinkt beim Schreiben natürlich.

framps, 2022-08-25:
> Aber s ist komsich, dass die Lampe im Adapter nicht flackert.

Schon merkwuerdig. Mit `nmon` kannst Du nachsehen ob was gelesen bzw geschrieben wird.

framps, 2022-08-25:
Auch solltest Du mal in /var/log/syslog nachsehen ob es irgendwelche Lese- oder Schreibfehler gibt.

closed 2022-09-10 by github-actions[bot]