framps/raspiBackup #534: Stop/Start Docker Container

pr3mi3r3, 2022-08-31:
Can you include a Start/Stop Feature for Docker Container?
They also should start after the reboot.
In the docker-compose file I have the option: restart-always.
But this only works, if the container is up and running after backup and before the system reboots.
Thanks.

framps, 2022-08-31:
You can start/stop docker already. Just add `start docker` and `stop docker` in the config file in options `START_SERVICES` and `STOP_SERVICES`.

pr3mi3r3, 2022-08-31:
"service docker stop" does not stopp all container.
Like the pihole container via docker-compose.yml.

framps, 2022-08-31:
I understand your issue. But I want to understand the problem domain first and then will come up with a solution.

Said this, I accept this feature request.

framps, 2022-08-31:
I just setup docker on my Laptop to recreate your scenario.

> Like the pihole container via docker-compose.yml.

Sounds like you have an issue with the pihole container. Do you have instructions how you setup the pihole container with docker-compose I can follow?

framps, 2022-08-31:
I just installed pihole in docker and verified you can stop pihole successfully before you stop docker and start the backup and start both when the backup finished in the following way:

Add following commands at the end of `DEFAULT_STOP_SERVICES`:
` && docker stop pihole && systemctl stop docker`

Add following commands at the end of `DEFAULT_START_SERVICES`:
` && systemctl start docker && docker start pihole`

pr3mi3r3, 2022-08-31:
For the reboot option, I need to add an `reboot` to `DEFAUL_START_SERVICES`?!

The Option
`DEFAULT_REBOOT_SYSTEM=0` ignores the `DEFAUL_START_SERVICES` and piHole will not start after reboot, because it was not restarted before reboot.

framps, 2022-09-02:
> For the reboot option, I need to add an `&& reboot` to the end of the value `DEFAUL_START_SERVICES`: ?!

As you already read this option is ignored when you use `DEFAULT_REBOOT_SYSTEM=1`. 

> and piHole will not start after reboot, because it was not restarted right before system reboot.

Hm ... docker starts all containers if they are configured accordingly. Do I get you correctly that pihole does not start because it was not restarted at the end of the backup?

framps, 2022-09-02:
I just found [this issue](https://github.com/pi-hole/docker-pi-hole/issues/684). There is some issue in your environment. Docker will start pihole during boot. You have to figure out why phihole fails to start during boot.

There is a possible workaround available for you: Either use the wrapper script and start the pihole container at the end again or just change the command in /etc/cron.d/raspiBackup to start raspiBackup followed by`; docker start pihole` . It's not the right way to go from my point of view but it's up to you to decide which way you want to go :wink:

pr3mi3r3, 2022-09-02:
I decided to not set the `DEFAULT_REBOOT_SYSTEM` Option.
And use `DEFAUL_START_SERVICES` to restart piHole after the Backup.
With the additional option `DEFAULT_AFTER_STARTSERVICES="reboot"`, I reboot the Pi.
And after reboot pihole starts up automatically.

framps, 2022-09-03:
> With the additional option `DEFAULT_AFTER_STARTSERVICES="reboot"`, I reboot the Pi.

Good catch :+1:  . That's another workaround. But keep in mind you still terminate raspiBackup when there still should some important code be executed. 

I checked the code and it's the cleanup code which deletes the latest backup when something went wrong and raspiBackup terminates with an error code. You also do not receive an eMail.

I will create an issue and add something like `DEFAULT_FINAL_STARTSERVICES` which allows to execute any command just before raspiBackup finishes. That would be the right place for your `reboot`.

I strongly suggest to use one of the two workarounds I described above. If a backup fails and an inconsistent backup is kept this may cause a restore to fail or the restored image not to start.

pr3mi3r3, 2022-09-03:
Thanks for your help 🙏
I will wait for the new feature

framps, 2022-09-04:
Would be great if you verify issue #535 when I'm done the next couple of days.

framps, 2022-09-04:
Would you please test the hotfix? See #535 how to execute the hotfix.

pr3mi3r3, 2022-09-05:
Thanks for the Hotfix.
I added DEFAULT_FINAL_COMMAND="reboot" .
The command is executed as final command. Seems to work. Thanks.

closed 2022-09-05 by pr3mi3r3

framps, 2022-09-05:
Great. Thank you for your verification. The issue will be available in next release. You can now use this hot fix until next release is available. The hotfix will be upgraded automatically to next release when it's available.