framps/raspiBackup #535: Create someting like DEFAULT_FINAL_COMMANDS

framps, 2022-09-03:
Add an option which allows to invoke any commands just before raspiBackup terminates. See #534 why this makes sense.

framps, 2022-09-04:
There is a hotfix available. To test the hotfix download [this file](https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/raspiBackupFromGitInvocation.sh)

and then execute

`./raspiBackupFromGitInvocation.sh master_535`

pr3mi3r3, 2022-09-05:
Thanks for the Hotfix.
I added `DEFAULT_FINAL_COMMAND="reboot" `.
The command is executed as final command. Seems to work. Thanks.

closed 2022-09-05 by framps

framps, 2022-11-28:
This extension is executed at the end of the backup process. Keep in mind there is additional important code executed later on when you execute a post extension (housekeeping and other stuff). Please read [this page](https://www.linux-tips-and-tricks.de/en/raspibackupcategorye/443-raspibackup-extensions).

> 2\. For a simple reboot, using `DEFAULT_REBOOT_SYSTEM=1` config option would be the way to go I guess

You're right. If you just want to reboot your system at the end of the backup you can call `shutdown -r now` in `DEFAULT_FINAL_COMMANDS` or use instead option `DEFAULT_REBOOT_SYSTEM`. But there are raspiBackup users which want to do much more sophisticated stuff at the end of a raspiBackup :wink:

framps, 2022-11-29:
It's up to the raspiBackup user which extension point makes sense to use. I think it's useful to have multiple extension points a user can choose from.