framps/raspiBackup #537: Scripten von Raspibackup mit Ansible

Springjunky, 2022-09-12:
Hallo,
ich habe ein Ansible-Script geschrieben um einen [Pi4 "Smarthome-Ready"](https://github.com/Springjunky/Pi4SmartHome) zu machen.

Dieses Script möchte ich gerne um raspi-backup erweitern .. ungefähr so.

Den Benutzer minimalisische Konfiguration machen lassen, also 
* wo soll das Backup hin als smb-Mount 
* an welchen Tagen und wieviele Backups halten
* evtl. noch eine Email
*  
typischer Ansible Eintrag wäre damit in etwa
```
raspi-backup:
   samba_sahre: //myshare/backup
   samba_user: username
   samba_password: password
   keep_backups: 3
   services_to_stop:
       - webmin
       - docker
   backup_cron:
         minute: "0"
         hour: "2"
         weekday: "0,2,5"
  notifaction_mail: john_do@google.de

 ```
Diese Werter setze ich dann in ein vorgegebenes Konfigurations-Template von raspi-backup.

#### Fragen
- wie komme ich an die UID ?
- welche Dateien brauche ich für /usr/local/bin .. reicht da raspiBackup.sh und raspiBackupInstallUI.sh 

Denke vorab

framps, 2022-09-12:
> Ich habe ein Ansible-Script geschrieben um einen [Pi4 "Smarthome-Ready"](https://github.com/Springjunky/Pi4SmartHome) zu machen.

Gute Idee. Sieht gut aus :+1: . Und das ganze Environment sollte natuerlich auch gesichert werden. raspiBackup (nicht raspi-backup :cry: ) ist sicherlich ein Tool was Du dazu nutzen kannst :smile: 

Zuerst mal zu Deinen Fragen und dann habe ich mehrere Hinweise zur Nutzung von raspiBackup:

> * wie komme ich an die UID ?

Welche UID meinst Du? Wozu denkst Du dass Du sowas brauchst?

> * welche Dateien brauche ich für /usr/local/bin .. reicht da raspiBackup.sh und raspiBackupInstallUI.sh

raspiBackup.sh ist notwendig um die Backups zu erstellen. Der Installer ist dafuer gedacht ein initiales Config fuer raspiBackup zu erstellen. So wie ich Dich verstehe willst Du das mit Ansible erstellen. Dann ist im Prinzip der Installer nicht notwendig.

Die Frage ist aber wie Du raspiBackup integrieren willst. Soll ein Nutzer von SmarthomeReady manuell raspiBackup per Installer installieren oder willst Du das per Ansible und CLI tun? Oder willst Du raspiBackup bundlen? Letzteres hat den Vorteil dass Du unabhaengig von raspiBackup Upgrades bist. Der Nachteil ist dass raspiBackup Nutzer irgendwann von raspiBackup darauf hingewiesen werden dass es eine neue Version gibt. Allerdings kann man diesen Hinweis per Option auschalten. 

Nun meine Hinweis zu raspiBackup:

> samba_sahre: //myshare/backup

Ich wuerde es nicht auf samba festlegen. Dort gibt es den Nachteil dass kein rsync funktioniert und immer ein Vollbackup erstellt wird. rsync ist die preferred Methode. D.h. aus meiner Sicht sollte der Backuppath und die Methode (rsync, tar oder dd) in Ansible definiert werden. Bei nfs sind keine Credentials notwendig. Fuer samba muessten dann noch die Credentials dazukommen.

> services_to_stop:
>        - webmin
>        - docker

Da fehlt noch IObroker et al - aber ich denke die Liste ist nur ein Beispiel fuer mich. Dir ist klar dass Du aus den Definitionen systemctl statetements generieren musst? Zu IOBroker sei noch [hierauf](https://www.linux-tips-and-tricks.de/de/alle-raspibackup-artikel/611-tipps-zu-verschiedenen-anwendungen/) verwiesen

Zu Thema Support: Wie hast Du Dir das fuer Pi4SmartHome vorgestellt? Du integrierst eine Menge Komponenten. Die Issues werden dann bei Dir im git aufschlagen. Wie soll das z.B. bei raspiBackup dann weitergehen :thinking:

Springjunky, 2022-09-12:
>  (nicht raspi-backup 😢 ) ist sicherlich ein Tool was Du dazu nutzen kannst 😄
Sorry, das habe ich im Ursprungs-Post mal korrigiert.

> Welche UID meinst Du? Wozu denkst Du dass Du sowas brauchst?

Hm. die aus der Konfiguration unter /usr/local/etc/raspiBackup.conf
```
# GENERATED - DO NOT DELETE 
UUID=2d8a95b4-e360-4dc4-a49c-2eeb710c5e1f
```
> So wie ich Dich verstehe willst Du das mit Ansible erstellen

genau so .. also richtig verstanden.

> Die Frage ist aber wie Du raspiBackup integrieren willst.

also eigentlich ist das egal, aber vielleicht sollte ich noch etwas mehr erklären.

Ansible stellt einen Zustand eines Systems her, das funktioniert so, das ich tatsächlich eigentlich gar kein Backup brauche, weil ich mein System jederzeit aus Git mit dem Ansible wieder genau so aufsetzen kann.
Ich setze also tatsächlich auch Konfigurationen per Ansible wie ssh, vnc oder sonstiges.
Na ja .. fast .... 
Sämtliche Docker-Container persistieren ihre Daten in genau einem Verzeichnis, dieses sichere bereits mit einem eigenen Script was im wesentliche alle Docker-Container runterfährt, dann ein tar.gz des Persistenz-Verzeichnisses macht (mit etwas Houskeeping) und alle Docker-Container wieder hochfärt.

In dem Fall ist das Restore recht einfach... Grundinstallation machen, Ansible laufen lassen .. das tar.gz einspielen und fertig .. ist ja alles Docker-basiert und eigentlich bin ich damit fein, aber dann kam ioBroker ...

ioBroker kann auch im Docker-Container laufen, ich finde das aber nicht so wirklich praktikable, wenn man sich nicht gut mit Docker auskennt, denn bei einigen Updates (js-Controller) muss man eine Shell im Docker-Container aufmachen und dort die Update durchführen ..oder bei ZigBee Modulen umständliche Mappings machen, damit alles im Docker-Container erreicht werden kann ... .. bei den anderen Docker-Container muss man eigentlich nie _in_ den Docker-Container rein, sondern mappt alles nur an Konfigurations-Dateien rein und gut ist. Da muss man sich nicht mit Docker auskennen.

Aus dem Grund habe ich ioBroker im neuen Setup mit Ansible nativ installiert und bisher noch keine gute Idee, wie ich die ioBroker Daten auch sichern kann ... ... möchte ich mein System wieder herstellen, muss ich also entweder ..
* ein Basis-System erstellen und Ansible laufen lassen
* die Docker-Daten wieder einspielen aus dem tar.gz
* die ioBroker Daten wieder einspielen.
oder
* ich verzichte auf Ansible bei einem Restore und auch auf die eigene Docker-Sicherung und nutze raspiBackup für die Komplettsicherung.

Ich möchte es aber auch dem Benutzer relativ einfach machen .. so nach dem Motto 
git clone & Ansible laufen lassen und der SmartHome-Raspi ist fertig mit ca. 7 Docker-Container (Influx, Grafana, PiHole, MotionEye,Portainer, DokuWiki...) 
und außerdem habe ich netterweise noch raspiBackup für Dich installiert, was alles auf Dein NAS spielt .. gerne fertig konfiguriert .. bitte schön .... 

... das wäre also der Hintergrund .. für Ideen bin ich offen .. und zu dem Support habe ich mit bisher tatsächlich noch keine Gedanken gemacht .. guter Aspekt

framps, 2022-09-12:
> Hm. die aus der Konfiguration unter /usr/local/etc/raspiBackup.conf
> 
> ```
> # GENERATED - DO NOT DELETE 
> UUID=2d8a95b4-e360-4dc4-a49c-2eeb710c5e1f
> ```

Die UUID ist mittlerweile obsolet :wink: 

> Ansible stellt einen Zustand eines Systems her, das funktioniert so, das ich tatsächlich eigentlich gar kein Backup brauche, weil ich mein System jederzeit aus Git mit dem Ansible wieder genau so aufsetzen kann.

So kenne ich auch Ansible das man damit ein System aufzusetzt. 

> dieses sichere bereits mit einem eigenen Script was im wesentliche alle Docker-Container runterfährt, dann ein tar.gz des Persistenz-Verzeichnisses macht (mit etwas Houskeeping) und alle Docker-Container wieder hochfärt.

Ok. Ich habe Nutzer von raspiBackup die sichern ihren ganzen Dockercontainer mit all seinen Daten. Aber so geht es natuerlich auch.

> ioBroker kann auch im Docker-Container laufen, ich finde das aber nicht so wirklich praktikable, wenn man sich nicht gut mit Docker auskennt, denn bei einigen Updates (js-Controller) muss man eine [...]

D.h. Du willst raspiBackup nutzen um native installed IOBroker zu sichern. Ich denke Du solltest nach einem userfriendly Weg suchen auch IOBroker in Docker laufen zu lassen wenn Du einen guten Backup fuer Docker dafuer hast. Dafuer jetzt raspiBackup zu nutzen geht - aber ist fuer mich da Du schon einen DockerBackup hast nicht sehr sinnvoll.

Ich kenne mich mit IOBroker nicht aus - habe aber gute Erfahrung mit Hilfe im IOBroker Forum gemacht. Hast Du dort mal Dein Problem geschildert ? Vielleicht habe die eine Idee wie Du IOBroker auch im Docker laufen lassen und sichern kannst :wink:

framps, 2022-09-12:
> und zu dem Support habe ich mit bisher tatsächlich noch keine Gedanken gemacht .. 

Wenn Du moechtest dass Dein SmarthomeReady genutzt wird wirst Du erhaltene git Issues zufriedenstellend beantworten muessen :wink: Und es wird sowohl Fragen als auch Problemberichte geben ...

Springjunky, 2022-09-13:
Prima, jede Menge Input zum Nachdenken und auch die Erkenntnis, das ich hier vermutlich mit Kanonen auf Spatzen schieße .. fassen wir mal zusammen:
* Docker-Backup ... einfache Möglichkeit, die im Rahmen meines Setup trägt 
* IOBroker Backup .. da habe ich doch noch etwas gefunden; das "hauseigene" Modul Backitup.

Ich habe in dem Sinne kein Problem mit IOBroker im Docker, aber habe festgestellt, das es für die Leute nicht so wirklich toll ist, die sich mit Pi4 und Linux beschäftigen, zusätzlich da noch die Docker-Hürde zu nehmen .. eineige Beispiele dazu.

* sämtliche Pfade, die im IOBroker angegeben werden sind erstmal _im_ Container, es sei denn man mappt sie explizit raus.. in etwa das History-Verzeichnis  
* gemountete SMB/NFS Verzeichnisse befinden sich dann ebenfalls nicht direkt im Zugriff durch den Container und hier muss man dann auch noch Wissen, das man im Docker-Container nich cifs/nfs Packages nachinstallieren muß.
* sämtliche Adapter, wie NodeRed, Vis, SimpleApi, Admin ... die Ports aufmachen müssen ebenfalls nach außen durchgereicht werden, ansonsten sind sie nur innerhalb eines Containers erreichbar, was keinen Sinn macht.
* usb-dev Devices wie z.B. ZigBee o.ä die auf /dev/ttyACM oder so liegen müssen wiedern _in_ den Container reingemappt werden .. das muss man wissen...auch, wie man dazu sein docker-compose anpasst.
* Verbindungen zu anderen Service, wie z.B. Influx müssen stets und *immer* per FQDN angegeben werden da localhost sich auf den Container beschränk.. klassiches Beispiel ist das Gespann aus Influx / Grafana. Wenn ioBroker nativ läuft ist Influx (im Docker) tatsächlich per localhost erreichbar, ebenso kann bei der Datenquelle in Grafana localhost angegeben werden. Läuft ioBroker im Container, dann *muss* immer die IP / der FQDN angeben werden in etwa raspberrpy.fritz.box:8086
.. da bin ich schon so oft selber drauf reingefallen
* Updates bedingen zum Teil eine Shell im Container.
* Docker selber hat da auch so seine Eigenheiten, was das pullen von neuen Images angeht .. 
* die Verwaltung des Docker-IOBroker-Images per Portainer ist sehr gut möglich, aber wieder ein neues Tool.
* und vor allem, wennn ich schaue, was für Probleme es hier in den [Kommentaren](https://smarthome.buanet.de/2020/10/iobroker-docker-container-updates-upgrades/) gibt nur, weil IOBroker im Container läuft .. dann würde ich jedem empfehlen, der sich nicht etwas mit Docker auskennt das einfach mal zu lassen

Vielen vielen Dank für die ganzen Inputs ... ich denke, man kann dann hier schließen .. ich werde in den nächsten Wochen denn mal ein POC dazu machen, ob wirklich alles so klappt .. also Desaster-Recovery mit
* Ansible laufen lassen
* Docker Sicherung einspielen
* BackitUp von IOBroker einspielen ...
wenn das alles nicht klappt, dann schreibe ich noch ein Plugin, im die Docker-Container Gracefull zu stoppen und zu starten .. dann ist raspiBackup wieder drann ;-)

Thanks framps !!

Springjunky, 2022-09-13:
Nachtrag:
Hier noch ein Snippet um Docker-Container runter- und wieder hochzufahren .. hatte da einen Issue gesehen .. funktionert dann auch mit docker-compose

Der erste Befehl liefert hierbei die Namen der aktuell laufenden Instanzen

```
RUNNING_CONTAINER=$(docker container ls --format '{{.Names}}')
# alle laufenden Container stoppen
for container in ${RUNNING_CONTAINER} ; do
  echo  "Stoppe => $container "
  docker container stop $container
done
```
und alle wieder starten
```
for container in ${RUNNING_CONTAINER} ; do
    docker container start $container
done
```

framps, 2022-09-13:
> man kann dann hier schließen

`man` bist `Du` :wink: 

> * dann ist raspiBackup wieder drann ;-)

Erstelle dann einfach einen neuen Issue.

Danke fuer den Tipp zum Stoppen der Container. Du beziehst Dich dabei vermutlich auf #534. Dort ist wohl das Problem das der PiHole Container sich automatisch restartet. Ansonsten werden alle Container beim Stoppen von Docker gestoppt.