framps/raspiBackup #538: rsync: (receiver|generator) chown file failed: Operation not permitted (1)

sirgooey, 2022-09-15:
## Note
raspiBackup is supported on RaspberryOS as operating system and Raspberry HW only. There are environments out there which successfully run raspiBackup but any support requests on unsupported environments will be rejected. 

If you run raspiBackup on RaspberryOS and a Raspberry please execute all following actions first and provide all requested information:

* Check whether the error messages of raspiBackup were listed on the messages page and all suggested actions don't fixes the issue. [English](https://linux-tips-and-tricks.de/en/rmessages) - [German](https://linux-tips-and-tricks.de/de/fehlermeldungen)
* Use your preferred search engine and look up the error message number
* Attach raspiBackup debug log. Just **DRAG AND DROP** raspiBackup.log into the issue and don't copy and paste the contents of raspiBackup.log into the issue. Don'y copy and paste the log into the issue.

#### If any requested information was not provided the issue will be rejected.

## Note
Issues are closed automatically after 2 weeks if there is no activity any more

First and foremost, thank you for all your work! I appreciate I am probably just one small mistake away from this working, so thank you in advance.

[raspiBackup_short.log](https://github.com/framps/raspiBackup/files/9578417/raspiBackup_short.log)

I have recently updated my Pi 3B from Stretch to Bullseye.
I was using raspiBackup for backups, so I set it up again on Bullseye.

The destination for the backups is the same, an unRAID share accessed via NFS.
However, any backups I run (rsync or tar) produce several of these kinds of errors, and fail:

`rsync: [generator] chown "/media/nfs/rpi01-3b-a020d3/rpi01-3b-a020d3-rsync-backup-20220915-172021/opt/pihole" failed: Operation not permitted (1)
rsync: [receiver] chown "/media/nfs/rpi01-3b-a020d3/rpi01-3b-a020d3-rsync-backup-20220915-172021/etc/dhcp/.dhclient.conf.C3aFTP" failed: Operation not permitted (1)
`
I've attached a somewhat sanitised output of raspiBackup.log which excludes most of the above errors.

The unRAID install (up to date) has the disks using XFS. 
The share is exported as below:
`root@host:~# exportfs -v
/mnt/user/netbackups
                192.168.88.166(async,wdelay,hide,no_subtree_check,fsid=100,sec=sys,rw,secure,root_squash,no_all_squash)`

On the RasPi, the fstab entry is (as suggested from one of your short articles):
192.168.88.90:/mnt/user/netbackups   /media/nfs  nfs  defaults,noatime,noauto,x-systemd.automount 0 0`

The Strech fstab entry was quite similar:
`192.168.88.90:/mnt/user/netbackups   /media/nfs  nfs  noauto,x-systemd.automount,x-systemd.device-timeout=10,timeo=14,x-systemd.idle-timeout=1min 0 0`

Mount output:
`pi@rpi01-3b-a020d3:~ $ mount | egrep nfs
192.168.88.90:/mnt/user/netbackups on /media/nfs type nfs4 (rw,noatime,vers=4.2,rsize=1048576,wsize=1048576,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,clientaddr=192.168.88.166,local_lock=none,addr=192.168.88.90)`

When the backup starts, folders, files etc are created fine on the destination, however permissions are as below (small extract, from the Pi):
`   0 drwx------ 1 nobody nogroup    0 Sep 15 17:20 boot
   0 drwx------ 1 nobody nogroup    0 Sep 15 17:20 dev
   0 drwx------ 1 nobody nogroup    0 Sep 15 17:20 etc
   0 drwx------ 1 nobody nogroup    0 Sep 15 17:20 home`

From the same Pi, I can see correct permissions on backups that were run under Stretch:
`   0 drwxr-xr-x 1 root   root      10 Jan  1  1970 boot
   0 drwxr-xr-x 1 root   root      10 Apr 28 16:32 dev
   0 drwxr-xr-x 1 root   root     148 May 21 21:26 etc
   0 drwxr-xr-x 1 root   root      51 May 21 21:33 home`
Similarly under _home_:
`total 4.0K
   0 drwxr-xr-x 1 root root   51 May 21 21:33 .
   0 drwxr-xr-x 1 root root   38 Jun 28  2021 ..
   0 drwxr-xr-x 1 pi   pi    293 May 21 21:33 pi
4.0K -rw-r--r-- 5 root root 1.9K May 29  2019 raspiBackup.log`

I have already tried to force mounting using NFSv3, no change.
Any help would be appreciated, I cannot seem to pin this down.
Thank you in advance!

sirgooey, 2022-09-15:
I've temporarily worked around this by adding an NFS rule on the unRAID server, though I am curious if there is another way.
`192.168.88.166(sec=sys,rw,**no_root_squash**)`

framps, 2022-09-15:
> I've temporarily worked around this by adding "**no_root_squash**" to the NFS rule

I have no clue why you were able to create rsync backups with Stretch without `no_root_squash` on your nfs server config. raspiBackup requires root access on the backupspace in order to save all Linux attributes in order to restore them. 

 `no_root_:squash` is mandatory on the nfs server side for raspiBackup.

sirgooey, 2022-09-16:
Acknowledged; you are obviously correct, and it's entirely possible I had some settings for this before the Pi upgrade which I've forgotten about.
For future reference / searches:
When backing up to an unRAID NFS share using raspiBackup, the share needs to be permissioned with `no_root_squash`; if it is not, `rsync: chown $filelocation failed: Operation not permitted (1)` will occur and the backup will fail.
The unRAID setting is under Shares > RelevantShare, under "NFS Security Settings".
I have mine configured as below, which in theory should block access to any IPs other than the one listed (this seems to be the case in my brief testing):
[image]

Thank you! Issue can be closed, not a software issue, PEBKAC...

closed 2022-09-16 by sirgooey