framps/raspiBackup #541: Rsync Restore with verbose - rsync: --deletev: unknown option

whowantsmybigdata, 2022-10-01:
Starting with version 0.6.7 I got an error when using rsync restore with `DEFAULT_VERBOSE=1`:
```
rsync: --deletev: unknown option
rsync error: syntax or usage error (code 1) at main.c(1749) [client=3.2.3]
```
Looking around (for the first time) in the raspiBackup.sh I think I found the problem (and a workaround solution).

[line 2706](https://github.com/framps/raspiBackup/blob/master/raspiBackup.sh#L2706) reads:
`DEFAULT_RSYNC_BACKUP_OPTIONS="-aHAx --delete"`

 [line 5459](https://github.com/framps/raspiBackup/blob/master/raspiBackup.sh#L5459) combines this options (among others not related to this issue) with $verbose (= `v` if triggered in the .conf) added to the rsync-options in the resulting command when triggered with `DEFAULT_VERBOSE=1` in the raspiBackup.conf:
```
local cmd="rsync $progressFlag --numeric-ids ${RSYNC_BACKUP_OPTIONS}${verbose} ${RSYNC_BACKUP_ADDITIONAL_OPTIONS} $excludePattern \"$ROOT_RESTOREFILE/\" $MNT_POINT"
```

If you put in the variables (everything non related to this issue is excluded) it combines to `rsync -aHAx --deletev` which doesn't exist and results in the error.

(My) **Solution:**
move `--delete` from lines 2706 to [line 2707](https://github.com/framps/raspiBackup/blob/c6deca69b95d7642706af61a68f7b6b199f1be5c/raspiBackup.sh#L2707) 
```
DEFAULT_RSYNC_BACKUP_OPTIONS="-aHAx"
DEFAULT_RSYNC_BACKUP_ADDITIONAL_OPTIONS="--delete"
```
the combined command will look like `rsync -aHAxv --delete` when `DEFAULT_VERBOSE=1` is set in raspiBackup.conf which does what it's supposed to do.

framps, 2022-10-02:
Good catch :+1: Your solution works but is a workaround.

I created a fix in branch #master_541. Would you please verify the fix?

whowantsmybigdata, 2022-10-03:
that was a workaround for sure as it was late and I'm by far no expert in scripting...haha
just verified your fix - thanks!

closed 2022-10-03 by whowantsmybigdata

framps, 2022-10-03:
> no expert in scripting

That's my job :wink: 

Thank you very much you verified the fix. I'll roll in the fix into 0.7.6 today and you can grab 0.6.7 which includes the fix with `sudo raspiBackup.sh -U -S` tomorrow.

whowantsmybigdata, 2022-10-03:
Fair enough...

Very nice! Would really wish one day most foss projects will be maintained in this manner! Even after several years... chapeau und danke dir vielmals