framps/raspiBackup #542: Incremental backup doesnt work

buckbaggen, 2022-10-08:
Hi there,

My backup location is an Synology NAS. I’ve mounted it with nfs.

The backup is configured in RaspiBackup as Rsync. 

The backup works daily without errors but are not incremental.

First day 25gb, second day 50gb, third day 75gb… 

Where can i setup to make incremental backups? 

Another (noob) question… once installed and configured RaspiBackup… how do i re-enter the configuration? Now i always put in the install command, but i guess there is an other, shorter way? 

Install command: curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/installation/install.sh | sudo bash

With kind regards,

Buck

framps, 2022-10-08:
> The backup works daily without errors but are not incremental.

Did you check [FAQ17](https://www.linux-tips-and-tricks.de/en/faq/#a17) ?

> how do i re-enter the configuration?

Don't reinstall raspiBackup all the time. Just call `sudo raspiBackupInstallUI.sh` to start the installer and modify any options. But please note: If you use any more advanced options you have to modify the config file /usr/local/etc/raspiBackup.conf with an editor.

buckbaggen, 2022-10-08:
Hi,

Thanks for the quick reply!

I not realy understand FAQ17, when i run the commands in the FAQ i get this results, but the backup directory shows this.

[image]
[image]

And thanks for the start command! :)

framps, 2022-10-08:
In your screenshot you see the first backup uses 4.9GB. The first backup is a full backup all the time. Next backups just use < 1GB so they are incremental backups.

At the end you see how much space the backups would use if no hardlinks would be used.

So everything is OK. Hardlinks are used in all backups other than the first backup :wink: 

Please close the issue if you're fine now :smile:

buckbaggen, 2022-10-08:
Hi,

Yes, that is confusing, it says:
4,9G 841M and 783M, so it should be in total around 6,5G

But the second screenshot (with the white background) says the backuo directory had 15,83GB on data on it. (it's directly on my Synology NAS)

framps, 2022-10-08:
3*5.6GB = 16.8GB which is a bit larger than 15.83GB which is reported on your Syno.
Keep in mind the numbers of `du` are rounded up by option `-h` and keep in mind the Syno display ignores the disk savings you get from hardlink usage everything is OK.

If you login into your Syno and execute `du -sh *` on the backup directory you will get the same output you get in your black screenshot.

Looks like this is an issue a lot of people complain about. Just read [this thread](https://community.synology.com/enu/forum/1/post/129217).

Again: Everything is OK. But you have to use the right tools/commands to respect hardlinks in the total space usage calculation :wink:

buckbaggen, 2022-10-08:
Thanks for the quick and clear support :)

Hopefully Synology will fix this sometimes...

Do you know if this is a Synology only problem? For example, if i use a QNAP NAS, wil the space allocation be displayed correctly?

framps, 2022-10-08:
Unfortunately not. I own a Syno but no QNAP. You may get an asnwer when you search for this topic and QNAP.

Anyhow it's not a bug. You usually want to know how much data you will copy when you copy a directory and then all hardlinks will be resolved and the number of bytes which will be copied is the size of all data inclusive the bytes linked with hardlinks. So the right way to solve this issue is to have a flag which toggles the inclusion and exclusion of hardlinked files in the directory size calculation.

buckbaggen, 2022-10-08:
Ah oke, clear. Learned something new thanks to you :-)

I’m not super experienced with Linux and hardlinks is new for me, but with your explaination it’s crystal clear for me, thanks a lot :)

closed 2022-10-08 by buckbaggen

framps, 2022-10-09:
Maybe [this page](https://linux-tips-and-tricks.de/en/raspibackupcategorye/572-how-do-hardlinks-work-with-rsync/) makes it even clearer how hardlinks work with rsync :wink: