framps/raspiBackup #544: Enhancement: docker container start und stop mit Extension

Springjunky, 2022-10-31:
Hallo,

könntest Du da einen Blick drüber werfen und evtl. aufnehmen... ein Merge-Request  ist aktuell nicht möglich bei mit (Layer 8 Problem)

#### raspiBackup_docker_pre.sh
```
#!/bin/bash
#######################################################################################################################
#
# Sample plugin for raspiBackup.sh
# called before a backup is started
#
# Function: Stops all running docker-container.
#
# See http://www.linux-tips-and-tricks.de/raspiBackup for additional information
#
#######################################################################################################################
#
#    Copyright (c) 2017 framp at linux-tips-and-tricks dot de
#
#    This program is free software: you can redistribute it and/or modify
#    it under the terms of the GNU General Public License as published by
#    the Free Software Foundation, either version 3 of the License, or
#    (at your option) any later version.
#
#    This program is distributed in the hope that it will be useful,
#    but WITHOUT ANY WARRANTY; without even the implied warranty of
#    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
#    GNU General Public License for more details.
#
#    You should have received a copy of the GNU General Public License
#    along with this program.  If not, see <http://www.gnu.org/licenses/>.#
#
#######################################################################################################################

GIT_DATE="$Date: 2022-05-12 21:36:43 +0200$"
GIT_COMMIT="$Sha1: 993abe4$"

MSG_EXT_DOCKER1="ext_docker_1"
MSG_EN[$MSG_EXT_DOCKER1]="RBK1000I: stopping Docker-Container: %s "
MSG_DE[$MSG_EXT_DOCKER1]="RBK1000I: beende Docker-Container: %s"

# gets a list of all running Docker container in one line sperated with blanks
function getRunningContainer () {
  if [[ -x  "$(which docker)"  ]] ; then
    local docker_bin=$(which docker)
    local running_container="$($docker_bin container ls --format '{{.Names}}' | tr '\n' ' ') "
    if [ ! -z "${running_container// }" ] ; then
     echo "$running_container"
    else
     echo ""
    fi
 fi 
}

# stops all given docker container
function stopRunningContainer () { 
  local running_container=$1	
  if [ ! -z "${running_container// }" ] ; then
    writeToConsole  $MSG_LEVEL_MINIMAL $MSG_EXT_DOCKER1 "$running_container"
    local docker_bin=$(which docker)
    $docker_bin container stop $running_container 2>&1>/dev/null
  fi
}

# ext_dockerContainer_pre is used in raspiBackup_docker_post.sh to start all 
# stopped container
ext_dockerContainer_pre="$(getRunningContainer)" 
stopRunningContainer "$ext_dockerContainer_pre"
```

#### raspiBackup_docker_post.sh
```
#!/bin/bash
#######################################################################################################################
#
# Sample plugin for raspiBackup.sh
# called after a backup finished
#
# Function: Display disk usage and % of disk usage change before and after backup
#
# See http://www.linux-tips-and-tricks.de/raspiBackup for additional information
#
########################################################################################################################
#
#    Copyright (c) 2017-2018 framp at linux-tips-and-tricks dot de
#
#    This program is free software: you can redistribute it and/or modify
#    it under the terms of the GNU General Public License as published by
#    the Free Software Foundation, either version 3 of the License, or
#    (at your option) any later version.
#
#    This program is distributed in the hope that it will be useful,
#    but WITHOUT ANY WARRANTY; without even the implied warranty of
#    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
#    GNU General Public License for more details.
#
#    You should have received a copy of the GNU General Public License
#    along with this program.  If not, see <http://www.gnu.org/licenses/>.#
#
#######################################################################################################################

GIT_DATE="$Date: 2022-05-12 21:36:43 +0200$"
GIT_COMMIT="$Sha1: 993abe4$"

# set any variables and prefix all names with ext_ and some unique prefix to use a different namespace than the script
# list of docker container stored in ext_dockerContainer_pre

MSG_EXT_DOCKER1="ext_docker_1"
MSG_EN[$MSG_EXT_DOCKER1]="RBK1000I: starting before stopped docker-container: %s "
MSG_DE[$MSG_EXT_DOCKER1]="RBK1000I: starte zuvor gestoppte Docker-Container: %s"

MSG_EXT_DOCKER2="ext_docker_2"
MSG_EN[$MSG_EXT_DOCKER2]="RBK1000I: Error with docker-container please check status: %s"
MSG_DE[$MSG_EXT_DOCKER2]="RBK1000I: Fehler bei Docker-Container bitte Status prüfen: %s"

# $MSG_LEVEL_MINIMAL will write message all the time
# $MSG_LEVEL_DETAILED will write message only if -m 1 parameter was used
if [ ! -z "${ext_dockerContainer_pre// }" ]; then
     writeToConsole  $MSG_LEVEL_MINIMAL $MSG_EXT_DOCKER1 "$ext_dockerContainer_pre"
     for container_to_start in $ext_dockerContainer_pre ; do 
       # the pre script checks, if docker was available at  /usr/bin/docker
       /usr/bin/docker start $container_to_start 2>&1>/dev/null
      if [ $? -ne 0 ] ; then
        writeToConsole  $MSG_LEVEL_MINIMAL $MSG_EXT_DOCKER3 ${container_to_start}
      fi
     done
fi
```
Beispiel Log-Ausgabe
```
--- RBK0009I: smartpi: raspiBackup.sh V0.6.7 (2022-10-11/0d32ab9) Mo 31. Okt 08:17:22 CET 2022 gestartet.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
--- RBK0151I: Backuppfad /mnt/SmarthomeAutoMount/SmartPiBackup/RaspiBackup mit Partitionstyp nfs4 wird benutzt.
--- RBK0008I: Services werden gestoppt: 'systemctl stop iobroker %1%1 systemctl stop webmin'.
...
--- RBK0267I: Erweiterung raspiBackup_docker_pre.sh wird aufgerufen.
--- RBK1000I: beende Docker-Container: DokuWiki Influxdb Deconz Grafana Portainer Motioneye PiHole  
--- RBK0081I: Backup vom Typ tgz wird in /mnt/SmarthomeAutoMount/SmartPiBackup/RaspiBackup/smartpi/smartpi-tgz-backup-20221031-081721 erstellt.
...
--- RBK0267I: Erweiterung raspiBackup_docker_post.sh wird aufgerufen.
--- RBK1000I: starte zuvor gestoppte Docker-Container: DokuWiki Influxdb Deconz Grafana Portainer Motioneye PiHole  
--- RBK0007I: Services werden gestartet: 'systemctl start webmin %1%1 systemctl start iobroker'.

```

... wenn nicht .. auch nicht schlimm .. bei mir klappt das prima ;-)

framps, 2022-10-31:
Der Code sieht doch sehr gut aus - und wie Du schreibst funktioniert er auch :+1: 

Zum docker Handlingcode kann ich nichts sagen da ich da nicht sattelfest bin.

Ein paar kleine Kommentare habe ich trotzdem:

1) Du solltest Messagenummern bei 2000 beginnen lassen. Die Sampleextensions nutzen schon den Numernraum 1000
2) Den Meldungstext wuerde ich immer mit einem Grossbuchstaben beginnen lassen
3) `if [ ! -z "${ext_dockerContainer_pre// }" ]; then` Warum nutzt Du nicht `[[ -n ${ext_dockerContainer_pre// }" }} ` ?

Wenn Du die Aenderungen noch vornimmst und den Code hier per dragndrop anhaengst und Du nichts dagegen hast wuerde ich ihn bei mir ins git stellen. Da Du den Code erstellt hast kannst Du gerne irgendetwas zu Dir als Autor in einem Kommentar schreiben :wink:

Springjunky, 2022-10-31:
Yeah, danke für den Review .. 

1. mache ich
2. auch das 
3. ... weil mein Audruck durchraus eine Zeichekette ergeben kann miz Blanks, also "  " und dann ist -n >0 .. sind zwar Blanks, aber eben nicht leer.

Aber das ganze gefällt mir nicht wirklich, ich werde das noch ändern, in dem ich die Anzal der laufenden Container zähle .. das ist dann auch besser nach zu vollziehen  ... (wc -l ist mein Favorit).

framps, 2022-10-31:
> 3\. weil mein Audruck durchraus eine Zeichekette ergeben kann mit Blanks, also "  " und dann ist -n >0 .. sind zwar Blanks, aber eben nicht leer.

Ok. Das war mit nicht bewusst. Vieleicht auch `getRunningContainer` so aendern dass keine Leerzeichen zurueckgegeben werden ? Nur so eine Idee. Aber `wc` funktioniert sicherlich auch.

Springjunky, 2022-10-31:
Habs mit wc gmacht .. siehe Pull-Request .. ;-)

framps, 2022-10-31:
Perfekt. Ist mir so auch lieber wenn ein PR reinkommt. Damit wird sichtbar dass nicht nur ich an raspiBackup arbeite und der Ersteller des Codes wird genant.

framps, 2022-10-31:
PR https://github.com/framps/raspiBackup/pull/546 merged

@Springjunky Please close the issue if you're fine now.

Springjunky, 2022-10-31:
Perfekt, nun bin ich auch in der Hall of Fame .. möge es den Docker-Nutzern helfen

commit e0e277c2 (2022-11-18): Fixed #544
commit c38cd280 (2022-11-18): Fixed #544
commit fbda2abf (2022-11-18): Fixed #544
closed 2022-10-31 by Springjunky