framps/raspiBackup #545: raspiBackup unable to write on NFS volume, manual operation OK

hmwelsch, 2022-10-31:
Fehlerbeschreibung und Diagnoseversuch:

**pi@rpi4sys1:~ $ sudo raspiBackup**
```
--- RBK0009I: rpi4sys1: raspiBackup.sh V0.6.7 (2022-10-11/0d32ab9) So 30. Okt 19:39:37 CET 2022 gestartet.
mkdir: das Verzeichnis „/mnt/nfsbak“ kann nicht angelegt werden: Die Datei existiert bereits
??? RBK0172E: Verzeichnis /mnt/nfsbak/rpi4sys1/rpi4sys1-rsync-backup-20221030-193935 kann nicht erstellt werden.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0010I: rpi4sys1: raspiBackup.sh V0.6.7 (0d32ab9) So 30. Okt 19:39:42 CET 2022 beendet mit Returncode 119.
--- RBK0026I: Debug Logdatei wurde in /home/pi/raspiBackup.log gesichert.
```
_Manuell kann ich das nicht nachvollziehen. Der Befehl läuft auf der Cmdline fehlerfrei durch. Ich habe folgende Tests gemacht._
**pi@rpi4sys1:~ $ sudo touch /mnt/nfsbak/test
```
pi@rpi4sys1:~ $ sudo ls -al /mnt/nfsbak/rpi4sys1/**
insgesamt 12
drwxr-xr-x 2 root root 4096 29. Okt 20:00 .
drwxrwxrwx 6 root root 4096 30. Okt 02:00 ..
```
**pi@rpi4sys1:~ $ sudo ls -al /mnt/nfsbak/**
```
insgesamt 28
drwxrwxrwx 6 root root 4096 30. Okt 02:00 .
drwxr-xr-x 3 root root 4096 29. Okt 19:39 ..
drwxrwxrwx 2 root root 4096  6. Mai 15:10 @Recycle
drwxr-xr-x 2 root root 4096 29. Okt 20:00 rpi4sys1
drwxr-xr-x 7 root root 4096 30. Okt 02:10 rpiserv3
drwxr-xr-x 7 root root 4096 30. Okt 01:21 rpiserv4
-rw-r--r-- 1 root root    0 30. Okt 19:42 test
```
_—> Zeigt problemlosen Zugriff inkl. Schreiben auf das Zielverzeichnis
—> Entferne Verzeichnis um raspiBackup von Scratch beginnen zu lassen_
**pi@rpi4sys1:~ $ sudo rm -r /mnt/nfsbak/rpi4sys1
pi@rpi4sys1:~ $ sudo rm /mnt/nfsbak/test
pi@rpi4sys1:~ $ sudo ls -al /mnt/nfsbak/**
```
insgesamt 24
drwxrwxrwx 5 root root 4096 30. Okt 19:44 .
drwxr-xr-x 3 root root 4096 29. Okt 19:39 ..
drwxrwxrwx 2 root root 4096  6. Mai 15:10 @Recycle
drwxr-xr-x 7 root root 4096 30. Okt 02:10 rpiserv3
drwxr-xr-x 7 root root 4096 30. Okt 01:21 rpiserv4
```
_—> Starte neu_
**pi@rpi4sys1:~ $ sudo raspiBackup**
```
--- RBK0009I: rpi4sys1: raspiBackup.sh V0.6.7 (2022-10-11/0d32ab9) So 30. Okt 19:44:49 CET 2022 gestartet.
--- RBK0031I: Prüfe ob eine neue Version von raspiBackup.sh verfügbar ist.
--- RBK0151I: Backuppfad /mnt/nfsbak mit Partitionstyp nfs wird benutzt.
df: /mnt/nfsbak/rpi4sys1: Datei oder Verzeichnis nicht gefunden
--- RBK0044I: Backup der Bootpartition wird in /mnt/nfsbak/rpi4sys1/rpi4sys1-rsync-backup-20221030-194448/rpi4sys1-backup.img erstellt.
??? RBK0178E: Erzeugung von /mnt/nfsbak/rpi4sys1/rpi4sys1-rsync-backup-20221030-194448/rpi4sys1-backup.img Datei endet fehlerhaft mit RC 1.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK1001I: Speicherauslastung vor dem Backup - Belegt: 107 MB Frei: 7656 MB
--- RBK1001I: Speicherauslastung nach dem Backup: Belegt: 110 MB Frei: 7653 MB
--- RBK1000I: CPU Temperatur vor und nach dem Backup: 46.7'C - 46.2'C
df: /mnt/nfsbak/rpi4sys1: Datei oder Verzeichnis nicht gefunden
(standard_in) 1: syntax error
(standard_in) 1: syntax error
(standard_in) 1: syntax error
(standard_in) 1: syntax error
(standard_in) 2: syntax error
--- RBK1001I: Partitionsauslastung vor dem Backup: Belegt: 0 Bytes Frei: 0 Bytes
--- RBK1002I: Partitionsauslastung nach dem Backup: Belegt: 0 Bytes Frei: 0 Bytes
/usr/local/bin/raspiBackup_disk_post.sh: Zeile 75: ((: != 0 : Syntaxfehler: Operator erwartet. (Fehlerverursachendes Zeichen ist "!= 0 ").
--- RBK0043I: Unvollständiges Backup in /mnt/nfsbak/rpi4sys1/rpi4sys1-rsync-backup-20221030-194448 wird gelöscht. Das kann etwas dauern. Bitte Geduld.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0010I: rpi4sys1: raspiBackup.sh V0.6.7 (0d32ab9) So 30. Okt 19:44:58 CET 2022 beendet mit Returncode 114.
--- RBK0026I: Debug Logdatei wurde in /home/pi/raspiBackup.log gesichert.
```
_—> Manuell das angemahnte Kommando_
**pi@rpi4sys1:~ $ sudo df /mnt/nfsbak/rpi4sys1**
```
Dateisystem           1K-Blöcke    Benutzt  Verfügbar Verw% Eingehängt auf
192.168.54.2:/rpibak 5765483616 4155890016 1609593600   73% /mnt/nfsbak
pi@rpi4sys1:
```
_—> User ratlos ;-)_

framps, 2022-10-31:
Interessanter Fehler :thinking: . Mir fehlen leider wichtige Informationen. Kannst Du das Debuglog von Deinem ersten und letzten Backupversuch mal per dragndrop anhaengen?

hmwelsch, 2022-11-01:
Hi, vielen Dank für die Reaktion. Leider habe ich wegen Nodered und M-Bus neu installieren müssen. Da ich zur Sicherheit komplett neu aufgesetzt habe, sind alle Daten weg. Das gilt leider auch für das Sicherungssarchiv, das ich intelligenterweise auch vor dem Neustart von raspiBackup ebenfalls gelöscht habe. Jetzt funktioniert bislang alles. Eine weitere Änderung war, dass ich einen anderen SSD-Adapter verwendet habe, da mir der erste früher schon mal Probleme bereitet hatte und das System jetzt aber produktiv genommen werden soll.

Alles in allem kann ich den alten Fehler nicht mehr reproduzieren und das neue System läuft problemlos.

Sorry, bzw. Gut so :-)

closed 2022-11-01 by hmwelsch

framps, 2022-11-01:
> sind alle Daten weg.

:cry: 

Fuer solche Faelle gibt es bei raspiBackup die Option `-M` mit der man einen quasi Snapshot erstellen kann der nicht dem normalen Backuprecycleprozess unterliegt. D.h. da haettest Du ein Backup auf irgendeine andere SSD, USB Platte o.ae. erstellen koennen. 

Ich habe schon geplant ein kleines Video ueber die Option `-M` auf YT zu erstellen um prominenter auf diese Option hinzuweisen. Auf der Webseite ist es schon beschrieben.

Natuerlich haette so ein Backup nichts genutzt wenn Du es auf demselben Backupspace wie die regelmaessigen Backups erstellt haettest :sob: 

Aber gut dass das Problem mit raspiBackup weg ist :+1: - obwohl es etwas unbefriedigend ist nicht zu wissen woran es lag. Hoffen wir mal dass es am SSD Adapter lag und der Fehler nicht irgendwann wieder auftritt. Jedenfalls koennen die Symptome durchaus darauf zurueckzufuehren sein.

hmwelsch, 2022-11-02:
> Aber gut dass das Problem mit raspiBackup weg ist 👍 - obwohl es etwas unbefriedigend ist nicht zu wissen woran es lag. Hoffen wir mal dass es am SSD Adapter lag und der Fehler nicht irgendwann wieder auftritt. Jedenfalls koennen die Symptome durchaus darauf zurueckzufuehren sein.

Hallo Peter,

ja, danke für den Hinweis. Hatte ich gerade kurz vor deiner Mail auch gesehen und gleich gemacht. Nutze ich gerade gerne heftig für die Neueinrichtung und Fortschreibung der Änderungen bzw. Erweiterungen.

BTW: Zu den Services hätte ich noch folgende Verständnisfragen.
Was genau sollte man denn bei einem jungfräulich aufgesetzten System an Services stoppen und wieder starten?
Ich richte das immer nur über dein Config Tool ein und mache das gerade
nur für Cron
und natürlich meine Erweiterungen, z.Zt. also Nodered.
SSH zu stoppen ist doch nur notwendig, wenn ich darüber zur Laufzeit Änderungen von anderer Stelle erwarten würde. Richtig?
NFS-Server stoppen, bedeutet doch hoffentlich auch nicht, dass mein Client den Backup auf ein NFS Target nicht mehr machen könnte. Das hat doch mit dem Server nichts zu tun. Oder?

NFS-Server und Samba will ich demnächst auch einrichten wenn ich mal die Serverlast durch Nodered und das Basissystem analysiert habe. Dann würde ich nämlich an dieses System eine alte große Festplatte hängen und als Target für den Backup all meiner RPIs und evtl. sogar MacOS einrichten. Muss dann mal schauen, ob das mit Sleep mode geht, oder ob ich ggf. per Raspi Port und Relais dafür Power ein-/ausschalten muss. Falls du da Erfahrung hast, kannst du mir gerne auch zu- oder abraten ;-) Ich lerne gerne aus Erfahrungen anderer und muss nicht alles selber erleiden.

Danke auf jeden Fall noch einmal für das super Tool und den Support.

LG Martin

framps, 2022-11-02:
Moin @hmwelsch 

> Was genau sollte man denn bei einem jungfräulich aufgesetzten System an Services stoppen und wieder starten?

Das kommt ganz darauf an was man auf seinem System am Laufen hat. I.d.R. sollten es Anwendungen sein die DBs nutzen oder sonstige Informationen im Speicher halten. 

ssh gehoert nicht dazu. nfsserver wohl nur wenn Deine Raspi diesen Dienst anbietet. Genaue Erfahrungen habe ich da nicht. 

> NFS-Server stoppen, bedeutet doch hoffentlich auch nicht, dass mein Client den Backup auf ein NFS Target nicht mehr machen könnte. Das hat doch mit dem Server nichts zu tun. Oder?

Nein. Dazu wird def NFS Client genutzt - nicht der Server.

> Falls du da Erfahrung hast, kannst du mir gerne auch zu- oder abraten ;-)

Nein. habe ich nicht. Du koenntest aber mal in [diesem Forum](https://forum-raspberrypi.de/forum/) vorbeisehen. Vielleicht findest Du da aehnliche Faelle oder Du erstellst einfach einen neuen Thread. Dort gibt es viele erfahrene und hilfsbereite Forenmitglieder die Dir auch in bei anderen Fragen zu Raspberry helfen koennen :wink:

PS: Bitte nutze hier meinen Nicknamen :wink: