framps/raspiBackup #551: Slack incoming webhook support for notification

sidevesh, 2022-11-13:
Can we add support for slack incoming webhook to this ?
Slack allows you to have separate webhooks send messages to separate channels which helps organise things a bit amongst different services pushing updates.
Adding support should be fairly simple and similar to other integrations since its just a webhook api call.

framps, 2022-11-13:
I like this idea. From a coding point of view I don't think it's a big issue. But it may be a configuration and testissue. Or even I'll not able to execute any tests because I only have a free Slack account and some features required a paid account.

I will investigate on this.

framps, 2022-11-13:
Just created a webhook in my free Slack account and sent me w test message via curl :smile: . So I don't think there is any show stopper to add this feature in next release. 

Unfortunately your feature request is too late to be included for next release 0.6.8.  A beta will be published in a couple of weeks.

framps, 2022-11-13:
Looks good. That's the format a message will look like:

[image]

New options will be:
```
DEFAULT_SLACK_WEBHOOCK_URL="<URL>"
DEFAULT_SLACK_NOTIFICATIONS="SFM"
```
Option `M` will append the raspiBackup messages to slack. Otherwise a success or failure message only will be sent.

@sidevesh Is there anything else you miss?

sidevesh, 2022-11-13:
@framps  that's perfect! and thanks for the incredibly swift response.
Really appreciate it and thanks for the amazing project that's already saved my Pi 4 twice!

framps, 2022-11-13:
I think I'm mostly done :wink: 

You can download the current code which supports slack with
`curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/raspiBackupDownloadFromGit.sh | sudo bash -s -- beta_slack
`

Would be great if you test it and give feedback.

framps, 2022-11-13:
Branch for this feature is https://github.com/framps/raspiBackup/tree/beta_slack

sidevesh, 2022-11-13:
thanks @framps , I am away from my Pi home server for some time but will test it and report back soon.

framps, 2022-11-13:
Take your time. But would be great if you would give feedback latest next week. I checked the updates I had to make for Slack and there are no changes in the main codepath. So I think I will merge the Slack support also in upcomming release 0.6.8 :wink:

closed 2022-11-20 by framps

sidevesh, 2023-01-11:
just an update, although very late, sorry for that, the slack support worked perfectly fine for my setup.
Thanks for implementing it.