framps/raspiBackup #553: Test and enable raspiBackup for Ubutunu

framps, 2022-11-16:


framps, 2022-11-17:
Issues:

1) LABELS not available for restore -> create .blkid file during backup
2) Installer does not accept keyboard if started directly from download
3) Remove `--unsupportedEnvornment` option requirement

framps, 2022-11-18:
There is an issue with `curl -L https://raspibackup.linux-tips-and-tricks.de/install | sudo bash`. Arrow keys are not processed.

Instead `curl -o install -L https://raspibackup.linux-tips-and-tricks.de/install; sudo bash ./install` has to be used.

framps, 2022-11-19:
Executed tests:

1) - [x] Backup and restore Ubuntu Desktop with rsync
2) - [ ] Backup and restore Ubuntu Desktop with rsync and -P
3) - [x] Backup and restore Ubuntu Server with rsync
4) - [x] Backup and restore Ubuntu Server with rsync and -P

framps, 2022-11-26:
The Ubuntu support is already available as an [inofficial beta](https://www.linux-tips-and-tricks.de/en/2-uncategorised/633-raspibackup-ubuntu-support).

framps, 2022-12-13:
> Debug Logdatei wurde in /home/plinux/raspiBackup.logr gesichert.

Bitte das Log uploaden.

framps, 2022-12-14:
Aus dem Log kann ich sehen dass Du offensichtlich den Restore auf einem Linux Mint vornimmst. Das mache ich auch bei mir und bekomme dasselbe Problem. Dummerweise ist die sfdisk Version auf dem Bullseye neuer und inkompatibel zum vorhergehende sfdisk was Du auf dem Mint hast.

```
20221213-132441 DBG 7124:                  --- label: dos
20221213-132441 DBG 7124:                  --- label-id: 0x083b206c
20221213-132441 DBG 7124:                  --- device: /dev/mmcblk0
20221213-132441 DBG 7124:                  --- unit: sectors
20221213-132441 DBG 7124:                  --- sector-size: 512
20221213-132441 DBG 7124:                  --- 
20221213-132441 DBG 7124:                  --- /dev/mmcblk0p1 : start=        8192, size=      524288, type=c
20221213-132441 DBG 7124:                  --- /dev/mmcblk0p2 : start=      532480, size=   121610240, type=83
```

Wenn Du im Backup die Zeile `sector-size: 512` auskommentierst wird der Restore erfolgreich sein :wink:

framps, 2022-12-14:
Ich habe uebrigens durch Deinen Fehlerbericht entdeckt dass ein paar Fehlermeldungen nur mit dem details Messagelevel ausgegeben werden :worried: . Im Log sieht man eine FehlerMeldung bzgl sfdisk - nur eben nicht auf der Konsole. Das habe ich eben gefixed. Wenn Du Dir noch mal den aktuellen Betacode downloadest hast Du die gefixte Version :wink:

Morpheus2018, 2022-12-14:
> Ich habe uebrigens durch Deinen Fehlerbericht entdeckt dass ein paar Fehlermeldungen nur mit dem details Messagelevel ausgegeben werden worried . Im Log sieht man eine FehlerMeldung bzgl sfdisk - [...]

Vielen Dank ich werde es beim nächsten Backup -d ausprobieren!
Andere Frage:
ist es möglich weitere exklude (Ordner / Pfade) fürs Backup anzugeben / auszuschließen?

framps, 2022-12-14:
> Andere Frage:

Bitte erstelle dazu ein Discussioneintrag :wink:

framps, 2023-06-24:
Created https://github.com/framps/raspiBackup/tree/master_ubuntu

framps, 2023-06-24:
Executed tests:

1) - [x] Backup and restore Ubuntu 23.04 Desktop with rsync
2) - [] Backup and restore Ubuntu 22.10 Desktop with rsync and -P
3) - [] Backup and restore Ubuntu 23.04 Server with rsync
4) - [] Backup and restore Ubuntu 23.04 Server with rsync and -P

closed 2023-06-26 by framps

framps, 2023-08-14:
Executed tests for 0.6.9

1) - [ ] Backup and restore Ubuntu 22.10 Desktop with rsync
2) - [ ] Backup and restore Ubuntu 22.10 Desktop with rsync and -P
3) - [ ] Backup and restore Ubuntu 22.10 Server with rsync
4) - [ ] Backup and restore Ubuntu 22.10 Server with rsync and -P