framps/raspiBackup #557: RBK0266E with NFS backup onto Synology even though no_root_squash is set

rossbacher, 2022-11-24:
## Setup

I'm trying to backup a raspberry pi onto a Synology NAS that is off site, so I did mount nfs through an ssh tunnel to make this more secure.

The exports setting for the NFS are:

```
/volume1/backup	localhost(rw,async,no_wdelay,insecure,no_root_squash,insecure_locks,sec=sys,anonuid=1025,anongid=100)
```

The `localhost` is due to the port forwarding via ssh.

nfs was mounted like this:

```
sudo mount -t nfs -o port=xxxx -o proto=tcp localhost:/volume1/backup /mnt/synology_backup
```

```
localhost:/volume1/backup on /mnt/synology_backup type nfs4 (rw,relatime,vers=4.0,rsize=131072,wsize=131072,namlen=255,hard,proto=tcp,port=xxxx,timeo=600,retrans=2,sec=sys,clientaddr=127.0.0.1,local_lock=none,addr=127.0.0.1)
```

I did look into the script what does lead to the error and I did my own experiments

if have a file on the rpi like this:

```
----r-xrwx 1 nobody nogroup    0 Nov 23 19:46 rossitest.fileattributes
```

and I copy it to the nfs share, I get

I oddly enough do not get the error that the script must be getting (see attached log file)

but I still do not get the same attributes:

```
----r-xr-x 1 root   root          0 Nov 24 21:13 rossitest.fileattributes
```

rossbacher, 2022-11-24:
Logfile: 

[raspiBackup.log](https://github.com/framps/raspiBackup/files/10087399/raspiBackup.log)

framps, 2022-11-24:
> I did mount nfs through an ssh tunnel to make this more secure.

I never had somebody who used a remote backup space but I don't see any reason why this shouldn't work with raspiBackup. 

I checked the log and found

`20221124-210156 DBG 3424:          --> supportsFileAttributes /mnt/<backup-path>
20221124-210156 DBG 3438:              --- ----r-xrwx # nobody # nogroup
cp: failed to preserve ownership for '//mnt/<backup-path>/raspiBackup.fileattributes': Invalid argument
20221124-210156 DBG 3444:              --- cp failed with rc 1
20221124-210156 DBG 3459:          <-- supportsFileAttributes 1
20221124-210156 DBG 3565:          --> findMountPath /mnt/<backup-path>
20221124-210156 DBG 3573:              --- Path: /mnt/<backup-path>
20221124-210156 DBG 3573:              --- Path: /mnt/synology_backup
20221124-210156 DBG 3581:          <-- findMountPath /mnt/synology_backup
20221124-210157 DBG 3832:          --> colorAnnotation 1
20221124-210157 DBG 3856:          <-- colorAnnotation 
??? RBK0266E: Access rights missing to create fileattributes on /mnt/synology_backup (Filesystem: nfs4).
20221124-210157 DBG 2363:          --> exitError 102
20221124-210157 DBG 2370:          <-- exitError 102`

raspiBackup creates a local file, copies this to the remote site and checks if the file attributes are preserved. The message `cp: failed to preserve ownership ` points out there is some issue on the server side. Did you read https://www.linux-tips-and-tricks.de/de/synology ? It's important do have no_squash defined for your exported nfs share.

rossbacher, 2022-11-24:
I think I did read most of everything on that page and that is my `/etc/exports` on the synology:

```
/volume1/backup	localhost(rw,async,no_wdelay,insecure,no_root_squash,insecure_locks,sec=sys,anonuid=1025,anongid=100)
```

so it is there. And when I replicated that behavior in `supportsFileAttributes` I did see the error one but now it actually works (when done from cli via sudo). Really odd.

We have Thanksgiving over here, I'll read some more on the synology page but I think I'm doing everything right. The export config is exactly the same as the one mentioned there.

framps, 2022-11-24:
> now it actually works (when done from cli via sudo). Really odd.

Strange ...

That's what I have in my Syno exports

```
/volume1/raspiBackup	192.168.0.0/24(rw,async,no_wdelay,no_root_squash,insecure_locks,sec=sys,anonuid=1025,anongid=100)
```
I think there is an issue with `localhost`. Try `*` first which will allow everybody to access the share and then nail it down to the IP used by the remote system.

rossbacher, 2022-11-30:
I don't think it is the ip, somehow the no root squash seems to not work or not work as expected the files get copied to the remote but they do have the wrong permissions.

Feels like a bug in Synology or something but I just can't get it to work and I seem to have the "correct" export settings. I also experimented with and without setting "unix permissions:" in the nfs setting but no difference.

framps, 2022-12-01:
> Feels like a bug in Synology 

Hm ... I think it's somehow an authorization issue. From time to time I get an issue because rsync doesn't work on a NAS. It's all the time `no_root_squash` was not used.

> I also experimented with and without setting "unix permissions:"

I created my nfs partition on the Syno just with the DSM UI - without any commands entered in the commandline. Then I also didn't modify any Linux access rights. I faced some issues I frankly didn't understand with the nfs versions. So maybe try to use a different nfs version (v3, v4 or v4.1).

If you plan to create a ticket for Synology please let me know the outcome of the discussion :wink: