framps/raspiBackup #560: Final Backup Notification not sent

framps, 2022-11-28:
### Discussed in https://github.com/framps/raspiBackup/discussions/559

<div type='discussions-op-text'>

<sup>Originally posted by **darkmoon2901** November 28, 2022</sup>
Hi guys,

in version V0.6.7 I got notifications (mail, telegram) when the backup was started and when it was finished. 

In the new version "raspiBackup.sh V0.6.8" I only get a notification, when the backup is started but not when it is finished.

Is there something I have to adjust in the config file after the upgrade from V0.6.7 to V0.6.8 to receive both messages again ?

Backup-Log:
######################################
--- RBK0009I: falcon1: raspiBackup.sh V0.6.8 - 2022-11-27 (25b237b) Mo 28. Nov 09:03:49 CET 2022 gestartet.
--- RBK0229I: Telegram benachrichtigt.
--- RBK0151I: Backuppfad /home/nfs/backup mit Partitionstyp nfs wird benutzt.
Interface:  wlan0
--- RBK0044I: Backup der Bootpartition wird in /home/nfs/backup/falcon1/falcon1-rsync-backup-20221128-090347/falcon1-backup.img erstellt.
256+0 Datensätze ein
256+0 Datensätze aus
1+0 Datensätze ein
1+0 Datensätze aus
--- RBK0158I: rsync Backup "/home/nfs/backup/falcon1/falcon1-rsync-backup-20221128-090347" wird erstellt.
--- RBK0085I: Backuperstellung vom Typ rsync gestartet. Bitte Geduld.
--- RBK0078I: Backupzeit: 00:36:22.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0159I: 5 Backups werden für den Backuptyp rsync aufbewahrt. Bitte Geduld.
Interface:  wlan0
--- RBK0017I: Backup erfolgreich beendet.
--- RBK0010I: falcon1: raspiBackup.sh V0.6.8 - 2022-11-27 (25b237b) Mo 28. Nov 09:53:38 CET 2022 beendet mit Returncode 0.
--- RBK0026I: Debug Logdatei wurde in /home/nfs/backup/falcon1/falcon1-rsync-backup-20221128-090347/raspiBackup.log gesichert.
######################################

Best regards Thomas 
</div>

framps, 2022-11-28:
There was some minor change in the notification handling: As long as you invoke raspiBackup in a command window there is no longer a notification sent. It's only sent if raspiBackup runs in the background or if you use the fake option.

But I see it's inconsistent because the start notification shouldn't be sent also. I will fix this.

framps, 2022-11-28:
Hm ... just detected even in the background there is no notification sent :cry: Will debug the issue right now.

framps, 2022-11-28:
See https://github.com/framps/raspiBackup/commit/089ce72dc5529e431fc36135fefc88c0c56e7836

closed 2022-11-28 by framps

PatrickGlatz, 2023-06-17:
Hi @framps 

what is the current status on this? I have just performed an interactive backup, but with the configuration which is working on my other raspis, my new raspberry pi didn't send a telegram message at the end:

My config looks like this

```
# Telegram token
DEFAULT_TELEGRAM_TOKEN="112233445:AAbbccdd-eeFFGGHHiijjKKllMMnn"
# Telegram target chatid
DEFAULT_TELEGRAM_CHATID="1234567890"
# Telegram notifications to send. S(uccess), F(ailure), M(essages as file), m(essages as text)
DEFAULT_TELEGRAM_NOTIFICATIONS="SF"
```

And the log file mentions
```
--- RBK0219I: Keine Backups werden smart recycled.
20230617-235720 DBG 6020:      <-- applyBackupStrategy
20230617-235720 DBG 4920:      --> cleanupTempFiles
20230617-235720 DBG 4927:      <-- cleanupTempFiles
20230617-235720 DBG 3369:      --> finalCommand
20230617-235720 DBG 3381:      <-- finalCommand
20230617-235720 DBG 4664:      --- Terminate now with rc 0
20230617-235720 DBG 3946:      --> colorAnnotation 1
20230617-235720 DBG 3970:      <-- colorAnnotation
--- RBK0017I: Backup erfolgreich beendet.
20230617-235721 DBG 3946:      --> colorAnnotation 1
20230617-235721 DBG 3970:      <-- colorAnnotation
--- RBK0010I: @HOSTNAME@: raspiBackup.sh V0.6.8 - 2023-06-01 (4de7751) Sat 17 Jun 23:57:20 BST 2023 beendet mit Returncode 0.
20230617-235721 DBG 4266:      --> sendEMail
20230617-235721 DBG 4388:      <-- sendEMail
20230617-235721 DBG 4487:      --- Masquerading some mount options
20230617-235721 DBG 4517:      --- Masquerading home directory name
20230617-235721 DBG 4522:      --- Masquerading hostname
20230617-235721 DBG 4527:      --- Masquerading sensitive non local IPs
20230617-235724 DBG 2097:      --> logFinish
20230617-235724 DBG 2108:          --- 0 2 0
20230617-235724 DBG 2117:          --- LOG_OUTPUT: 2
20230617-235724 DBG 2144:          --- DEST_LOGFILE: /backup/printpi/printpi-rsync-backup-20230617-235003/raspiBackup.log
20230617-235724 DBG 2145:          --- DEST_MSGFILE: /backup/printpi/printpi-rsync-backup-20230617-235003/raspiBackup.msg
20230617-235724 DBG 2150:          --- Logfiles used: /backup/printpi/printpi-rsync-backup-20230617-235003/raspiBackup.log and /tmp/raspiBackup.msg
20230617-235724 DBG 2155:          --- Logfiles used: /backup/printpi/printpi-rsync-backup-20230617-235003/raspiBackup.log and /backup/printpi/printpi-rsync-backup-20230617-235003/raspiBackup.msg
20230617-235724 DBG 2162:              *** cat /tmp/raspiBackup.logf
20230617-235724 DBG 2162:
20230617-235724 DBG 3946:          --> colorAnnotation 1
20230617-235724 DBG 3970:          <-- colorAnnotation
20230617-235724 DBG 2177:      <-- logFinish
20230617-235724 DBG 4624:      --> unLockMe
20230617-235724 DBG 4626:      <-- unLockMe
20230617-235724 DBG 4759:  <-- cleanup
```

Do you have any idea, why i don't receive a telegram message?

Thanks
Patrick

framps, 2023-06-18:
@PatrickGlatz There is no Telegram notification sent in interactive mode. 

If your other Raspberries send a notification it's most probably because they run an older raspiBackup version.

PatrickGlatz, 2023-06-20:
> I just changed the code and now a notification is sent all the time if a backup is created (https://github.com/framps/raspiBackup/commit/81636e6145737fdc55a158b72df29d63b3a6af76)

Hi @framps , I've read the above comment as if the notification is sent all the time and not only in scheduled run.

I will run a one time cronjob and see if that sends the telegram messages.

Thanks
Patrick