framps/raspiBackup #563: DEFAULT_ZIP_BACKUP=1 ,  Configuration Samples

jalsco, 2022-12-01:
https://www.linux-tips-and-tricks.de/en/configuration-samples

**1c. A Windows user wants to use pishrink to create a minimal backup image.**

"but this compressed dd image cannot be used by windisk32image"  ?????

So WHAT can be done with the compressed dd image ??  As it cannot be used by windisk32image..

framps, 2022-12-01:
> So WHAT can be done with the compressed dd image ?? As it cannot be used by windisk32image..

You save diskspace. The compressed file is smaller than the uncompressed image. If you need the image for restore uncompress it first and then use win32diskimager :wink:

jalsco, 2022-12-01:
ah - just - uncompress it first with zip,  and then **CAN** use it with Win32diskimager ?
I assume that applies to zip and dd. 
What about pishrink ? It seems easy enough to create an image with RaspiBackup and Pishrink...  but - again - the webpages are unclear about how  to - restore - a dd image that was post-processed by pishrink . 

BTW - raspibackup is amazing !  all your hardwork is appreciated . thank you ! 😊🙏🏻 

Final question : I read that the rsync backup are: first one, Full and then the following ones are Incrememental. 

I processed - to an NFS share - 3 times today ... all rsync... and all were about same size - 11.5GB. 
I expected first one to be 11.5GB (full system size)  and then 2nd and 3rd to be Incremental, so probably only about  1gb each, since they were done within a couple hours of the first one. But all were size of the full system. Is there some reason that they not being Incremental by default  ?  .. I set the the number of backups to keep at 3.

framps, 2022-12-01:
> What are the the pros and cons of each compression tool, Zip v PiShrink. man_shrugging

zip is an old rock solid Linux tool. PiShrink is a tool which will shrink you image as small as possible but it's not maintained any longer. There are a lot open PRs but last commit is more than one year old. 

> restore - a dd image that was post-processed by pishrink .

Just use the same tool you use for a dd image - disk32imager or the other Windows tools which can do this.

> But all were size of the full system. Is there some reason that they not being Incremental by default ?

Please read [FAQ17](https://www.linux-tips-and-tricks.de/en/faq#a17)

jalsco, 2022-12-01:
Ok - I read all that on FAQ17...
None of it seems to apply to the problem I am seeing. 
IS there something specific on that page that you are asking me to look at ? 

The problem is that `rsynch` backups alway do a full backup, and not do an incrememntal. 
The NFS share (  mounted to /backup ) is EXT4

Please help 🙏🏻

framps, 2022-12-01:
Please provide the debug log