framps/raspiBackup #564: Die Option "--rebootSystem" funktioniert zum falschen Zeitpunkt

bracewGitHub, 2022-12-01:
Hallo framp,

ich lasse ein CronJob gesteuertes raspiBackup über das"/usr/local/bin/raspiBackupNfsWrapper.sh"-Skript einmal die Woche in der Nacht laufen.
Im Wrapper-Skript ist der Aufruf: `raspiBackup.sh --timestamps --notifyStart --rebootSystem` hinterlegt.

Nach dem Backup von letzter Woche und auch diese Woche habe ich mich gewundert, dass ich zwar die E-Mail: "_raspiBackup.sh...gestartet_" erhalte, aber nicht die "_Backup erfolgreich beendet_".

Im Zielverzeichnis wurde zwar ein neues Backup angelegt. Es hatten sich dort aber bereits 6 Stück angesammelt, obwohl nur 3 Stück da seien sollten (`DEFAULT_KEEPBACKUPS="3"`). Die alten Backups wurden nicht aufgeräumt.

Sobald ich die Option `--rebootSystem` rausnehme und im Wrapper ein reboot an späterer geeigneter Stelle:
```
              raspiBackup.sh --timestamps --notifyStart
		rc=$?
		if (( $rc > 0 )); then
			echo "raspiBackup failed with rc $rc"
			exit $rc
		fi
              shutdown -r +2
```
nutze, wird ein neues Backup im Zielverzeichnis angelegt, die alten Backups aufgeräumt und neu gestartet. Also, alles wie gewünscht und Ok.

Deshalb bin ich der Meinung, dass der reboot mit der Option `--rebootSystem` zu früh geschieht, noch vor dem Aufräumen?

framps, 2022-12-01:
Peinlich :cry: Ja, Du hast Recht. 

So sieht der Code am Ende aus:

```
doit

if (( ! $RESTORE && $REBOOT_SYSTEM && ! $FAKE )); then
	if [[ -n "$DEFAULT_EMAIL" ]]; then
		sleep 1m								# allow MTA to send notification email
	fi
	reboot
fi
```
Das dumme ist dass `doit ` nie wieder rurueckkommt bei einem Backup aufruf :cry: 

Ich hatte da mal einen entsprechenden Kommentar stehen - aber irgendwann habe ich ihn wohl offensichtlich geloescht.

Aber die Option ist doch wegen Dir reingekommen und Du hattest sie verifiziert :thinking:

bracewGitHub, 2022-12-01:
`...und Du hattest sie verifiziert` 
Ja, hat auch bis 24.11.22 funktioniert! Danach nicht mehr. Hab heute auch mit 0.6.7 und 0.6.8 versucht. Bei beiden das gleiche.
Warum es auf einmal nicht mehr geht. Keine Ahnung!
Ich habe in der Zwischenzeit an diesem System nichts geschraubt, nichts gedreht, nichts gemacht. Ehrlich.

framps, 2022-12-01:
Ist kein Vorwurf. Ich versuche nur zu verstehen wie das passieren konnte. 

Der Fix ist kein Problem.

framps, 2022-12-02:
Fixed in https://github.com/framps/raspiBackup/commit/f70731ac6c868fb6c84de30b4cafc61e68a8e75e

framps, 2022-12-02:
@bracewGitHub  Der Reboot funktioniert nun. Im gegensatz zu damals habe ich es eben auch verfiziert.

Wenn Du willst kannst Du es auch noch verifizieren: Dazu folgendes ausfuehren:

```
curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/raspiBackupDownloadFromGit.sh | sudo bash -s -- master_reboot
--- Downloading raspiBackup.sh from git branch master_reboot into current directory ...
--- Retrieving commit meta data of raspiBackup.sh ...
--- Inserting commit meta data into downloaded file raspiBackup.sh ...
--- Use 'sudo ./raspiBackup.sh' now to start Version: 0.6.8 CommitSHA: f70731a CommitDate: 2022-12-02 CommitTime: 11:45:57Z$
```

Danach steht die gefixte Version in Deinem aktuellen Verzeichnis welches Du mit `sudo ./raspiBackup.sh` aufrufen kannst. Vorher nicht vergessen den Reboot per Config wierder einzuschalten :wink: 

Achte aber darauf dass dort ein Punkt am Anfang steht. Ohne den nutzt Du Deine alte raspiBackup Verison.

bracewGitHub, 2022-12-02:
Wie im ersten Post geschrieben, nutze ich raspiBackup über das "/usr/local/bin/raspiBackupNfsWrapper.sh"-Skript. 
Ich vermute, wenn ich die neue Version via curl runterlade und im /home Verzeichnis ausführe, wird die Verbindung via NFS vorher nicht hergestellt, oder?

framps, 2022-12-02:
Stimmt. Das habe ich vergessen :cry: D.h. Du muesstest den mount noch vorher manuell ausfuehren.

bracewGitHub, 2022-12-02:
`curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/raspiBackupDownloadFromGit.sh | sudo bash -s -- master_reboot`
??? Missing jq. Please install jq first. Try 'sudo apt-get install jq'.

framps, 2022-12-02:
Das kleine Programm benoetigt noch jq. Wie in der Meldung steht musst Du noch `sudo apt-get install jq` eingeben :wink:

bracewGitHub, 2022-12-02:
War nur zur Info.

Ist der Aufruf: `sudo ./raspiBackup.sh` richtig, oder mit `--rebootSystem`, 
also z.B.: `sudo ./raspiBackup.sh  --timestamps --notifyStart --rebootSystem`?

framps, 2022-12-02:
> Ist der Aufruf: `sudo ./raspiBackup.sh` richtig,

Ja wenn Du in der Config alles entsprechend eingetragen hast.

> sudo ./raspiBackup.sh --timestamps --notifyStart --rebootSystem

So ist es egal was in der Config steht denn Du ueberschreibst die Config Optionen durch die Aufrufoptionen. Fuer den Test ist das die schnellste Methode.

bracewGitHub, 2022-12-02:
```
curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/raspiBackupDownloadFromGit.sh | sudo bash -s -- master_reboot
--- Downloading raspiBackup.sh from git branch master_reboot into current directory ...
--- Retrieng commit meta data of raspiBackup.sh ...
--- Inserting commit meta data into downloaded file raspiBackup.sh ...
--- Use 'sudo ./raspiBackup.sh' now to start Version: 0.6.8 CommitSHA: f70731a CommitDate: 2022-12-02 CommitTime: 11:45:57Z$

sudo mount -t nfs -o soft 192.168.0.20:/media/USBHDD1/raspiBackup /backup

sudo ./raspiBackup.sh --timestamps --notifyStart --rebootSystem
12-02-2022 14:09:52 --- RBK0009I: epi: raspiBackup.sh V0.6.8 - 2022-12-02 (f70731a) Fr 2. Dez 14:09:52 CET 2022 gestartet.
12-02-2022 14:09:53 --- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
12-02-2022 14:09:56 --- RBK0031I: Prüfe ob eine neue Version von raspiBackup.sh verfügbar ist.
12-02-2022 14:09:59 --- RBK0151I: Backuppfad /backup mit Partitionstyp nfs4 wird benutzt.
12-02-2022 14:09:59 --- RBK0272I: Das System wird am Ende des Backuplaufes neu gestartet.
12-02-2022 14:10:00 --- RBK0008I: Services werden gestoppt: 'systemctl stop cups-browsed ; systemctl stop cups ; systemctl stop motioneye ; systemctl stop motion ; systemctl stop webmin ; systemctl stop tailscaled ; systemctl stop cron'.
12-02-2022 14:10:05 --- RBK0267I: Erweiterung raspiBackup_mem_pre.sh wird aufgerufen.
12-02-2022 14:10:06 --- RBK0267I: Erweiterung raspiBackup_temp_pre.sh wird aufgerufen.
12-02-2022 14:10:06 --- RBK0267I: Erweiterung raspiBackup_disk_pre.sh wird aufgerufen.
12-02-2022 14:10:06 --- RBK0081I: Backup vom Typ rsync wird in /backup/epi/epi-rsync-backup-20221202-140949 erstellt.
12-02-2022 14:10:07 --- RBK0044I: Backup der Bootpartition wird in /backup/epi/epi-rsync-backup-20221202-140949/epi-backup.img erstellt.
12-02-2022 14:10:33 --- RBK0044I: Backup des Partitionlayouts wird in /backup/epi/epi-rsync-backup-20221202-140949/epi-backup.sfdisk erstellt.
12-02-2022 14:10:33 --- RBK0046I: Backup des Masterbootrecords wird in /backup/epi/epi-rsync-backup-20221202-140949/epi-backup.mbr erstellt.
12-02-2022 14:10:34 --- RBK0133I: Verzeichnis /backup/epi/epi-rsync-backup-20221201-122618 wird für Hardlinks benutzt.
12-02-2022 14:10:34 --- RBK0158I: rsync Backup "/backup/epi/epi-rsync-backup-20221202-140949" wird erstellt.
12-02-2022 14:10:34 --- RBK0085I: Backuperstellung vom Typ rsync gestartet. Bitte Geduld.
12-02-2022 14:37:00 --- RBK0078I: Backupzeit: 00:26:54.
12-02-2022 14:37:01 --- RBK0033I: Bitte warten bis aufgeräumt wurde.
12-02-2022 14:37:01 --- RBK0267I: Erweiterung raspiBackup_mem_post.sh wird aufgerufen.
12-02-2022 14:37:02 --- RBK1001I: Speicherauslastung - Vor dem Backup - Belegt: 163 MB Frei: 169 MB - Nach dem Backup: Belegt: 166 MB Frei: 105 MB
12-02-2022 14:37:02 --- RBK0267I: Erweiterung raspiBackup_temp_post.sh wird aufgerufen.
12-02-2022 14:37:02 --- RBK1000I: CPU Temperatur vor und nach dem Backup: 47.2'C - 47.2'C
12-02-2022 14:37:02 --- RBK0267I: Erweiterung raspiBackup_disk_post.sh wird aufgerufen.
12-02-2022 14:37:02 --- RBK1001I: Partitionsauslastung vor dem Backup: Belegt: 616.08 GiB Frei: 2.79 TiB
12-02-2022 14:37:03 --- RBK1002I: Partitionsauslastung nach dem Backup: Belegt: 616.52 GiB Frei: 2.79 TiB
12-02-2022 14:37:03 --- RBK1003I: Änderung freier Platz: -448.50 MiB (0,00 %%)
12-02-2022 14:37:04 --- RBK0159I: 3 Backups werden für den Backuptyp rsync aufbewahrt. Bitte Geduld.
12-02-2022 14:37:04 --- RBK0205I: Älteste Backup /backup/epi in wird gelöscht. Das kann etwas dauern. Bitte Geduld.
12-02-2022 14:51:17 --- RBK0017I: Backup erfolgreich beendet.
12-02-2022 14:51:17 --- RBK0010I: epi: raspiBackup.sh V0.6.8 - 2022-12-02 (f70731a) Fr 2. Dez 14:51:17 CET 2022 beendet mit Returncode 0.
12-02-2022 14:51:20 --- RBK0026I: Debug Logdatei wurde in /backup/epi/epi-rsync-backup-20221202-140949/raspiBackup.log gesichert.
Reboot scheduled for Fri 2022-12-02 14:54:20 CET, use 'shutdown -c' to cancel.

Connection to epi closed by remote host.
Connection to epi closed.
```
Backup wurde im Zielverzeichnis erstellt.
Es sind 3 Backups vorhanden, aufräumen hat funktioniert.
Es ist keine Start-E-Mail angekommen.
Es ist keine Stop-E-Mail angekommen.
Ich glaube der reboot ist von meinem `shutdown -r +2` im Wrapper her gekommen.

bracewGitHub, 2022-12-02:
Nein, der reboot ist doch aus dem ./raspiBackup.sh Skript gekommen, da Du dort +3 drinn stehen hast, und ich +2. Es waren aber 3 Minuten bis zum Shutdown.

framps, 2022-12-02:
Perfekt. Vielen Dank fuer Deine Verifikation :+1: 

Ich war mit zu 99.9% sicher das es jetzt funktioniert. Aber ich habe es immer lieber wenn derjenige, der das Problem gemeldet, hat den Fix verifiziert. Es liegen manchmal Randbedingungen vor die ich bei meinen Tests nicht habe. Z.B. kabe ich kein Wrapperscript genutzt.

D.h. ich werde diesen Fix in 0.6.8 noch reinrollen. D.h. morgen kannst Du Dir die aktuelle Version mit `sudo raspiBackup.sh -U -S` holen :wink:

framps, 2022-12-02:
> P.S. Den timestamp finde ich im Amerikanischem System nicht so gut.

Ich habe dazu mal [einen Issue](https://github.com/framps/raspiBackup/issues/566) erstellt :wink:

framps, 2022-12-02:
Published in 0.6.8

closed 2022-12-02 by framps

bracewGitHub, 2022-12-03:
Ich habe die Version via `sudo raspiBackup.sh -U -S` aktualisiert.

Ein vorgezogenes Cron gesteuertes raspiBackup hat heute funktioniert. 
Beide E-Mails sind angekommen.
Es sind die letzten 3 Backups im Verzeichnis.
Der Reboot hat zum richtigen Zeitpunkt funktioniert.

Alles wieder prima :slightly_smiling_face: