framps/raspiBackup #569: Fehler beim aktualisieren von raspiBackupInstallUI.sh

bracewGitHub, 2022-12-03:
Hallo,

ich habe heute
`sudo raspiBackupInstallUI.sh` 
gestartet und 
`Aktualisiere Komponenten`, `P2 Aktualisiere raspiBackupInstallUI.sh (0.4.3.6 -> 0.4.4)` 
ausgewählt.

Beim erneuten Aufruf wird ein Fehler angezeigt:
```
sudo raspiBackupInstallUI.sh
/usr/local/bin/raspiBackupInstallUI.sh: 2: Syntax error: newline unexpected
```

Der Inhalt von `/usr/local/bin/raspiBackupInstallUI.sh` sieht nun ganz anders aus, als der Inhalt der gleichnamigen Datei im letzten Backup.

Ersten Zeilen der neuen `/usr/local/bin/raspiBackupInstallUI.sh`:
```
<!DOCTYPE html>
  
<html lang="en-us" dir="ltr">
<head>

<!-- 
	<script async src="//pagead2.googlesyndication.com/pagead/js/adsbygoogle.js"></script>
-->
  
  <!-- 02122020 
  <script data-ad-client="ca-pub-3885210057785859" async src="https://pagead2.googlesyndication.com/pagead/js/adsbygoogle.js"></script>
-->
```

Erste Zeilen der Datei im letzten Backup:
```
#!/bin/bash
#######################################################################################################################
#
# Script to download, install, configure and uninstall raspiBackup.sh using windows.
# Commandline installation is also possible. Use option -h to get a list of all options.
#
# Visit http://www.linux-tips-and-tricks.de/raspiBackup for latest code and other details
#
# Credits to following people for their translation work
#	  FI - teemue
#
#######################################################################################################################
```

framps, 2022-12-03:
Es sind offensichtlich mehrere Sachen zusammen die zu dem Problem gefuehrt haben.

1. Meine Webseite liefert keinen 404 zurueck sondern eine Fehlerwebseite wenn eine Datei nicht gefunden wird
2. In der Release 0.4.3.6 wird das leider noch nicht abgefangen. Deshalb hast Du diese Fehlerseite downloaed.
3. Die DownloadURLs wurden geaendert so dass immer ein 404 zurueckgegeben wird.

Ich habe versucht rauszufinden welche URL bei Dir nicht stimmte aber alle URLs die ich bei 0.4.3.6 getestet habe existieren. D.h. ich kann Dein Problem leider nicht nachvollziehen :cry: 

Du kannst wie folgt manuell den Installer auf die aktuelle Version updaten:

```
 wget https://www.linux-tips-and-tricks.de/raspiBackup/raspiBackupInstallUI.sh
chmod +x raspiBackupInstallUI.sh
sudo mv raspiBackupInstallUI.sh /usr/local/bin
```

bracewGitHub, 2022-12-03:
> Du kannst wie folgt manuell den Installer auf die aktuelle Version updaten:

Ok, so habe ich die V0.4.4.1 (2022-11-27/21:03:30) bekommen. Danke!

framps, 2022-12-03:
:+1: Dann close ich den issue jetzt.

closed 2022-12-03 by framps

marco79cgn, 2023-01-05:
Zur Info: Ich hatte heute genau das gleiche Problem. Scheint also kein Einzelfall zu sein.

framps, 2023-01-05:
Vielen Dank fuer den Hinweis. Leider kann ich ohne ein Debuglog die Ursache nicht herausfinden. Ich denke Du nutzt einen aelteren Installer wo noch die alten Downloadlinks genutzt werden und irgendwo habe ich ein Redirect vergessen :cry: 

Update den Installer bitte manuell wie ich oben geschrieben habe damit diese moegliche Ursache ausgeschlossen ist. Wenn es dann weiterhin Probleme gibt bitte das Debuglog vom Installer zeigen :wink:

framps, 2023-01-07:
Ich glaube ich habe eben den fehlenden redirect gefunden und geadded :wink:

marco79cgn, 2023-01-07:
Super, freut mich. Ich kann's jetzt leider nicht mehr testen, aber meine Version war ziemlich sicher auch die `0.4.3.6`. War aber dank dieses Issues sehr schnell gefixt wie oben beschrieben und lief dann auch direkt wieder.

Habe mir vor wenigen Tagen den Pi geschrottet beim Update von buster auf bullseye und musste alles neu aufsetzen. Habe zwar das raspiBackup image nicht komplett eingespielt (weil das vermutlich nicht geklappt hätte aufgrund der breaking changes in bullseye), aber alleine dass alle Daten und Ordner noch auf File-Ebene vorhanden waren auf meiner Backup-Platte hat mir sehr viel Zeit gespart.

framps, 2023-01-07:
> aber alleine dass alle Daten und Ordner noch auf File-Ebene vorhanden waren auf meiner Backup-Platte hat mir sehr viel Zeit gespart.

Es ist immer gut wenn man mindestens ein - besser die letzten 4 woechentlichen Backups zu Hand hat. 

Ich habe gerade einen neuen Installer published und den auch auf all meinen Raspis hochgezogen und habe dabei das problem mit dem fehlenden Redirect gefunden. Auf einer Raspi die einen 0.4.3.6 Stand hatte konnte ich den Fehler trotzdem reproduzieren :cry: Es liegt noch an was anderem. 

Jedenfalls hilft mir dabei auch mein Backup dabei den vorherigen Stand des Installers zu holen und das Problem genauer zu analysieren. Es scheint noch ein anderer redirect zu fehlen :sad:

framps, 2023-01-07:
Es hat sich herausgestellt dass es ein Bug im Installer ist wenn er sich updated und nicht an einem fehlendem Redirect liegt. Das ist im neuen Installer gefixed. Nur leider kann ich den alten Installer der auf den Raspis liegt nicht fixen.

Somit bleibt in diesem Falle nur [diesem Hinweis wie man manuell den Installer updaten kann](url) zu folgen.