framps/raspiBackup #570: pushover in raspiBackup V. 0.6.8

franjo-G, 2022-12-06:
The pushover function works correctly in the simulation (-F).

--- RBK0283I: Pushover notified.

pushover is sent, and also received at the same time.

In the real backup without (-F) there is (with the same notification settings (SFM)) no send and logically also no receive.
(tested several times)

` :~ $ sudo raspiBackup -F
--- RBK0009I: bullseye64bit: raspiBackup.sh V0.6.8 - 2022-12-02 (f70731a) Tue  6 Dec 00:36:25 CET 2022 gestartet.
!!! RBK0124W: Simulationsmodus an.
--- RBK0151I: Backuppfad /backup mit Partitionstyp nfs4 wird benutzt.
!!! RBK0157W: Keine Systemd Services sind zu stoppen.
--- RBK0078I: Backupzeit: 00:00:00.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0159I: 10 Backups werden für den Backuptyp rsync aufbewahrt. Bitte Geduld.
--- RBK0017I: Backup erfolgreich beendet.
--- RBK0010I: bullseye64bit: raspiBackup.sh V0.6.8 - 2022-12-02 (f70731a) Tue  6 Dec 00:36:28 CET 2022 beendet mit Returncode 0.
--- RBK0283I: Pushover benachrichtigt.
--- RBK0026I: Debug Logdatei wurde in /home/franjo64/raspiBackup.log gesichert.

:~ $ sudo raspiBackup
--- RBK0009I: bullseye64bit: raspiBackup.sh V0.6.8 - 2022-12-02 (f70731a) Tue  6 Dec 00:37:15 CET 2022 gestartet.
--- RBK0151I: Backuppfad /backup mit Partitionstyp nfs4 wird benutzt.
!!! RBK0157W: Keine Systemd Services sind zu stoppen.
--- RBK0044I: Backup der Bootpartition wird in /backup/bullseye64bit/bullseye64bit-rsync-backup-20221206-003714/bullseye64bit-backup.img erstellt.
--- RBK0158I: rsync Backup "/backup/bullseye64bit/bullseye64bit-rsync-backup-20221206-003714" wird erstellt.
--- RBK0085I: Backuperstellung vom Typ rsync gestartet. Bitte Geduld.
--- RBK0078I: Backupzeit: 00:22:02.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0159I: 10 Backups werden für den Backuptyp rsync aufbewahrt. Bitte Geduld.
--- RBK0017I: Backup erfolgreich beendet.
--- RBK0010I: bullseye64bit: raspiBackup.sh V0.6.8 - 2022-12-02 (f70731a) Tue  6 Dec 01:05:46 CET 2022 beendet mit Returncode 0.
--- RBK0026I: Debug Logdatei wurde in /backup/bullseye64bit/bullseye64bit-rsync-backup-20221206-003714/raspiBackup.log gesichert.`

framps, 2022-12-06:
See #560

closed 2022-12-06 by framps

framps, 2022-12-06:
I just changed the code and now a notification is sent all the time if a backup is created (https://github.com/framps/raspiBackup/commit/81636e6145737fdc55a158b72df29d63b3a6af76)