framps/raspiBackup #571: Checking for hard link capability doesn't give enough time for the test link to be created

MxMarx, 2022-12-09:
I'm getting an issue relating when the script checks if the backup directory supports hard links. For some reason, it takes a couple seconds for the hard link to be created, so `supportsHardlinks()` checks for the test link before it shows up, so rsync fails with error RBK0164E.

I'm getting around this by adding ` sleep 3` after the link is copied. It's working fine with this addition, so I'm just posting this in case anyone else has the same issue.

```
touch /$1/$MYNAME.hlinkfile &>>$LOG_FILE
cp -l /$1/$MYNAME.hlinkfile /$1/$MYNAME.hlinklink &>>$LOG_FILE
sleep 3
links=$(ls -la /$1/$MYNAME.hlinkfile | cut -f 2 -d ' ')`
```

For reference, I'm backing up to an NFS share on OpenMediaVault using mergerfs.

framps, 2022-12-09:
Thank you very much you report this issue :+1: . This may also be the root cause for another issue I got recently (See https://github.com/framps/raspiBackup/discussions/568) . 

In any case I will change the test logic in next release to repeat the test 3 times with a delay of 3 seconds if the initial test fails without any delay. That way your mergerfs will be supported by raspiBackup without any manual patch.

rossbacher, 2022-12-16:
Hm, I made this change: 
```
        touch /tmp/$MYNAME.fileattributes &>>"$LOG_FILE"
        chown 65534:65534 /tmp/$MYNAME.fileattributes &>>"$LOG_FILE"
        chmod 057 /tmp/$MYNAME.fileattributes &>>"$LOG_FILE"

        sleep 3
```

and that is still failing with the same error. I also made some more changes to add more sleep calls but still the same result.

rossbacher, 2022-12-16:
I'll test this again once 0.6.9 is out, this was with 0.6.8.

framps, 2023-06-26:
Merged into development branch

closed 2023-06-26 by framps