framps/raspiBackup #577: raspiBackupDialog doesn't check whether the backup parition is mounted

framps, 2022-12-10:
On a system which uses dynamic mount and thus the backup parition is not mounted raspiBackupDialog creates following dialog

```
sudo ./raspiBackupDialog.sh 

 
 Please choose your preferred language
 Bitte waehle deine bevorzugte Sprache 
 

 German  = 1
 English = 2 
 
 
2
find: ‘/backup/idefix/idefix*’: No such file or directory
 Should a backup be created or an existing backup restored? 

 backup    1
 restore   2 
 
2
 Should the last backup be restored? y/N  
```

Would make sense to check whether the partition is available and quit with an error message otherwise.

framps, 2022-12-10:
@franjo-G I think it makes sense to add this missing check :wink:

franjo-G, 2022-12-10:
Ok i will add a message that raspiBackup is startet.

franjo-G, 2022-12-10:
I have added a mountcheck.
If the backup directory is not mounted a corresponding message appears

framps, 2022-12-10:
Implemented with https://github.com/framps/raspiBackup/commit/4602bb3370f87cb4b24c1a0e20083bf5f2bad897

closed 2022-12-10 by framps