framps/raspiBackup #599: [New Feature] - Discord Notification

Tytouc, 2023-01-13:
** New Feature: Discord Notification **
As a user I'd like to have the `raspiBackup.sh` status notified through a Discord text channel.
I expect this message to contain the same details as the email notification message.
This Discord notification needs to use Discord WebHook method to post message on a given channel.
The Discord message shall be colored depending on the backup status: `FAILURE `in `RED`, `SUCCESS` in `GREEN`

Exemple of messages screenshots:
Success (fake mode on):

[image]

Failure (because of CTRL-C):
[image]

framps, 2023-01-16:
I reviewed your [draft PR](https://github.com/Tytouc/raspiBackup/pull/1) you created on your repo and it looks fine to me. Do you plan to create a PR on this repo such that I can merge your feature request implementation :wink:

Tytouc, 2023-01-16:
Hi, yes, I'm ok with that. But, don't know what/how I have to do

framps, 2023-01-16:
Just create a PR on my repo instead of yours :wink:

Tytouc, 2023-01-17:
done

framps, 2023-01-17:
Great. Thank you very much :+1: . I just merged your PR.

closed 2023-01-17 by framps

framps, 2023-02-13:
@Tytouc FYI: https://github.com/framps/raspiBackup/issues/612