framps/raspiBackup #604: RC 32 when going to a different size SD card

p1r473, 2023-01-21:
**Note**
raspiBackup is supported on RaspberryOS and Ubuntu as operating system and Raspberry HW only. There are environments out there which successfully run raspiBackup but any support requests or issues on unsupported environments will be rejected. 

*If you have questions about raspiBackup please create a `discussion` entry instead of an `issue` entry in github*

If you have an issue with raspiBackup execute following actions first
1. Read the FAQ pages of raspiBackup 
   * [English](https://www.linux-tips-and-tricks.de/en/faq)
   * [German](https://www.linux-tips-and-tricks.de/de/faq)
1. Check whether the error messages of raspiBackup were listed on the messages page and all suggested actions don't fixes the issue
   * [English](https://www.linux-tips-and-tricks.de/en/raspibackupmessagese) I cant find anything in your FAQ on RC 32
   * [German](https://www.linux-tips-and-tricks.de/de/raspibackupmeldungen)
1. Use your preferred search engine and look up the error message number or the error message text you receive

If you still don't get any idea how to solve your issue with raspiBackup
1. Explain in detail what's going wrong - I am trying to restore a backup to a different sized card and I am getting error RC 32
1. Attach raspiBackup debug log. Just **DRAG AND DROP** raspiBackup.log into the issue and don't copy and paste the contents of raspiBackup.log into the issue. **If you don't attach a debug log the issue will be rejected**
2. [raspiBackup.log](https://github.com/framps/raspiBackup/files/10471555/raspiBackup.log)

**Note**
Issues are closed automatically after 2 weeks if there is no activity any more

p1r473, 2023-01-21:
Error seems to be `mount: /tmp/raspiBackup: wrong fs type, bad option, bad superblock on /dev/sdb2, missing codepage or helper program, or other error.`
I tried to change the fs of the sd card prior to recovery with `mkfs.ext4 /dev/sdb`

p1r473, 2023-01-21:
I can replicate error by manually doing:
```
mount /dev/sdb2 /tmp/raspiBackup
mount: /tmp/raspiBackup: wrong fs type, bad option, bad superblock on /dev/sdb2, missing codepage or helper program, or other error.
```
[image]

framps, 2023-01-21:
In the debug log I find

```
20230120-233415 MSG 5771:                  --- RBK0054I: Formating second partition (root partition) /dev/sdb2. 
mke2fs 1.46.2 (28-Feb-2021)
Creating filesystem with 32701952 4k blocks and 8175616 inodes
Filesystem UUID: a1051992-5d75-4336-a5b7-f1a8c0272db6
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208, 
	4096000, 7962624, 11239424, 20480000, 23887872

Allocating group tables:   0/998       done                            
Writing inode tables:   0/998       done                            
```
and then sudddenly 

```
Creating journal (131072 blocks): 20230120-233737 DBG 2332:      *** cat /etc/os-release
20230120-233737 DBG 2332:          PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
```

which means when the second partition was formatted successfully with ext4 for some unknown reasons the restore doesn't continue the normal process. All of the sudden the restore starts again :thinking: 

For me it looks like you invoked raspiBackup with all correct parameters for the restore,  the restore started and formatted the second partition with ext4 successfully. But then for some unknown reasons the raspiBackup seems to be called again with the same parameters. Did you do anything special when the restore was in progress?

p1r473, 2023-01-22:
No sir, I am invoking with this. Nothing special. Using latest version (0.6.8)
`sudo raspiBackup.sh -d /dev/sdb /mnt/tmp/Harbormaster/Harbormaster/Harbormaster-tgz-backup-20230107-030001/`

I tried the restore again with the same card, just in case anything may be different, but it fails with the same error. I also see `Terminate now with rc 102`
[raspiBackup.log](https://github.com/framps/raspiBackup/files/10475234/raspiBackup.log)

```
--- RBK0009I: Monkeebutt: raspiBackup.sh V0.6.8 - 2022-12-15 (120287b) started at Sun 22 Jan 2023 01:55:35 PM EST.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
--- RBK0138I: Using bootbackup /mnt/tmp/Harbormaster/Harbormaster/Harbormaster-tgz-backup-20230107-030001/Harbormaster-tgz-backup-20230107-030001.tgz.
--- RBK0068I: Using bootpartition backup files starting with /mnt/tmp/Harbormaster/Harbormaster/Harbormaster-tgz-backup-20230107-030001 from directory Harbormaster-backup.
!!! RBK0018W: Target /dev/sdd with 125.00 GiB is larger than backup source with 119.08 GiB. root partition will be expanded accordingly to use the whole space.
!!! RBK0065W: Device /dev/sdd will be repartitioned and all data will be lost.
--- RBK0067I: Current partitions on /dev/sdd:
Number  Start   End       Size      Type     File system  Flags
 1      1.05MB  269MB     268MB     primary  fat16        lba
 2      269MB   134217MB  133947MB  primary  ext4
--- RBK0066I: Device /dev/sdd will be overwritten with the saved boot and root partition.
--- RBK0069I: Bootpartition /dev/sdd1 will be formatted and will get the restored Boot partition.
--- RBK0070I: Rootpartition /dev/sdd2 will be formatted and will get the restored Root partition.
--- RBK0038I: Are you sure? y/N
y
--- RBK0050I: Restoring backup from /mnt/tmp/Harbormaster/Harbormaster/Harbormaster-tgz-backup-20230107-030001.
--- RBK0052I: Creating partitions on /dev/sdd.
--- RBK0004I: Adjusting second partition from 118.83 GiB to 124.74 GiB.
--- RBK0053I: Restoring first partition (boot partition) to /dev/sdd1.
--- RBK0054I: Formating second partition (root partition) /dev/sdd2.
--- RBK0055I: Restoring second partition (root partition) to /dev/sdd2.
??? RBK0213E: Mount /dev/sdd2 to /tmp/raspiBackup failed. RC 32.
--- RBK0033I: Please wait until cleanup has finished.
??? RBK0077E: Restore failed. Check previous error messages.
--- RBK0010I: Harbormaster: raspiBackup.sh V0.6.8 - 2022-12-15 (120287b) stopped at Sun 22 Jan 2023 02:00:16 PM EST with rc 102.
--- RBK0026I: Debug logfile saved in /root/raspiBackup.logr.
```

p1r473, 2023-01-22:
I just went out and bought the exact same SD card model as before, as I thought it may be due to a size difference, and this time I got error: 
```
tar: z: Not found in archive
tar: Exiting with failure status due to previous errors
??? RBK0035E: Backupprogram tgz failed during restore with RC 2.
```

But I also see in the logs:
`Terminate now with rc 117`
Here is the logs for the same exact sd card as original one that failed with RC 2/117:
[raspiBackup.log](https://github.com/framps/raspiBackup/files/10475213/raspiBackup.log)

p1r473, 2023-01-22:
Heres my config, if it helps. 
[raspiBackup.conf.txt](https://github.com/framps/raspiBackup/files/10475269/raspiBackup.conf.txt)

p1r473, 2023-01-22:
Ive also tried both cards with the backup made the day prior, and both backups give the same errors.
So it does not appear the issue is the backup itself.

framps, 2023-01-22:
I've never seen this error message `z not found` before.

It shouldn't be an issue with the SD card size you restore. As long as the data allocated on the source SD card and the restore SD card is able to receive the data this should work.

Unfortunately the man description of tar for rc 2 is quit useless :cry: 

```
RC2 - Fatal error.  This means that some fatal, unrecoverable
              error occurred
```

Please add option `-v` when you start your restore and provide the new debug log.

p1r473, 2023-01-22:
The one that had error RC 2 also had RC 117 in the logs.

Ive tried it -v on both cards, the exact same model and the slightly different model as they both gave slightly different errors. I invoked -v with `sudo raspiBackup.sh -v -d /dev/sdb /mnt/tmp/Harbormaster/Harbormaster/Harbormaster-tgz-backup-20230107-030001/`. I hope thats where I was supposed to put the -v as the output still looks very similar to previous logs.

On the new card thats the exact same model as the damaged original card: (RC 2, / 117)
[raspiBackup.log](https://github.com/framps/raspiBackup/files/10475451/raspiBackup.log)

On the first card I tried thats a slightly different model as the damaged original card:
[raspiBackup.log](https://github.com/framps/raspiBackup/files/10475459/raspiBackup.log)

framps, 2023-01-22:
> I hope thats where I was supposed to put the -v as the output still looks very similar to previous logs.

Yes. That's fine.

Looks like tar terminates immediately for some unknown reasons.

The command executed by raspiBackup to restore the backup is

 tar --exclude /boot --same-owner --same-permissions --numeric-owner  -x -v z -f "/mnt/tmp/@HOSTNAME@/@HOSTNAME@/@HOSTNAME@-tgz-backup-20230107-030001/@HOSTNAME@-tgz`

Please check whether the tar backup is OK:

If you use

 tar -t -z -f "/mnt/tmp/@HOSTNAME@/@HOSTNAME@/@HOSTNAME@-tgz-backup-20230107-030001/@HOSTNAME@-tgz

you should get a list of all files included in the backup. Do you get a huge file list?

p1r473, 2023-01-22:
Yes, I do get a huge file list when I perform that.

framps, 2023-01-22:
> I think the backup is ok.

Agree with you :+1: Let me check the debug log carefully now.

framps, 2023-01-23:
I created a branch with updated raspiBackup which includes additional debug statements which hopefully help to isolate your issue.

Pleas execute `curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/raspiBackupDownloadFromGit.sh | bash -s -- master_604` to download updated raspiBackup and invoke it with `./raspiBackup.sh` (note the leading period :wink: 

Please attach the debug log again.

p1r473, 2023-01-23:
Just want to say thanks Framps for your excellent and fast support and great communication!

Here are the logs.
On the new card thats the exact same model as the damaged original card: (RC 2, RC 117)
[raspiBackup.log](https://github.com/framps/raspiBackup/files/10482369/raspiBackup.log)

On the first card I tried thats a slightly different model as the damaged original card (RC 32, RC 102)
[raspiBackup.log](https://github.com/framps/raspiBackup/files/10482325/raspiBackup.log)

p1r473, 2023-01-24:
> PS: Thank you very much for your donation.
Thanks for the great support

I downloaded the new debug the same way (let me know if this is wrong) `curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/raspiBackupDownloadFromGit.sh | bash -s -- master_604`

`lsblk -f` all cards show up and look good to me
One of these (I think sdb) is original card from original post from this thread. Other one (I think sdd) is a brand new card I just opened this afternoon.
```
NAME        FSTYPE FSVER LABEL UUID                                 FSAVAIL FSUSE% MOUNTPOINT
sdb         ext4   1.0         2689dd1c-750e-409a-8d4d-4148cfa57b01
├─sdb1      vfat   FAT16       86B4-B88F
└─sdb2      ext4   1.0         d3ef2d5b-213c-49cc-8904-cb430836f04a
sdd
├─sdd1      vfat   FAT16       86B4-B88F
└─sdd2      ext4   1.0         c79ecf50-7c58-4493-a80a-6b5e451ee0ac
mmcblk0
├─mmcblk0p1 vfat   FAT16       B857-5569                             205.5M    20% /boot
└─mmcblk0p2 ext4   1.0         db59553d-536c-4c51-b083-c7d956b71544   95.2G    13% /
```

`df -h` drives dont show up.
```
df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       117G   16G   96G  14% /
devtmpfs        3.7G     0  3.7G   0% /dev
tmpfs           3.7G  3.9M  3.7G   1% /dev/shm
tmpfs           1.5G   11M  1.5G   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
/dev/mmcblk0p1  256M   51M  206M  20% /boot
log2ram         4.9G  330M  4.6G   7% /var/log
tmpfs           747M   32K  747M   1% /run/user/1000
tmpfs           747M   28K  747M   1% /run/user/0

```
rmdir /media
rmdir /mnt
mkdir /media
mkdir /mount
sudo mount -v -t cifs //192.168.1.69/backup /media/ -o user=JJJ
sudo mount -v -t cifs //192.168.1.69/Harbormaster /mnt/ -o user=JJJ
tar --exclude /boot --same-owner --same-permissions --numeric-owner  -x -v z -f "/mnt/Harbormaster/Harbormaster-tgz-backup-20230107-030001/Harbormaster-tgz-backup-20230107-030001.tgz" /media
```

Wow, this weird error again. But I figured it out. The commands I copy and pasted from you say `-x -v z -f ` and not `-x -v 'z -f `. Also it doesnt like the `/media`.  And this didnt generated any logs to send in /root/
```
tar: z: Not found in archive
tar: /media: Not found in archive
tar: Exiting with failure status due to previous errors
```
So here it is with a `-` in front of the `z`. **But it still doesnt like the /media argument**
`tar --exclude /boot --same-owner --same-permissions --numeric-owner  -x -v z -f "/mnt/Harbormaster/Harbormaster-tgz-backup-20230107-030001/Harbormaster-tgz-backup-20230107-030001.tgz" /media`
```
tar: /media: Not found in archive
tar: Exiting with failure status due to previous errors
```

Note about cards:
Original card that died was Sandisk Pro 128gb
First post in this thread is Sandisk 128gb (not pro)
The 2 new cards I bought and tried are both also Sandisk Pro 128gb
So when I say slightly different size, it is not a big different, they are all 128gb, but may be slight variations (get message target ... with 125.00gb is larger than backup source with 119.08bg)

> Just to make sure the SD cards are not buggy
I bought **another brand new card** (different from before; fresh out of package from Amazon) to try it on (dont want to use USB and complicate the process) and it fails the same. This card is the same size as the original card that died that the backup was from

On the brand new NEW card (fresh card right out of package) thats the exact same model as the original backup: (RC 2, RC 117) - Sandisk Pro 128gb with Jan 7 backup
[raspiBackup.log](https://github.com/framps/raspiBackup/files/10484651/raspiBackup.log)

On the first card I tried thats a slightly different model as the damaged original card (RC 32, RC 102) - Sandisk 128gb with Jan 7 backup
[raspiBackup.log](https://github.com/framps/raspiBackup/files/10484745/raspiBackup.log)

And I have opened another brand new card, this time 256gb, which Ive tried to restore January 6 instead of January 7 to (**this will eliminate the backup and the card as fault, as its a different backup and another brand new card** - Sandisk Pro 256gb with Jan 6 backup : (RC 2, RC 117)
[raspiBackup.log](https://github.com/framps/raspiBackup/files/10484602/raspiBackup.log)

p1r473, 2023-01-24:
> The second one fails because the mount of the second partition on the restore SD card fails even its allocated and formatted.

And the first one is a generic tar error?

framps, 2023-01-24:
> > The second one fails because the mount of the second partition on the restore SD card fails even its allocated and formatted.
> 
> And the first one is a generic tar error?

Yes. Unfortunately I have no idea right now what's the root cause. That's why I need your help to debug the issue.

So now I think we're sure it's not the SD card which is buggy. Sorry I stress this but I have had a lot of issues because buggy SD cards were used for the restore.

I tried to recreate your issue (have just 16GB SD cards, no 128GB or 256GB card) and created a tgz from a tiny bullseye image. Then restored it with the updated script I created for you and detected there is a missing dash for z which caused the restore to fail. I fixed this and now I can restore the tgz with the updated script successfully on my side. I'm not able to recreate your issue :cry: 

That's why I asked you to restore the backup on the root partition manually. I executed the steps you executed on my side and detected there are some typos which I corrected below. And I also detected the target partition you use to restore the image is /media which is a cifs partition. Please use /dev/sdb2 because that's the partition created and formatted by raspiBackup.  This test should prove the tgz is OK and the target partition is correct formatted. At least that's what `lsblk -f` tells me. And then I hope to get a better error message that just RC2 from tar.

Please try again the following commands:
```
umount /media
umount /mnt
sudo mount /dev/sdb2 /media
sudo mount -v -t cifs //192.168.1.69/Harbormaster /mnt/ -o user=JJJ
tar --exclude /boot --same-owner --same-permissions --numeric-owner  -x -v -z -f "/mnt/Harbormaster/Harbormaster-tgz-backup-20230107-030001/Harbormaster-tgz-backup-20230107-030001.tgz" -C /media
```

p1r473, 2023-01-24:
> So now I think we're sure it's not the SD card which is buggy. 

For sure, fully understand. I have now tried the January 7 restore on 4 different cards. the original sandisk 128gb from original post, a sandisk pro 128gb that is same model as card from backup that was damaged/bad blocks, another brand new from package sandisk pro 128gb (same model as original backup), and finally, another brand new from package 256gb with January 6 backup. So I think it is not the card and also not the backup, as Ive tried 2 backups and 4 cards.

Clean up:
```
umount /media
umount /mnt
rm -rf /media
rm -rf /mnt
```

now the good stuff:
`sudo mount /dev/sdb2 /media`
```
mount: /media: wrong fs type, bad option, bad superblock on /dev/sdb2, missing codepage or helper program, or other error.
```

`rmdir /media`

`sudo mount /dev/sdb2 /media`
```
mount: /media: mount point does not exist.
```

`mkdir /media`

`sudo mount /dev/sdb2 /media`
```
mount: /media: wrong fs type, bad option, bad superblock on /dev/sdb2, missing codepage or helper program, or other error.
```

framps, 2023-01-24:
Ok. Now we know the filesystem created on /dev/sdb2 is not OK - forever unknown reasons.

>  I really hope we can save these raspbian installation and not have to start from scratch D:

Don't panic. All backups of raspiBackup can be restored manually with existing Linux tools (see [here](https://www.linux-tips-and-tricks.de/en/raspibackupcategorye/498-raspibackup-manual-restore) for a how to for a partition oriented rsync backup). 

As long as the backup is OK we will be able to get the backup restored :wink: There is one thing which puzzles me: Did you ever test the restore before and were you able to start the restored system? I strongly recommend to do this all over the place :wink: 

I now have to write down all steps how to restore a tgz backup manually and this will take some time. I will write theses instructions on my website and link to it when I'm done.

p1r473, 2023-01-24:
> Did you ever test the restore before and were you able to start the restored system?

I tested that raspiBackup worked a long time ago, when I first tested it, but I have not tested it in a while. Lots of things have changed on the system since then, but its not anything crazy. These devices are running Pihole, Apache Webservers, but Im not doing anything too crazy, I am just a regular prosumer/advanced user.

framps, 2023-01-24:
Great. So we know it should work. 

raspiBackup reminds you every 6 months to test a backup to make sure any updates on the system don't screw up the backup. But given the fact you didn't change much on the system we should be able to restore the backup.

framps, 2023-01-24:
Here we are :wink: [raspiBackup - Manual restore of a tgz backup ](https://www.linux-tips-and-tricks.de/en/raspibackupcategorye/646-raspibackup-manual-restore-of-a-tgz-backup)

If you face any issues just let me know.

Frankly I'm keen to know why raspiBackup restore fails for you ... but let's execute the manual restore first. Maybe we get some indications why raspiBackup fails during restore.

p1r473, 2023-01-24:
Further fleshing out the documentation, feel free to use if you feel its useful and appropriate.
`Next mount the backuppartition on a mountpoint. I use /mnt in the following description and hostname raspberry.`
```
sudo mnt /mnt/raspberry/raspberry-tgz-backup-20170812-134552/raspberry-backup.sfdisk /mnt/
```
If its a CIFS share:
```
sudo mount -v -t cifs //$HOST/$BACKUP_NAME$ /mnt/ -o credentials=/home/pi/.credentials
sudo mount -v -t cifs //$HOST/$BACKUP_NAME$ /mnt/ -o user=$USER$
sudo mount -v -t cifs //$HOST/$BACKUP_NAME$ /mnt/ -o user=$USER$,password="REDACTED"
```

I think a typo- `sde` should read `sdb`
[image]

My journey:
`mkdir /mnt/`
`sudo mount -v -t cifs //192.168.1.69/Harbormaster /mnt/ -o user=JJJ`
`sudo sfdisk /dev/sdb < /mnt/Harbormaster/Harbormaster-tgz-backup-20230107-030001/Harbormaster-backup.sfdisk`

```
Checking that no-one is using this disk right now ... OK

The device contains 'ext4' signature and it may remain on the device. It is recommended to wipe the device with wipefs(8) or sfdisk --wipe, in order to avoid possible collisions.

Disk /dev/sdb: 125 GiB, 134217728000 bytes, 262144000 sectors
Disk model: USB3.0 CRW   -SD
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x264c47cc

Old situation:

Device     Boot  Start       End   Sectors   Size Id Type
/dev/sdb1         2048    526335    524288   256M  c W95 FAT32 (LBA)
/dev/sdb2       526336 262141951 261615616 124.7G 83 Linux

>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Created a new DOS disklabel with disk identifier 0x264c47cc.
The device contains 'ext4' signature and it may remain on the device. It is recommended to wipe the device with wipefs(8) or sfdisk --wipe, in order to avoid possible collisions.

/dev/sdb1: Created a new partition 1 of type 'W95 FAT32 (LBA)' and of size 256 MiB.
Partition #1 contains a vfat signature.
/dev/sdb2: Created a new partition 2 of type 'Linux' and of size 118.8 GiB.
Partition #2 contains a ext4 signature.
/dev/sdb3: Done.

New situation:
Disklabel type: dos
Disk identifier: 0x264c47cc

Device     Boot  Start       End   Sectors   Size Id Type
/dev/sdb1         2048    526335    524288   256M  c W95 FAT32 (LBA)
/dev/sdb2       526336 249737215 249210880 118.8G 83 Linux

The partition table has been altered.
Calling ioctl() to re-read partition table.
Re-reading the partition table failed.: Device or resource busy
The kernel still uses the old table. The new table will be used at the next reboot or after you run partprobe(8) or partx(8).
```

That didnt work, let me wipe the disk.
`fdisk /dev/sdb`
```
[... 164 lines cut ...]

New situation:
Disklabel type: dos
Disk identifier: 0x264c47cc

Device     Boot  Start       End   Sectors   Size Id Type
/dev/sdb1         2048    526335    524288   256M  c W95 FAT32 (LBA)
/dev/sdb2       526336 249737215 249210880 118.8G 83 Linux

The partition table has been altered.
Calling ioctl() to re-read partition table.
Syncing disks.
```

Before I can mount to /mnt, I need to first unmount /mnt. This wasnt covered in the documentation @framps 
`umount /mnt`

`sudo mount /dev/sdb1 /mnt;`
```
mount: /mnt: wrong fs type, bad option, bad superblock on /dev/sdb1, missing codepage or helper program, or other error.
```

This error again!

Trying on a different card:

p1r473, 2023-01-24:
New card repeated all the steps up until
` sudo dd of=/dev/sdb if=Harbormaster-backup.img bs=1MB`
```
268+1 records in
268+1 records out
268435456 bytes (268 MB, 256 MiB) copied, 6.39675 s, 42.0 MB/s
```

That wiped the sd card, and was too fast.
`
Disk /dev/sdb: 238.3 GiB, 255869321216 bytes, 499744768 sectors
Disk model: USB3.0 CRW   -SD
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000
`

` ls -lsa`
```
total 5253929
      4 drwxr-xr-x 2 root root       4096 Jan 21 02:00 .
      4 drwxr-xr-x 2 root root       4096 Jan 20 03:49 ..
 262144 -rwxr-xr-x 1 root root  268435456 Jan  7 03:01 Harbormaster-backup.img
      1 -rwxr-xr-x 1 root root        512 Jan  7 03:01 Harbormaster-backup.mbr
      1 -rwxr-xr-x 1 root root        212 Jan  7 03:01 Harbormaster-backup.sfdisk
4991696 -rwxr-xr-x 1 root root 5111495595 Jan  7 03:22 Harbormaster-tgz-backup-20230107-030001.tgz
     76 -rwxr-xr-x 1 root root      75634 Jan  7 03:22 raspiBackup.log
      4 -rwxr-xr-x 1 root root       1360 Jan  7 03:22 raspiBackup.msg
```

The DD command took only 6 seconds and erased the partitions?

p1r473, 2023-01-24:
Its possible that the Sandisk 128gb may be at fault, as thats the only card that gives the fs error I think. I will proceed with only the brand new cards for further testing.

framps, 2023-01-24:
I'm sorry. There is a major typo: (I updated already my website). You're the guinea pig who verifies the instructions on my website are without any typos :wink: 

> sudo dd of=/dev/sdb if=/mnt/Harbormaster/Harbormaster-tgz-backup-20230107-030001/Harbormaster-backup.img  bs=1MiB

should read

> sudo dd of=/dev/sdb1 if=/mnt/Harbormaster/Harbormaster-tgz-backup-20230107-030001/Harbormaster-backup.img  bs=1MiB

Otherwise you erase the whole partition table and wipe out all partitions created before :cry:

p1r473, 2023-01-24:
No worries, happy to help. I thought that a 1 might have been missing there.

I would also add an `umount`
between `sudo dd of=/dev/sdb1 if=/mnt/raspberry/raspberry-tgz-backup-20170812-134552/raspberry-backup.img bs=1MiB`
and `sudo mount /dev/sdb1 /mnt; ls /mnt; sudo umount /mnt`

Because if /mnt has the remote backup, do we really want to mount /dev/sdb to it? This may confuse beginers.

For example:
`restore the tgz Backup on the partition created just now`
`sudo mount /dev/sdb2 /mnt; df -h /mnt`
`sudo tar --exclude /boot --same-owner --same-permissions --numeric-owner -x -v -z -f /mnt/raspberry/raspberry-tgz-backup-20170812-13455/raspberry-tgz-backup-20170812-134552.tgz -C /mnt`

Can you really have many things mounted directly to /mnt? Dont they need to be sub-mounts of /mnt? For example, /mnt/remotebackup and /mnt/sdb2mount
`sudo tar --exclude /boot --same-owner --same-permissions --numeric-owner -x -v -z -f /mntremotebackup/Harbormaster/Harbormaster-tgz-backup-20230107-030001/Harbormaster-tgz-backup-20230107-030001.tgz -C /mnt/sdb2backup`

Also, why do we need to specify bs=1MiB? Is default block size not best for a generic walkthrough like your guide? Isnt speed dependant on file size? It is faster to write though it seems.

Assuming that the one sd card (Sandisk 128 not-pro) that was giving the weird  `wrong fs type, bad option, bad superblock on /dev/sdb2, missing codepage or helper program, or other error.` error was a bad SD card...

it looks like the manual restore worked great. My Pi has been restored and it back up and running. I would however like to keep figuring out and helping you test what is wrong with the script. Why did the manual restore work but the script fails?

Ive checked the syslog, and it seems nothing bad in there like I was getting before (i/o errors writing)
[image]

p1r473, 2023-01-24:
Assuming that the one sd card (Sandisk 128 not-pro) that was giving the weird  `wrong fs type, bad option, bad superblock on /dev/sdb2, missing codepage or helper program, or other error.` error was a bad SD card...

it looks like the manual restore worked great. My Pi that was down has been restored and it back up and running. I would however like to keep figuring out and helping you test what is wrong with the script. Why did the manual restore work but the script fails?

Ive checked the syslog, and it seems nothing bad in there like I was getting before (i/o errors writing). Looks like a successful restore!
[image]

p1r473, 2023-01-24:
Even though restore seems successful, I would however like to keep figuring out and helping you test what is wrong with the script. Why did the manual restore work but the script fails?

framps, 2023-01-25:
> Because if /mnt has the remote backup, do we really want to mount /dev/sdb to it? This may confuse beginers.
> Can you really have many things mounted directly to /mnt?

Good catch. I updated the page and use /mnt  for the backup partition and /media for the target SD card partitions. 

Yes, all mounts are stacked. But you can access only the last mounted partition and there is no way to access the previously mounted partition in parallel. That's why I use /media now for the target device partitions.

> This may confuse beginers.

Actually these instructions are a fallback if the restore fails for some reasons and you have to have at least some basic experiences with Linux. I don't own and also don't have time to test a restore with all HW. Maybe your HW has some special features which cause the restore to fail?

> Also, why do we need to specify bs=1MiB? Is default block size not best for a generic walkthrough like your guide? Isnt speed dependant on file size? It is faster to write though it seems.

Well, I don't think speed is an issue here. The boot partition is quite small. I'm used to use 1MiB whenever I copy a huge RaspbianOS Image from my desktop to a SD card. And there this speeds up the copy process. I removed `bs=2MiB` .

> Assuming that the one sd card (Sandisk 128 not-pro) that was giving the weird `wrong fs type, bad option, bad superblock on /dev/sdb2, missing codepage or helper program, or other error.` error was a bad SD card...

:+1:

> it looks like the manual restore worked great on my new SD card. My Pi that was down has been restored and it back up and running.
> 
> Ive checked the syslog, and it seems nothing bad in there like I was getting before (i/o errors writing). Looks like a successful restore!

Glad we managed to get your Raspberry up and running again :+1: 

> Even though restore seems successful, I would like to keep figuring out and helping you test what is wrong with the script. Why did the manual restore work but the script fails?

:+1: Same to me. I never faced this issue and want to know what's the root cause and fix it. Glad you want to help on this issue. Without your help I will not be able to locate and fix the root cause.

As far as I can see there are following possible root causes:

1) The resizing fails given you use a 128GB SD card. I checked the debug log and it looks fine to me. But we have to test.
2) There is some kind of timing issue. Maybe it takes some time until the format was persisted on the SD card.
3) There is some special HW capability which has to be triggered in some way to persist the new generated filesystem
4) The Linux kernel does not get notified about the format of the second partition 

If you have any other idea please let me know.

I will now update raspiBackup and add additional debug statements. When I'm done I will let you know which steps to execute on your site :wink:

framps, 2023-01-25:
When I added the debug statements in master_604 I also went through your debug logs and suddenly detected the root cause for your issue: It's a bug in the restore code for tgz :worried: . Everything is OK on your site - other than the SD card which is broken indeed.

I created a fix and it's published already. Please update your local raspiBackup with `sudo raspiBackup -U -S` and try again. You now should be able to restore the backup aith raspiBackup. The manual restore was successfully because the syntax for tar was OK - which was not true in the old raspiBackup version because raspiBackup passed invalid options which caused RC2 by tar. Learning never stops :smile: 

Please close this issue when you successfully verified the fix.

Thank you very much for your patience and you offered to help to isolate the issue. Please let me know if I should return your donation :wink:

p1r473, 2023-01-25:
Just please give me a few days to verify, somehow the pi I was using yesterday wont boot anymore, so I need to restore that one from backup! Maybe one of the commands I did accidently messed it up.

Keep the donation and thanks for your help!

I will have this tested and verified this weekend I hope.

framps, 2023-01-25:
No hurry. I was able to reproduce your issue, created the fix and tested the fix. Your test is just the final test of the issue creator before he closes the issue.

framps, 2023-01-25:
General comment:

This bug affects all users who use backup type tgz. The bug was introduced in release 0.6.7 on 3.10.2022 unfortunately. If you have a 0.6.7 release with a newer commit date (check with `raspiBackup.sh --version`) or if you use release 0.6.8 you have to update raspiBackup with `sudo raspiBackup.sh -U -S` to get the fix.

p1r473, 2023-01-28:
Looks good!
[image]

closed 2023-01-28 by p1r473

framps, 2023-01-28:
Perfekt. Thank you very much for your verification :+1: 

Sorry for the inconvenience :slightly_frowning_face: