framps/raspiBackup #609: RBK0178E: Creation of PATH/nginx-backup.img failed with RC 1

Chr1sCon, 2023-01-29:
I am trying to run a backup of a raspberry pi nginx server.

I am backing up to a smb share on my nas.

The backup fails with error RBK0178E

This is a snippet of the log:
```
--- RBK0081I: Creating backup of type tar in /home/chris/nas_backup/nginx/nginx/nginx-tar-backup-20230129-215544.
--- RBK0044I: Creating backup of boot partition in /home/chris/nas_backup/nginx/nginx/nginx-tar-backup-20230129-215544/nginx-backup.img.
??? RBK0178E: Creation of /home/chris/nas_backup/nginx/nginx/nginx-tar-backup-20230129-215544/nginx-backup.img failed with RC 1.
```

I have the same setup running on other raspberry pi's, and they run fine.

I haven't been able to figure out what RBK0178E or RC1  means.

Any tips to further debug this?

Chr1sCon, 2023-01-29:
[debug.log](https://github.com/framps/raspiBackup/files/10530898/debug.log)

adding debug log

framps, 2023-01-29:
Please check [this page](https://www.linux-tips-and-tricks.de/en/raspibackupmessagese#m0178)

Next check the debug log whether you find any additional messages from dd. Just search for RBK0178E and scroll back :wink:

framps, 2023-01-30:
I would either try to use option -B to create the backup and cross fingers it succeeds.

Otherwise I would setup a new SD card as long as you can access the config data.

Your SD card is buggy and will die sooner or later.

And yes, I will update the English and German message. All the other translated messages I have to keep ... Unfortunately

framps, 2023-01-31:
Thank you very much for your donation. 

But ... as far as I now understand you have a backup of your system created with raspiBackup handy. Why do you plan to set up the system from scratch and don't restore the backup?

Chr1sCon, 2023-02-01:
No worries :) 

I actually don't have a backup. It fails every time, so I am left with a system that is not backed up... Unless you have a way to force it to backup what it can...?

framps, 2023-02-01:
> Unless you have a way to force it to backup what it can...?

Try to copy the root partition (second partition) with `cp -a` (You have to use a Linux system. Windows is not able to access the second partition) Then you have at least all the configuration files handy and they will help you to setup your system from scratch much faster :wink:

closed 2023-02-17 by github-actions[bot]