framps/raspiBackup #615: raspiBackupWrapper.sh

MrR341-GER, 2023-02-01:
well... I have to go to school, so lets make this short:

the raspiBackupWrapper.sh no longer shrinks the images... also doesnt restart services properly...
this "problem" occured the first time a few days ago...

Tytouc, 2023-02-03:
Hi @MrR341-GER
This wrapper is NOT for shrinking images etc. it 'only' calls raspiBackup.sh as per your configuration and notify the backup status by sending a Discord message to the channel which has been cofigured.
For shrinking image, you should use the other wrapper: [raspiBackupWrapper.sh] (https://github.com/framps/raspiBackup/blob/master/helper/raspiBackupWrapper.sh) .

raspiDiscordWrapper.sh does not add any command line argument while calling raspiBackup, so ensure your configuration is correct so that services are stopped as restarted properly. Maybe you can try calling the raspiBackup.sh script directly to check if services are stoped/restarted, and give us the status.
Cheers,

framps, 2023-02-03:
Please provide the debug log

closed 2023-02-18 by github-actions[bot]