framps/raspiBackup #617: raspibackup doesn't clean old backups

DarthWeber, 2023-02-10:
Hi Framp,

although i use 

`
20230210-011916 DBG 2555:      --- KEEPBACKUPS=10
20230210-011916 DBG 2556:      --- KEEPBACKUPS_DD=0
20230210-011916 DBG 2557:      --- KEEPBACKUPS_DDZ=3
20230210-011916 DBG 2558:      --- KEEPBACKUPS_TAR=0
20230210-011916 DBG 2559:      --- KEEPBACKUPS_TGZ=5
20230210-011916 DBG 2560:      --- KEEPBACKUPS_RSYNC=10
`
my raspibackup doesn't clean up old backups and keeps every backup for months. How can i force raspibackup to delete them ?

[raspiBackup.log](https://github.com/framps/raspiBackup/files/10703858/raspiBackup.log)

framps, 2023-02-10:
I checked the log and found

```
20230210-015209 DBG 4090:          --> cleanupBackupDirectory 
20230210-015209 DBG 4092:                  *** ls -la /mnt/sshfs/GymWild/@HOSTNAME@/@HOSTNAME@-tgz-backup-20230210-011912
20230210-015209 DBG 4092:                      ls: Zugriff auf '/mnt/sshfs/GymWild/@HOSTNAME@/@HOSTNAME@-tgz-backup-20230210-011912' nicht möglich: Datei oder Verzeichnis nicht gefunden
20230210-015209 DBG 4109:          <-- cleanupBackupDirectory 
20230210-015209 DBG 4565:      <-- cleanupBackup 
20230210-015209 DBG 5514:      --> applyBackupStrategy 
20230210-015209 DBG 5569:          --- Deleting oldest directory in /mnt/sshfs/GymWild/@HOSTNAME@
20230210-015209 DBG 5570:              *** ls -d /mnt/sshfs/GymWild/@HOSTNAME@/*
20230210-015209 DBG 5570:                  ls: Zugriff auf '/mnt/sshfs/GymWild/@HOSTNAME@/*' nicht möglich: Datei oder Verzeichnis nicht gefunden
20230210-015209 MSG 5572:          --- RBK0159I: 5 Backups werden für den Backuptyp tgz aufbewahrt. Bitte Geduld. 
20230210-015210 MSG 5575:          --- RBK0205I: Älteste Backup /mnt/sshfs/GymWild/@HOSTNAME@ in wird gelöscht. Das kann etwas dauern. Bitte Geduld. 
/usr/local/bin/raspiBackup.sh: Zeile 5576: pushd: /mnt/sshfs/GymWild/@HOSTNAME@: Datei oder Verzeichnis nicht gefunden
/usr/local/bin/raspiBackup.sh: Zeile 5576: popd: Der Verzeichnisstapel ist leer.
/usr/local/bin/raspiBackup.sh: Zeile 5581: pushd: /mnt/sshfs/GymWild/@HOSTNAME@: Datei oder Verzeichnis nicht gefunden
20230210-015210 DBG 5590:          --- Skipping pishrink.log
20230210-015210 DBG 5590:          --- Skipping raspiBackup.log
20230210-015210 DBG 5590:          --- Skipping raspiBackup.msg
/usr/local/bin/raspiBackup.sh: Zeile 5611: popd: Der Verzeichnisstapel ist leer.
```

For some reasons the backup partition is no longer available when raspiBackup wants to cleanup the backups. That's why there is no cleanup executed. 

Do you have any idea why the partition is gone?

I also noticed you use an old release of raspiBackup
```
20230210-011912 DBG 2004:  --- raspiBackup.sh 0.6.7, 2022-06-19/19:37:48 - 7e37673
```

Please update to latest release 0.6.8 with `sudo raspiBackup.sh -U` and try again. If the issue is still there please provide the new debug log for further analysis.

DarthWeber, 2023-02-17:
maybe due to ip change ? Maybe because i use the same method with same sshfs connection at same time - but this should not be a real problem.

I use the new version and different cron times now. Will report ....

DarthWeber, 2023-02-17:
Hm, jetzt stehe ich etwas auf dem Schlauch:
```
tar: /var/log/daemon.log: Datei hat sich beim Lesen geändert.
tar: /var/log/syslog: Datei hat sich beim Lesen geändert.
```
ich hatte das immer unsportlich mit TAR_IGNORE_ERRORS="1" ignorieren lassen, aber diese Option scheint es ja in der config version 0.1.7 nicht mehr zu geben. Wie gehe ich damit am besten um ? log Verzeichnis komplett excluden, da ich die Prozesse ja kaum fürs Backup beenden kann, oder doch ?

framps, 2023-02-17:
> aber diese Option scheint es ja in der config version 0.1.7 nicht mehr zu geben.

Wie ich [hier](executeTar) dokumentiert habe sollte man diese Option nicht nutzen sondern z.B. die Dateien vom Backup exluden. Deshalb ist die Option auch aus der Config verschwunden damit sie nicht mehr so offensichtlich ist. Du kannst sie aber weiterhin nutzen :wink: . Ich habe bislang keine jemans eingefuehrte Option entfernt obwohl es manchmal ganz gut ist alte Zoepfe abzuschneiden.

Einfach diese Option wieder in die Config eintragen und der Fehler wird wieder ignoriert. Allerdings wird diese Option bei jedem Upgrade wieder entfernt und muss dann immer wieder manuell eingefuegt werden :cry:

framps, 2023-02-17:
> maybe due to ip change ?

An IP change will cause the mounted backuppartition to become unavailable :cry: 

You have to modify your environment such that there is no IP change during the backup run. In order to have raspiBackup to cleanup the backups raspiBackup has to have access to the backup partition. 

> Maybe because i use the same method with same sshfs connection at same time

I don't know. Please check your environment setup. Maybe it helps to serialize the access to the backup partition via sshfs, i.e. make sure there are no backup runs executed in parallel.

framps, 2023-02-17:
> Allerdings wird diese Option bei jedem Upgrade wieder entfernt und muss dann immer wieder manuell eingefuegt werden 

Ausser Du nutzt eine der weiteren Configdateien (Siehe dazu [hier](https://www.linux-tips-and-tricks.de/de/aufruf-und-optionen)) mit Prio 3 oder 4. Die werden bei einem Upgrade nicht geaendert :wink:

p1r473, 2023-03-04:
Hi @framps 

Is there any way to trigger the smart backup recycle from a cron job without having to actually do a backup?

Ideally I would like to have a crime job that runs every week or month that just does smart recycle. I have one rig that's very difficult to back up and it's constantly failing to back up. Likely due to Wi-Fi networking issues not raspibackup. It's creating a ton of unfinished backups that it doesn't clean up because this pi in particular fails and reboots before the smart backup happens

framps, 2023-03-04:
> Is there any way to trigger the smart backup recycle from a cron job **without having to actually do a backup?**

It's not supported to trigger a recycle only without any backup just created. A recycle is coupled on a new backup. 

I see a workaround for you: As far as I understand you want to get rid of the incomplete backups. If raspiBackup terminates unsuccessfully the backup directory just created will be deleted. But this doesn't happen because your system unfortunately dies before.

There is an easy way to detect backup directories which are incomplete: raspiBackup creates the message and debug log in /tmp and in the latest step these files will be copied from /tmp into the backup directory if the backup finishes with RC0. If the backup fails it's copied into the home directory of the caller. You have to write a small script which traverses all backup directories and checks for the existence of raspiBackup.msg and raspiBackup.log. If they don't exist it's an incomplete backup and you can remove this directory.

closed 2023-03-14 by github-actions[bot]