framps/raspiBackup #619: Sender's email Address not changing in the email header

facundo59, 2023-02-20:
Hello there,

Thank you very much for this plug-in. I found an issue I thought I may share with you.

The -F backup runs smoothly, but I was getting an error message for the Email notification. I read about FAQ47, I still think the issues lays not with my configuration.

My email provider (Infomaniak) hard checks the sender email, it has to be registered. I successfully created an alias that is also used in SSMTP, so the sent emails are being translated to "raspberry@mydomain.de" (r@@@e). I added the corresponding configuration in the raspiBackup file (DEFAULT_SENDER_EMAIL), so it is forced to used this email.

I still get the error that the sender's email is invalid. I do think I have to check it with you, because otherwise SSMPT works very well without any extra parameters. 

When debugging SSMTP and checking the email's header, I see the following (domains redacted)
[raspibackup.log](https://github.com/framps/raspiBackup/files/10781580/raspibackup.log)

```
...
Feb 20 08:38:04 rpi sSMTP[9960]: MAIL FROM:<**root@mydomain.de**>
Feb 20 08:38:04 rpi sSMTP[9960]: 250 2.1.0 Ok
Feb 20 08:38:04 rpi sSMTP[9960]: RCPT TO:<me@mydomain.de>
Feb 20 08:38:04 rpi sSMTP[9960]: 250 2.1.5 Ok
Feb 20 08:38:04 rpi sSMTP[9960]: DATA
Feb 20 08:38:04 rpi sSMTP[9960]: 354 End data with <CR><LF>.<CR><LF>
Feb 20 08:38:04 rpi sSMTP[9960]: Received: by mydomain.de (sSMTP sendmail emulation); Mon, 20 Feb 2023 08:38:03 +0100
Feb 20 08:38:04 rpi sSMTP[9960]: Date: Mon, 20 Feb 2023 08:38:03 +0100
Feb 20 08:38:04 rpi sSMTP[9960]: To: me@mydomain.de
Feb 20 08:38:04 rpi sSMTP[9960]: From: **raspberry@mydomain.de**
Feb 20 08:38:04 rpi sSMTP[9960]: Subject: rpi: Backup finished successfully.
...
```

As you see, the email is actually sent from "root" in the header and masking it as "raspberry" in the body. 

Troubleshooting this issue I found the workarounds, it is no longer a problem, but as a good Samaritan, I think actually changing the address in the header is the intended functionality and may help others in the future.

I attach the logs so you can see the error message from the email server. 

Thank you for this awesome utility :)

framps, 2023-02-20:
> Thank you very much for this plug-in.

Which plugin?

> I found an issue I thought I may share with you.

Great :+1: Thank you very much.

> As you see, the email is actually sent from "root" in the header and masking it as "raspberry" in the body.

I'm a noob regarding eMail handling. 

> Troubleshooting this issue I found the workarounds,

Would be great if you share your workaround. This maybe enable me to create a fix :wink:

facundo59, 2023-02-21:
I meant app lol. Sorry, I lurke too much into smarthome "plug-ins" :)

There was the possibility of creating a alias for `root@mydomain.de` on the server side but I didn't like that. I went on and modified the `/etc/ssmtp/revaliases` file to force the conversion of the address that raspiBackup was trying to use into what I originally wanted. It looks something like this:

`root:raspberry@mydomain.de:mail.infomaniak.com:587`

I should have posted the solution in the original post!

framps, 2023-02-21:
Thank you very much for your explanation of your workaround :+1:

closed 2023-03-09 by github-actions[bot]