framps/raspiBackup #620: Out of memory crash

p1r473, 2023-02-28:
Hi Framps! You might remember me from #604
I have 3 Pis and they all have the exact same config for raspiBackup
However I noticed that one of my healthchecks has been down every day for the last day, on my primary Pi. This started ever since I ran a raspi -d restoration on this Pi I think last week.
Every time I run raspiBackup on this Pi, eventually I run out of memory and the Pi starts crashing and shuting down programs

I am watching it with `glances` and `watch top` and it will be running fine and smooth as butter then all of a sudden my Pi will just start ravaging and destroying itself. 

[image]

Syslog from right before to during the terrible crash:
https://pastebin.com/raw/h8XV3X8R
[raspiBackup.log](https://github.com/framps/raspiBackup/files/10847725/raspiBackup.log)

If you can help me fix this, I will be happy to donate, same as last time- this is my primary production Pi.

p1r473, 2023-02-28:
Worth noting: just this week I added the following line to my config:
`exclude=/var/www/apache/public/BrownNoise.mp3"`. This is a 1.3GB file. But it is excluded, and I am running a 8gb Pi. You can see in the screenshots I have nearly 6.6gb free.

I have REMOVED this line from the config, and just tried to raspiBackup again. The results:

framps, 2023-02-28:
That's bad. I unfortunately get 403 when I try to access your l logs. 

You're the first one who reports an OOM with raspiBackup. I'm sure it's one of the Linux tools used by raspiBackup which causes the OOM.

p1r473, 2023-02-28:
@framps sorry, I fixed the log access.

framps, 2023-02-28:
> I fixed the log access.

I just downloaded the files. You can now either delete them or protect them again. I just did a quick scan and for me there are sensitive details you most probably don't want to share with the public :wink:

framps, 2023-02-28:
I merged the two logs - syslog and raspiBackup log and that's what I see:

```
2:27:13 - raspiBackup starts (syslog)
2:27:24 - tar starts to backup the root partition (raspiBackup log)
2:37:37 - First OOM activities (syslog)
2:37:46 - tar terminates unsuccessfull with RC 42 (raspiBackup log)
2:37:49 - raspiBackup finished (raspiBackup log)
2:37:51 - Homeassistent and other services startup again (syslog)
```
As you can see it took 13 seconds from tar start to OOM start. It may be tar is the root cause ... but not sure. 

RC 42 from raspiBackup should not be returned normally. If tar terminates the return code is set to the tar return code. Looks like this did not happen because raspiBackup was killed my the OOM.

Question now is why the OOM started.

You have a system with a lot of active services, pihole, apache, chromium, nginx, ... As of now I never had somebody to backup a system with this load and you don't stop any running services. As you wrote there is enough free memory which suddenly vanishes.

There is something which may be the root cause for your issue: You use tar and zip the resulting image. You also wrote you now have a huge file for about 1.3GB and the issue popped up when you excluded the file. Maybe it popped up because you added this file?

Please execute following tests and let me know the outcome:
1) Temporarily remove this huge file
2) Don't zip the backup and just create a tar file
3) Create a dd backup on some spare disk

p1r473, 2023-02-28:
Regarding high load: I am running Apache, Pihole, and Homeassistant, yes, but I have 6.6gb free of RAM (see above screenshot). I also tried disabling HomeAssistant, rebooting, and then backing up again, to test the backup without HomeAssistant and same issue (OOM). HomeAssistant is a new addition to the system, however, PiHole and APache have been there for years and never had a raspibackup issue with them.

Test 1: Temporarily remove this huge file
`rm /var/www/apache/public/BrownNoise.mp3/`
`raspiBackup`
Results: PENDING

Test 2: Don't zip the backup and just create a tar file
Results: PENDING

Test 3: Create a dd backup on some spare disk
Results: PENDING

framps, 2023-02-28:
> Regarding high load: I am running Apache, Pihole, and Homeassistant, yes, but I have 6.6gb free of RAM (see above screenshot).

I see. It's nasty. A detailed analysis of the OOM log messages may help to locate the root - but I never analyzed a OOM log. I did some investigations on this ... but it's hard stuff ...

p1r473, 2023-02-28:
Something is wrong for sure, the system is barely under any load during raspiBackup
[image]
[image]

p1r473, 2023-02-28:
Test 1: Temporarily remove this huge file
`rm /var/www/apache/public/BrownNoise.mp3/`
`raspiBackup`
Results: SUCCESS

What? Success? I will try again to be safe. But, I excluded this file?? And it is only 1.3gb??

framps, 2023-02-28:
> What? Success? I will try again to be safe. But, I excluded this file?? And it is only 1.3gb??

Try again to confirm it's this file which causes the issue.  Why? As of now I have no idea ...

p1r473, 2023-02-28:
Just tried again and it crashed again :( May be was a fluke success

framps, 2023-02-28:
:-( You may now either skip to step 3 to test whether it's no an issue with raspiBackup/tar or continue with step 2.

And please note: You test right now at 4 PM but raspiBackup ran last night at 2 AM ...

p1r473, 2023-02-28:
Only thing that runs in the morning on the Pi is RaspiBackup every night at 3PM
Ill try step 3 and 4 now
Thanks!

framps, 2023-02-28:
> every night at 3PM

According the logs at 2AM - you may adjust your timzone :wink:

p1r473, 2023-02-28:
> 

Thanks, will check the timezone on my Pis
`cat /etc/cron.d/raspiBackup`
```
0 3 * * * root /usr/local/bin/raspiBackup
```

framps, 2023-02-28:
Extract from your syslog:

```
Feb 28 02:27:13 Harbormaster raspiBackup: Started 0.6.8 (138a72b)
```

p1r473, 2023-02-28:
> As you can see it took 13 seconds from tar start to OOM start. It may be tar is the root cause ... but not sure.

It runs smoothly for about 15-20 min and fails at the end
Im wondering if I messed up the logs last night and should send a new batch
Was up until 4AM last night and could easily have messed up the log gathering due to it being 4AM and trying multiple times with multiple crashed

p1r473, 2023-02-28:
Test 2: Don't zip the backup and just create a tar file
Results: CRASH

> As you can see it took 13 seconds from tar start to OOM start. It may be tar is the root cause ... but not sure.
It usually lasts for 15-20 min and fails near the end. Though it literally just failed within 1 min this last attempt I just tried

p1r473, 2023-02-28:
I will gather a new set of logs for
-tar
-tar without gzip
-dd

I will donate heavily for your help ;)

p1r473, 2023-02-28:
POST IN PROGRESS - COME BACK LATER :)

[New config](https://pastebin.com/raw/jh2Nmj5x)
Changes 
- Backup strategy temporarily set to 99 1 0 0
- Added some new exclusions

[Attempt 1 - tar with compression]
[raspilog](url)
[Syslog](url)

[Attempt 2 - tar withOUT compression]
[raspilog](url)
[Syslog](url)

[Attempt 3 - dd with compression]
[raspilog](url)
[Syslog](url)

[Attempt 4 - dd withOUT compression]
[raspilog](url)
[Syslog](url)

[Attempt 5 - rsync with compression]
[raspilog](url)
[Syslog](url)

[Attempt 6 - rsync  withOUT compression]
[raspilog](url)
[Syslog](url)

framps, 2023-03-01:
> It usually lasts for 15-20 min and fails near the end. Though it literally just failed within 1 min this last attempt I just tried.

Ok. According the first log you provided it took just seconds and that's why I thought it's not raspiBackup. Please make sure the syslog and raspiBackup log are in the same time slot and I can merge them :wink: 

> Worst case (I HOPE) is I just keep raspiBackup restoring until I find an old backup that doesnt crash

This may only work if there is not any external environment stuff the root cause for this issue. But let's be optimistic. Given the fact raspiBackup succeeds with dd it sounds like there is an issue with tar.

I'm busy today and will be able to check the logs at around 8 pm my time (CET).

If you find time you may also execute a plain tar backup. Search for RBK0085I in the debug log. The debug line which includes the `--exclude` statements is the command executed by raspiBackup. For example that's the rsync command executed in my environment:
```
--- RBK0085I: Backup of type rsync started. Please be patient.
20221204-202706 DBG 2489:                      --> executeRsync 
20221204-202706 DBG 2492:                          --- rsync --exclude="/backup/*"                      --exclude="/tmp/raspiBackup.log"                        --exclude="/tmp/raspiBackup.msg"                
        --exclude='.gvfs'                       --exclude=/proc/*                       --exclude=/lost+found/*                         --exclude=/sys/*                        --exclude=/dev/*                
        --exclude=/boot/*                       --exclude=/tmp/*                        --exclude=/run/*                        --exclude=/media/*                      --exclude=/@HOSTNAME@-backup.img --exclude=/@HOSTNAME@-backup.tmg --exclude=/@HOSTNAME@-backup.sfdisk --exclude=/@HOSTNAME@-backup.mbr --exclude=/raspiBackup.log --exclude=/raspiBackup.msg                     --exclude /var/log/journal             
                                --numeric-ids                   -aHAx --delete                                          --info=NAME0                    /                       "/backup/@HOSTNAME@/@HOSTNAME@-rsync-backup-20221204-202630"                    
```

p1r473, 2023-03-01:
Sure I will make the logs private around 8PM CET

p1r473, 2023-03-01:
I must be missing something:
RBK0085I:
`--- RBK0085I: Backup of type tgz started. Please be patient.
20230228-184213 DBG 2505:                  --> executeTar 
20230228-184213 DBG 2508:                      --- tar 		-cpi --one-file-system 		 		-z 		 		-f "/mnt/@HOSTNAME@/@HOSTNAME@/@HOSTNAME@-tgz-backup-20230228-184200/@HOSTNAME@-tgz-backup-20230228-184200.tgz" 		--warning=no-xdev 		--numeric-owner 		--exclude="/mnt/@HOSTNAME@/*" 		--exclude="/tmp/raspiBackup.log" 		--exclude="/tmp/raspiBackup.msg" 		--exclude='.gvfs' 		--exclude=/proc/* 		--exclude=/lost+found/* 		--exclude=/sys/* 		--exclude=/dev/* 		--exclude=/tmp/* 		--exclude=/boot/* 		--exclude=/run/* 		--exclude=/media/* 		--exclude=/mnt/* --exclude=/var/lib/samba/private/msg.sock/* --exclude=/etc/pihole/* --exclude=/var/log/* --exclude=/var/lib/php/sessions/* --exclude=/var/cache/* --exclude=/var/spool/postfix/dev/log --exclude=/home/@USER@/.local/share/Trash/*--exclude=/root/.local/share/Trash* --exclude=/.Trash/* --exclude=/var/www/apache/public/BrownNoise.mp3 		/`
`tar -cpi --one-file-system -z -f "/mnt/Harbormaster/Harbormaster/Harbormaster-tgz-backup-20230228-184200/Harbormaster-tgz-backup-20230228-184200.tgz" --warning=no-xdev --numeric-owner --exclude="/mnt/Harbormaster/*" --exclude="/tmp/raspiBackup.log" --exclude="/tmp/raspiBackup.msg" --exclude='.gvfs' --exclude=/proc/* --exclude=/lost+found/* --exclude=/sys/* --exclude=/dev/* --exclude=/tmp/* --exclude=/boot/* --exclude=/run/* --exclude=/media/* --exclude=/mnt/* --exclude=/var/lib/samba/private/msg.sock/* --exclude=/etc/pihole/* --exclude=/var/log/* --exclude=/var/lib/php/sessions/* --exclude=/var/cache/* --exclude=/var/spool/postfix/dev/log --exclude=/home/@USER@/.local/share/Trash/*--exclude=/root/.local/share/Trash* --exclude=/.Trash/* --exclude=/var/www/apache/public/BrownNoise.mp3 /`
```
tar: Removing leading `/' from member names
tar (child): /mnt/Harbormaster/Harbormaster/Harbormaster-tgz-backup-20230228-184200/Harbormaster-tgz-backup-20230228-184200.tgz: Cannot open: No such file or directory
tar (child): Error is not recoverable: exiting now
tar: /mnt/Harbormaster/Harbormaster/Harbormaster-tgz-backup-20230228-184200/Harbormaster-tgz-backup-20230228-184200.tgz: Cannot write: Broken pipe
tar: Child returned status 2
tar: Error is not recoverable: exiting now

```

p1r473, 2023-03-01:
Looks like the issue is with tar, the tar failed.

framps, 2023-03-01:
I just downloaded the logs. As you wrote even if you start tar alone you get an OOM :cry: tar is a very very old Linux backup tool and I'm surprised it causes the OOM. Maybe it's some change in your environment which causes tar to crash? But nevertheless tar shouldn't create an OOM.

I will check the logs now.

p1r473, 2023-03-01:
I did do lots of tinkering on the pi this week and I'm not sure what I've changed that caused this 😞
This is exactly why I love and use your tool so heavily. Cuz I can always just rollback the entire image.
Hopefully something the logs stands out and if not I will start using a second SD card and trying to roll back until I find an image that doesn't crash on tar

Usually Linux is so stable...

framps, 2023-03-01:
```
20230228-191150 DBG 2505:                  --> executeTar 
20230228-191150 DBG 2508:                      --- tar 		-cpi --one-file-
20230228-191819 DBG 4626:                      --> cleanup 
```
According the logs it takes 6:29 minutes for a tar backup to fail 
```
20230228-184213 DBG 2505:                  --> executeTar 
20230228-184213 DBG 2508:                      --- tar 		-cpi --one-file-system 		 		20230228-185359 DBG 4626:                      --> cleanup 
```
and 11:46 minutes for the tgz backup to fail. That's interesting because I expected the tar backup to fail later because zip will use more memory. 

Size of tar file when tar aborted was 
`20230228-191821 DBG 4391:                                          -rwxr-xr-x 1 root root 6617088000 Feb 28 19:11 @HOSTNAME@-tar-backup-20230228-191138.tar`

and size of tgz when tar aborted was 
`20230228-185402 DBG 4391:                                          -rwxr-xr-x 1 root root 3553509376 Feb 28 18:42 @HOSTNAME@-tgz-backup-20230228-184200.tgz`

Root file system size is 
`20230228-184205 DBG 5566:                          /dev/root                  117G   15G   97G  13% /`

I found some comments about tar and OOM (see [here](https://serverfault.com/questions/8577/why-does-tar-csf-file-tar-source-run-out-of-memory)). Not sure whether this is applicable for you. A split of tar backups will require a feature request for raspiBackup.  You may try to use these Options when you use the plain tar command to check whether this helps.

15G is nothing for tar and tar should be able to save this amount of data without any issues. 

I see following options for you to get rid of the issue:

1) Use rsync instead of tar 
I see you use cifs which causes you to use tar. Can't you use nfs which allows to use rsync? In addition rsync will create incremental backups in contrast to tar which creates a full backup all the time.

2) Execute some additional tests just with the tar command (without raspiBackup involvement) 
2a) Restore on a spare disk instead of the cifs partition. Maybe cifs runs out of memory because of some kind of buffering.
2b) Reduce the number of exclude Options. It's just a guess but you use a lot of exclude statements.

3) I noticed something in the logs. Maybe you know what's causing this:
3a) Check for swap usage
```
Mar  1 12:24:03 Harbormaster kernel: [ 1114.758132] 0 pages in swap cache
Mar  1 12:24:03 Harbormaster kernel: [ 1114.758135] Free swap  = 102396kB
Mar  1 12:24:03 Harbormaster kernel: [ 1114.758137] Total swap = 102396kB
Mar  1 12:24:03 Harbormaster kernel: [ 1114.758140] 1949696 pages RAM
```
No swap space is used. Did you disable swap usage?

3b) Disable RAM logging
```
20230228-184204 DBG 6201:              /dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
20230228-184204 DBG 6201:              /dev/mmcblk0p2 on /var/hdd.log type ext4 (rw,noatime)
20230228-184204 DBG 6201:              log2ram on /var/log type tmpfs (rw,nosuid,nodev,noexec,noatime,size=5120000k,mode=755)
2
```
You use RAM for logging. I use a plain RaspbianOS lite on my servers and they don't do any RAM caching. Maybe tar just makes the barrel to overflow?

3) I spend time to understand the OOM system messages and hopefully find something which help

4) Revert your system from existing backups until raspiBackup with tar will work again

framps, 2023-03-01:
> did do lots of tinkering on the pi this week and I'm not sure what I've changed that caused this

I detected in the logs you now use a lot of additional features of raspiBackup. Great you love them. But as we see that's not the root cause of your issue :cry: 

> What tool do you use to piece the logs together?

Just a plain editor and copy and paste :smile: . Given the fact a plain tar command also fails for you a merge of the two logs is no longer needed. I thought maybe raspiBackup causes in a later step (backup cleanup) a OOM.

p1r473, 2023-03-01:
The reason I am using CIFS (I think) is because my file share is on Windows. It looks like NFS is for Linux mounts, I think?

framps, 2023-03-01:
Yes. nfs is a Linux feature. I thought you maybe use some NAS and usually they run on Linux and support nfs. Maybe there is support on Windows for nfs but I don't know. I'm not familiar with windows.

p1r473, 2023-03-01:
Thanks. Im going to move to rsync in the future, but I will start rolling back through my backups until tar is fixed.

p1r473, 2023-03-01:
> I see you use cifs which causes you to use tar. Can't you use nfs which allows to use rsync? In addition rsync will create incremental backups in contrast to tar which creates a full backup all the time.

Based on this, I guess I cant use rsync with CIFS then?
https://www.linux-tips-and-tricks.de/images/decisiontree_en.dia.jpg

framps, 2023-03-01:
> I would much prefer this actually. Especially as you implemented my feature request here #366

It's not implemented. It's on my backlog list. As of now it's only you who asked for this feature. I don't start to work on any feature where where the quotient #number of users/#effort is nearly zero.

> Could you elaborate? Right now I am hosting my backups on my Windows PC server as it has a lot of hard drive space. Can you point me in the direction of the command?

It's just a guess. So it's up to you to decide whether you want to test this: Attach a spare disk, mount this disk and then create the backup on the disk instead of the cifs mounted partition.

> I actually just added 4 of those excludes yesterday. My other Pis have them too with no issues.

As I wrote: It's just a guess. I think you can ignore this.

> I just noticed this today too. I enabled the swap and still crashes [#620 (comment)](https://github.com/framps/raspiBackup/issues/620#issuecomment-1450543979)

Did you verify it's used in the OOM messages? Frankly I think it's the wrong approach to add additional swap space. If a system start to swap there is a memory issue. 

> I really like log2ram, it increases the life of my SD cards. Other 2 Pis work with it very well and this one used to too!

I see your point. Do you know you can configure an overlay filesystem with `raspi-config` ? Not sure whether this is an option for you.

> I am happy to compensate you for your time doing this! But this is up to you as I will start on 4 - reverting. Im going to take out this SD card and I can plug it back in for testing if you find anything.

In one of my FAQ I wrote I only support any issues related with raspiBackup. As we found out you have an issue which is not related  to raspiBackup but tar. OOM is an interesting topic and I will look into the OOM logs. But this will take time :cry: If it's OK with you to revert until you get a system which can be saved with raspiBackup again - that's my preference :wink: Crossing fingers you are successful. According the logs you have about 17 backups to go back ...

p1r473, 2023-03-01:
No problems at all. Thanks for your help!

closed 2023-03-01 by p1r473

framps, 2023-03-01:
> Is the issue with hard links? Based on these articles showing rsync to CIFS, perhaps it doesnt require ext3/4 but could work on NTFS?

rsync can be used on a CIFS mounted partition. But this will be a full backup all the time because of missing support for hardlinks. 
A user of raspiBackup had serious trouble when he restored his system which he saved on an NTFS formatted partition. Root cause was all file attributes were silently changed to root, including any user files :cry: Therefore I added a check in raspiBackup and now any rsync backup to an NTFS partition is rejected with an error message.

In former times of raspiBackup there was code which allowed to backup with rsync even the target filesystem doesn't support hardlinks and a full backup was created. It's no longer supported by raspiBackup. tar is a full backup and rsync is an incremental backup all the time created by raspiBackup.

framps, 2023-03-01:
Just let me know if you succeeded to revert :wink:

framps, 2023-03-01:
Would be interesting whether you can backup your system with rsync. If you have a spare disk handy just create a ext4 partition on this disk and create a rsync backup :wink:

framps, 2023-03-02:
Thank you very much for your donation but I also would have helped you on this issue anyhow. I try to fix any issue with raspiBackup if possible. Unfortunately it looks like there is some issue with tar :cry:

p1r473, 2023-03-02:
> Just let me know if you succeeded to revert 😉

Succeeded, had to go all the way back to 24th of last month.
Thank goodness for rapsiBackup!!!!

framps, 2023-03-02:
Do you remember what you changed/updated? Would be interesting to know what's the root cause of your issue and whether it's a tar issue or something else.

p1r473, 2023-03-02:
I did a lot of tinkering last 2 weeks 🤓 I played around with a lot of new things and I'm not sure which broke that

framps, 2023-03-02:
According the logs your Raspberry seems to be a production system which should run 7/24. 

Whenever I have major updates on one of my production Raspberries  I create a raspiBackup snapshot first just to have a backup with some meaningful comment for a longer period just in case I have to revert the update. Then I clone the system with raspiBackup (restore latest backup on another SD card) and update the cloned system running on a spare Raspberry. When all regression tests were successful  I shut down the production server,  insert the SD card with the system I updated and start the server again.

This reduces the downtime of the server to a minimum and protects to get any issues because of the update :wink: