framps/raspiBackup #621: Can't backup because of change in /var/log/lightdm

xMadNet, 2023-02-28:
Hi,
Tue Feb 28 14:08:40 pi@HomeBridge: ~ $ uname -a
Linux HomeBridge 6.1.11-v7l+ #1630 SMP Fri Feb 10 12:07:50 GMT 2023 armv7l GNU/Linux

Below log. 
Please help

BR

Tue Feb 28 14:02:37 pi@HomeBridge: ~ $ sudo raspiBackup.sh -m detailed
--- RBK0009I: HomeBridge: raspiBackup.sh V0.6.8 - 2023-02-09 (4b9c22a) started at Tue 28 Feb 2023 02:02:46 PM CET.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
--- RBK0031I: Checking whether a new version of raspiBackup.sh is available.
--- RBK0151I: Using backuppath /backup with partition type cifs.
--- RBK0008I: Stopping services: 'systemctl stop cron && systemctl stop cups && systemctl stop mosquitto && systemctl stop homebridge'.
--- RBK0081I: Creating backup of type tar in /backup/HomeBridge/HomeBridge-tar-backup-20230228-140244.
--- RBK0044I: Creating backup of boot partition in /backup/HomeBridge/HomeBridge-tar-backup-20230228-140244/HomeBridge-backup.img.
--- RBK0045I: Creating backup of partition layout in /backup/HomeBridge/HomeBridge-tar-backup-20230228-140244/HomeBridge-backup.sfdisk.
--- RBK0046I: Creating backup of master boot record in /backup/HomeBridge/HomeBridge-tar-backup-20230228-140244/HomeBridge-backup.mbr.
--- RBK0158I: Creating native tar backup "/backup/HomeBridge/HomeBridge-tar-backup-20230228-140244/HomeBridge-tar-backup-20230228-140244.tar".
--- RBK0085I: Backup of type tar started. Please be patient.
tar: /var/log/lightdm: file changed as we read it
tar: /var/log: file changed as we read it
??? RBK0021E: Backupprogram for type tar failed with RC 1.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0007I: Starting services: 'systemctl start homebridge && systemctl start mosquitto && systemctl start cups && systemctl start cron'.
--- RBK0043I: Removing incomplete backup in /backup/HomeBridge/HomeBridge-tar-backup-20230228-140244. This may take some time. Please be patient.
??? RBK0005E: Backup failed. Check previous error messages for details.
--- RBK0010I: HomeBridge: raspiBackup.sh V0.6.8 - 2023-02-09 (4b9c22a) stopped at Tue 28 Feb 2023 02:08:37 PM CET with rc 109.
--- RBK0167I: Sending email.
--- RBK0026I: Debug logfile saved in /home/pi/raspiBackup.log.

framps, 2023-02-28:
Please read [FAQ32](https://www.linux-tips-and-tricks.de/en/faq#a32)

xMadNet, 2023-02-28:
Hi, Thanks
1/ I can't stop lightdm !
2/ I do not have specific files to exclude. I'll have a look to DEFAULT_EXCLUDE_LIST so as to exclude /var/log

framps, 2023-02-28:
> I'll have a look to DEFAULT_EXCLUDE_LIST

Yes. Because the two files are just log files they shouldn't create any issue when they are excluded :wink:

p1r473, 2023-03-09:
`nano /usr/local/etc/raspiBackup.conf`

Dont back up any logs:
`DEFAULT_EXCLUDE_LIST="--exclude=/var/log/*"`

@xMadNet

closed 2023-03-26 by github-actions[bot]