framps/raspiBackup #626: Feature request - label/comment backups

p1r473, 2023-03-02:
Hi Framps!
I am starting to create a raspiBackup after every major system work. I would like a way to invoke raspiBackup and label or comment the backup
ex) 
`raspiBackup -comment "installed PiHole"
`raspiBackup -comment "removed Apache"
`raspiBackup -comment "installed HomeAssistant"
`raspiBackup -comment "configuredUnbound"

Somehow, this should be listable:
`raspiBackup -list`
```
20230223-180002-"installed PiHole"
20230224-210410-"removed Apache"
20230228-215801--"installed HomeAssistant"
20230227-030002-"configuredUnbound"
```

Thanks!

framps, 2023-03-02:
You can already add comments if you use [raspiBackup snapshots](https://www.linux-tips-and-tricks.de/en/snapshots) :wink: 

As of now you have to maintain them manually. I frankly don't want to add additional options to manage them. Then I have to add option `--remove <regular expression>`,  `--list <regular expression>` and actually duplicate a subset of the  `ls` and  `rm` commands.

closed 2023-03-02 by p1r473

p1r473, 2023-03-04:
Thanks so much for this, I missed this feature!