framps/raspiBackup #628: Use a temporary backup directory

framps, 2023-03-04:
As of now the new backup directory is created below the backupdirectory of the saved system. If the backup fails this directory is deleted.

In order to catch failed backups which leave an incomplete backupdirectory the backups should be created in a temp directory on the backupspace and if the backup succeeds this directory should be moved into the server backup directory. If the backup fails this temp directory is deleted. 

That way incomplete backups don't screw up the backup directories.

p1r473, 2023-03-05:
I have a question related to backup dirs. I didnt want to create an entire issue for it as it is related to backup directory but let me know if youd rather questions be in their whole own issue.
I have my backup directory set to /mnt/$HOSTNAME
But, the backup seems to go into /mnt/$HOSTNAME/$HOSTNAME
Is this working as intended?
What I ended up doing was making on my remote server /backups/hostname/hostname and then in /etc/fstab mounting \\REMOTESERVER\HOSTNAME as /backups/hostname
That way, the backup ends up going into /mnt/$HOSTNAME instead of /mnt/$HOSTNAME/$HOSTNAME

p1r473, 2023-03-05:
Thanks!
Im using /mnt/$HOSTNAME because I have 2 of my Pis mount both of each of their mountpoints, in case I need to use Pi1 to restore Pi2 or Pi2 to restore Pi1!

framps, 2023-03-09:
Merged into development branch

framps, 2023-06-29:
Given the fact next release 0.6.9 should be shipped this enhancement is delayed because the current implementation doesn't allow concurrent backups to happen from different systems.

closed 2024-10-05 by framps