framps/raspiBackup #629: Feature request - implement unfinished backup cleanup

p1r473, 2023-03-05:
> Is there any way to trigger the smart backup recycle from a cron job **without having to actually do a backup?**

It's not supported to trigger a recycle only without any backup just created. A recycle is coupled on a new backup. 

I see a workaround for you: As far as I understand you want to get rid of the incomplete backups. If raspiBackup terminates unsuccessfully the backup directory just created will be deleted. But this doesn't happen because your system unfortunately dies before.

There is an easy way to detect backup directories which are incomplete: raspiBackup creates the message and debug log in /tmp and in the latest step these files will be copied from /tmp into the backup directory if the backup finishes with RC0. If the backup fails it's copied into the home directory of the caller. You have to write a small script which traverses all backup directories and checks for the existence of raspiBackup.msg and raspiBackup.log. If they don't exist it's an incomplete backup and you can delete this directory.

_Originally posted by @framps in https://github.com/framps/raspiBackup/issues/617#issuecomment-1454844935_
            

I have written the script:
```
#!/bin/bash

backup_dir="/mnt/backup"

# check if backup directory exists
if [ ! -d "$backup_dir" ]; then
  echo "$backup_dir does not exist"
  exit 1
fi

# iterate through direct children subdirectories of backup_dir
for dir in "$backup_dir"/*/; do
  # check if raspiBackup.msg and raspiBackup.log files exist in the directory
  if [ ! -f "${dir}raspiBackup.msg" ] || [ ! -f "${dir}raspiBackup.log" ]; then
    # if not, delete the directory
    echo "Deleting directory: $dir"
    rm -rf "$dir"
  fi
done
```

Can we implement it into the main raspiBackup tool?

framps, 2023-03-05:
I agree there is an issue with incomplete backups. It's a race condition which unfortunately happens for you. Not seen any complaints about this before. That's why I recently created #628 to make sure a new backup directory is created only if the backup finished successfully.

In this issue I will create a warning message if raspiBackup detects incomplete backups. A raspiBackup user then can delete the incomplete backups in the command line. I don't want to add this removal in raspiBackup because this may increase the backup run time of raspiBackup significantly. There are users of raspiBackup out there who care about the backup time because they shut down important services and want to have them up as fast as possible.

For now - until I implemented #628 - you should kick off your code in a cron job. 

When #628 is implemented and included in next release raspiBackup will write a warning if some incomplete backups exist and warn in a message. This will be a new message number and documented accordingly.

p1r473, 2023-03-05:
> I don't want to add this removal in raspiBackup because this may increase the backup run time of raspiBackup significantly. 

Sure, but I added an `exit 0` to the script, meaning this cleanup tool should be run not as a part of a backup, but by itself.
So running `raspiBackup --cleanup` only does a cleanup, and does not create a backup so it would not increase backup time

> That's why I recently created https://github.com/framps/raspiBackup/issues/628 to make sure a new backup directory is created only if the backup finished successfully.
Thanks for creating this! This will fix the issue instead.

closed 2023-03-05 by p1r473

framps, 2023-03-05:
> So running `raspiBackup --cleanup` only does a cleanup, and does not create a backup so it would not increase backup time

You're right. But I don't want to add any management features to raspiBackup. If you invoke raspiBackup it's task is to create a backup or restore a backup. That's all.

What you're asking for is a management option in raspiBackup to manage the backups. This would be a nice feature but I don't spend any effort in backup management. It's easy to manage the backups on the command line.

framps, 2023-03-05:
> In the mean time, Ive added `systemctl stop watchdog` and `systemctl start watchdog` to the start and stops to try and disable this race condition.

Sounds good :+1:

p1r473, 2023-03-05:
> What you're asking for is a management option in raspiBackup to manage the backups. This would be a nice feature but I don't spend any effort in backup management. It's easy to manage the backups on the command line.

No worries,  this issue will be resolved with #628 so this feature is not needed anymore