framps/raspiBackup #631: Get Time Elapsed: user provided extension

p1r473, 2023-03-13:
Get Time Elapsed: user provided extension

This pull request adds a user extension that shows time elapsed for a backup
Backup was created and then when backup ended, I pasted the output into the issue here.

```
--- RBK0267I: Extension raspiBackup_time_post.sh called.
--- RBK2000I: Total runtime: 00:12:34 (hh:mm:ss)
```

framps, 2023-03-13:
Nice tiny extension :+1: 

You may have seen
```
--- RBK0078I: Backup time: 00:05:15.
```

Would be great if you document here which added value a raspiBackup user of your extension will get.

p1r473, 2023-03-13:
> RBK0078I: Backup time: 00:05:15.

I have not seen this. Does it require a certain debug status?
Or, maybe it only shows on successful backup? Recommend to show it on faiure too.

p1r473, 2023-03-13:
Ok I see this on successful backup
I missed it during testing as I was cancelling backups for faster testing

Recommend to add this on failed backup too. Oops! At least I learned stuff in creation of this extension.

closed 2023-03-13 by p1r473

framps, 2023-03-14:
> Does it require a certain debug status?

You will get this message only if raspiBackup terminates successfully.

> Recommend to add this on failed backup too.

I will take care of this :wink: 

> At least I learned stuff in creation of this extension.

:+1: So now you're ready now for another extension :smile: