framps/raspiBackup #633: Resume functionality

p1r473, 2023-03-14:
Hi Framps,
How does raspiBackup handle internet disconnections?
I have one Pi that is on Wifi and is constantly getting disconnected. When it resumes, I am unsure if raspiBackup is resuming as its just kind of hanging and I dont know if its still processing. 
According to the logs, it failed. and already cleaned up.
But I wouldnt know this if I didnt look at the logs
[raspiBackup.log](https://github.com/framps/raspiBackup/files/10963184/raspiBackup.log)
`--- RBK0043I: Removing incomplete backup`
Is there some way you can either resume the session upon internet reconnect, or resume the backup?

Because if you just look at the ssh session, it looks like its still running:
[image]

```
Mar 13 20:27:07 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
Mar 13 20:27:11 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
Mar 13 20:29:45 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
Mar 13 20:29:46 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
Mar 13 20:29:48 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
Mar 13 20:29:50 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
Mar 13 20:36:15 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
Mar 13 20:36:15 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
Mar 13 20:41:46 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
Mar 13 20:41:47 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
Mar 13 20:42:25 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
Mar 13 20:42:27 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
Mar 13 20:46:28 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
Mar 13 20:46:28 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
Mar 13 20:50:09 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
Mar 13 20:50:11 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
Mar 13 20:54:50 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
Mar 13 20:54:50 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
Mar 13 21:00:45 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
Mar 13 21:00:45 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
Mar 13 21:03:44 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
Mar 13 21:03:46 Tron dhcpcd[750]: wlan0: XXXXXXXXXXXXXXXXXXXXXXX is reachable again
```

Digging into the why, is its likely USB3 causing Wifi Disconnects:
https://www.pcmag.com/opinions/wireless-witch-the-truth-about-usb-30-and-wi-fi-interference
https://raspberrypi.stackexchange.com/a/120971

p1r473, 2023-03-14:
Edit: Im looking through the logs and it seems it was tarring while the internet was flickering on and off, so probably working ok
Will try to move to Ethernet...
```
--- RBK0085I: Backup of type tgz started. Please be patient.
20230313-195223 DBG 2505:                  --> executeTar 
20230313-195223 DBG 2508:                      --- tar 		-cpi --one-file-system 		 		-z 		 		-f "/mnt/@HOSTNAME@/@HOSTNAME@/@HOSTNAME@-tgz-backup-20230313-195030/@HOSTNAME@-tgz-backup-20230313-195030.tgz" 		--warning=no-xdev 		--numeric-owner 		--exclude="/mnt/@HOSTNAME@/*" 		--exclude="/tmp/raspiBackup.log" 		--exclude="/tmp/raspiBackup.msg" 		--exclude='.gvfs' 		--exclude=/proc/* 		--exclude=/lost+found/* 		--exclude=/sys/* 		--exclude=/dev/* 		--exclude=/tmp/* 		--exclude=/boot/* 		--exclude=/run/* 		--exclude=/media/* 		--exclude=/mnt/* --exclude=/var/lib/samba/private/msg.sock/* --exclude=/etc/pihole/* --exclude=/var/log/* --exclude=/var/lib/php/sessions/* --exclude=/var/cache/* --exclude=/var/spool/postfix/dev/log --exclude=/home/@USER@/.local/share/Trash/*--exclude=/root/.local/share/Trash* --exclude=/.Trash/* --exclude=/var/www/apache/public/BrownNoise.mp3 		/
20230313-211125 DBG 4626:                      --> cleanup 
20230313-211125 DBG 3941:                          --> colorAnnotation 1
20230313-211125 DBG 3965:                          <-- colorAnnotation 
??? RBK0163E: Script execution canceled with CTRL C.
20230313-211125 DBG 3941:                          --> colorAnnotation 1
20230313-211125 DBG 3965:                          <-- colorAnnotation 
--- RBK0033I: Please wait until cleanup has finished.
20230313-211125 DBG 4891:                          --> cleanupBackup 
20230313-211125 DBG 4893:                              --- Got trap SIGINT
20230313-211125 DBG 4894:                              --- rc: 103
20230313-211125 DBG 2195:                              --> callExtensions post
20230313-211125 DBG 2225:                                  --- Calling raspiBackup_mem_post.sh 103
20230313-211125 DBG 3941:                                  --> colorAnnotation 1
20230313-211125 DBG 3965:                                  <-- colorAnnotation 
```

closed 2023-03-14 by p1r473

framps, 2023-03-14:
> How does raspiBackup handle internet disconnections?

There is no special handling of any networking interruptions or uther unhandled failures. It's a nightmare to test this. Actually I know it's still an open issue in raspiBackup 0.6.8 because there is no kind of commit of the backups. That's among other things why I implemented #629. This makes sure there will be "commited" backups only in the backup directory. 

> Is there some way you can either resume the session upon internet reconnect, or resume the backup?

If any failure is caught raspiBackup deletes the incomplete backup. If there is an uncaught failure ( lost network connection, hard stop of system, ...) the backup will stay in the temp directory with #629 in next release and a manual cleanup has to be done. There is no way to resume a backup. 

Do you think it's worth to spend some time to add an additional option which will cause raspiBackup to delete the temp backup directory and thus there is no manual cleanup required? Keep in mind depending on the size of the backup data this may increase the backup time. I know there are users out there who want to have the backup time as short as possible because they shut down important services which should be up as fast as possible.

p1r473, 2023-03-14:
I know I'd use that automatic cleanup on my Wifi Pi

p1r473, 2023-03-14:
> Do you think it's worth to spend some time to add an additional option which will cause raspiBackup to delete the temp backup directory and thus there is no manual cleanup required? 

I know I'd use that automatic cleanup on my Wifi Pi!