framps/raspiBackup #635: Secure certificate expired, unable to install

g6nhu, 2023-03-14:
Hi, just trying to do an update on one of my Pis and I'm getting the following.   Has your certificate expired?

Thanks.

curl -L https://raspibackup.linux-tips-and-tricks.de/install | sudo bash

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (60) SSL certificate problem: certificate has expired
More details here: https://curl.haxx.se/docs/sslcerts.html

curl performs SSL certificate verification by default, using a "bundle"
 of Certificate Authority (CA) public keys (CA certs). If the default
 bundle file isn't adequate, you can specify an alternate file
 using the --cacert option.
If this HTTPS server uses a certificate signed by a CA represented in
 the bundle, the certificate verification probably failed due to a
 problem with the certificate (it might be expired, or the name might
 not match the domain name in the URL).
If you'd like to turn off curl's verification of the certificate, use
 the -k (or --insecure) option.

framps, 2023-03-14:
Looks like you use an old RaspbianOS which was not updated. 9/21 the [root certificates of Letsencrypt expired](https://letsencrypt.org/docs/dst-root-ca-x3-expiration-september-2021/). 

Try
```
sudo update-ca-certificates -f
```
to update your local root certificates and then try again :wink:

g6nhu, 2023-03-14:
Oh, very good spot.  Yes, I'm that Pi is using Stretch because the software it's running won't work on anything newer.

I've done that but it still doesn't update....

$ sudo update-ca-certificates -f
Clearing symlinks in /etc/ssl/certs...
done.
Updating certificates in /etc/ssl/certs...
126 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.

$ curl -L https://raspibackup.linux-tips-and-tricks.de/install | sudo bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (60) SSL certificate problem: certificate has expired
More details here: https://curl.haxx.se/docs/sslcerts.html

curl performs SSL certificate verification by default, using a "bundle"
 of Certificate Authority (CA) public keys (CA certs). If the default
 bundle file isn't adequate, you can specify an alternate file
 using the --cacert option.
If this HTTPS server uses a certificate signed by a CA represented in
 the bundle, the certificate verification probably failed due to a
 problem with the certificate (it might be expired, or the name might
 not match the domain name in the URL).
If you'd like to turn off curl's verification of the certificate, use
 the -k (or --insecure) option.

framps, 2023-03-14:
:cry: 

Please make sure you have the latest Stretch updates:

```
sudo apt-get update
sudo apt-get upgrade
```
and then try the above command again.

It's a very well know issue. If you search in the net for `letsencrypt root certificate expired` you will get a lot of hits. I just found [this page](https://serverfault.com/questions/1079199/client-on-debian-9-erroneously-reports-expired-certificate-for-letsencrypt-issue) which mentions Stretch. Please read the thread and try the suggested solutions. 

As you can see it's an issue with LetsEncrypt and not raspiBackup :wink:  . Crossing fingers you find a solution in the thread. Otherwise please try to find the solution in the net. In any case I suggest to upgrade to latest Bullseye.

Another alternative would be to download the latest installer and latest raspiBackup with a browser and copy the files to your Raspberry. Unfortunately then all the new options will not be added to the config file :sad: . 

So please try to update the root certificate and let me know which steps updated the root certificate and I can point another raspiBackup user who also faces this issue to this issue. You're the second one who faces this issue. The command above fixed the issue for the first one.

g6nhu, 2023-03-14:
Once again, I've demonstrated that I should practise what I preach.  Yes, that was it.  That Pi had been up for around 250 days which was when it was last updated.  I bounced it, updated, bounced again and it's all good.  Sorry for the dumb question, it's something I should have done without even thinking about it.

framps, 2023-03-14:
> Sorry for the dumb question,

No problem. Glad I was able to help. Please close the issue if you're fine now.

closed 2023-03-14 by g6nhu