framps/raspiBackup #636: Add tar diff backup as an extension to the normal tar backup

framps, 2023-03-18:
As discussed in https://github.com/framps/raspiBackup/issues/625 a tar inc backup will require too much effort to implement. But tar diff will be able to add to raspiBackup in a reasonable time.

framps, 2023-03-20:
`TAR_BACKUP_DIFFERENTIAL_PERIOD` will be a new option and define how many differential tar backups  are created until a new full backup is created. 

Typical use will be 4 for weekly backups such that every month a new full backup is created.

framps, 2023-03-22:
The restore of a tar diff backup is easier than an inc backup - but the handling of the full backup and diff backup creates similar effort as the inc backup in raspiBackup. Therefore this feature will also be moved to the backlog.

closed 2023-03-22 by framps

p1r473, 2023-03-28:
Thanks Framp for looking into this, happy to set a donation bounty on implementation of inc+diff as time permits!
If 30% of raspiBackup users are indeed using tar this will save a lot of time, diskspace, cpu and bandwidth.

framps, 2023-12-04:
By error I deleted my answer 

That's mostly what I answered:

Looks like you're keen to get this feature implemented :wink: 

raspiBackup was designed to have all backups independent from each other. This feature requires a major redesign of raspiBackup. Would be interesting to implement it. But given the huge effort to implement this feature and the small number of users it doesn't make sense.