framps/raspiBackup #640: gzip: stdout: Input/output error...tar: Child returned status 1

p1r473, 2023-03-23:
Hey Framps!
Had a backup go bad last night and wanted to show you the logs
```
gzip: stdout: Input/output error
tar: Child returned status 1
tar: Error is not recoverable: exiting now
```

Ever see this one before?
Ive tried googling it but it seems inconclusive.
[raspiBackup.log](https://github.com/framps/raspiBackup/files/11051871/raspiBackup.log)

framps, 2023-03-23:
It's tar which reports an io error when gzip was used to compress the tar file. Unfortunately tar doesn't tell you whether it's an input or output error. But it's usually an input error because the SD card dies. 

You also can temporarily use a different backup target to verify it's and input  error. Then it's time to restore your latest backup on a new SD card.

closed 2023-03-23 by p1r473

framps, 2023-03-23:
> Hey Framps!

My nick name is `framp` :wink: . Unfortunately I had to use `framps` for my github account because it's already registered by somebody else :cry:

p1r473, 2023-03-23:
> > Hey Framps!
> 
> My nick name is `framp` 😉 . Unfortunately I had to use `framps` for my github account because it's already registered by somebody else 😢

Could do fr4mp!