framps/raspiBackup #642: Dateisystem cifs auf /backup unterstüzt keine Linux Dateiattribute / File system cifs on /backup does not support Linux file attributes

Niriuqq, 2023-03-29:
Hallo 
Wenn ich das Toool starten will kommt die Meldung.
Dateisystem cifs auf /backup unterstüzt keine Linux Dateiattribute
Das Netzlaufwerk ist auch per CiFs gemounted. 
Was kann ich da machen?

Hello 
When I want to start the Toool I get the message.
File system cifs on /backup does not support Linux file attributes.
The network drive is also mounted via CiFs. 
What can I do there?

framps, 2023-03-29:
If you use a CIFS mounted backup partition you cannot use rsync. (See [here](https://www.linux-tips-and-tricks.de/en/raspibackupcategorye/580-which-filesystem-can-be-used-on-the-backup-partition))

Either you use a nfs mounted partition or you use tar instead of rsync. See the pros and cons of the different backup types [here](https://www.linux-tips-and-tricks.de/en/backup#butypes).

Niriuqq, 2023-03-29:
Mega!
Now it works.

closed 2023-03-29 by Niriuqq