framps/raspiBackup #647: Some directory are not backup

Blackkout, 2023-04-05:
Hello,

I have a little problem with my raspibackup script.
The first 5 directories of the root are not backed up.
Do you have an idea?

On my raspberry :
[image]

RaspiBackup result : 
[image]

Thank you in advance,

framps, 2023-04-05:
Please provide 

1) the debug log 
2) output of `ls -la /` on your Raspberry
3) output of `ls -la <backup>`

Blackkout, 2023-04-05:
1. the debug log
   
[raspiBackup.log](https://github.com/framps/raspiBackup/files/11161405/raspiBackup.log)

   2. output of ls -la / on your Raspberry
   
[image]

    3. output of ls -la <backup>
    
[image]

framps, 2023-04-05:
```
??? RBK0163E: Script execution canceled with CTRL C.
```
I need the log of a completed backup run.

framps, 2023-04-05:
```
20230405-115140 DBG 5563:                      *** blkid
20230405-115140 DBG 5563:                          /dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="4BBD-D3E7" TYPE="vfat" PARTUUID="738a4d67-01"
20230405-115140 DBG 5563:                          /dev/mmcblk0p2: LABEL="rootfs" UUID="45e99191-771b-4e12-a526-0779148892cb" TYPE="ext4" PARTUUID="738a4d67-02"
20230405-115140 DBG 5563:                          /dev/sda1: UUID="6b7a55e4-c11c-483f-891f-d099d79e7b35" TYPE="ext4" PARTUUID="738a4d67-01"
20230405-115140 DBG 5563:                          /dev/mmcblk0: PTUUID="738a4d67" PTTYPE="dos"
```
You have two partitions with the same PARTUUID: /dev/sda1 and /dev/mmcblk0p1.  This causes raspiBackup to select /dev/sda1 as the root partition whereas it should be /dev/mmcblk0p2. 

I have no clue what /dev/sda1 is used for.  But please change the PARTUUID of /dev/sda1 and try again.

Blackkout, 2023-04-06:
mmcblk0p1 is the boot partition on the sdcard.
sda1 is the root partition on the hdd drive.

Blackkout, 2023-04-06:
Hmmm for the same PARTUUID is strange my actual system is a restore from raspibackup.
The raspibackup puts the same PARTUUID ?

framps, 2023-04-07:
According the debug log you run a Raspberry 4. Why do you use mixed mode - boot from sd card and use the root partition on an USB disk? A Raspberry4 supports USB boot mode and you don't have to use an external boot partition on SD card.

Anyhow - I don't know what you have done. raspiBackup keeps the same UUID when a system is restored. If you don't want the restored system to use the same UUID use option `--updateUUIDs` to generate a different UUID.

Blackkout, 2023-04-12:
Yes it's a mistake from me to used a mixed mode like my rpi 2. I didn't succeed to swap to usb boot mode.

Can I change my sda1 PARTUUID without issue ? 

(Sorry to disturb you with my personal configuration problems)

framps, 2023-04-12:
> Can I change my sda1 PARTUUID without issue ?

raspiBackup will create a random PARTUUID if you use option `--updateUUIDs`  for restore and update /boot/commandline.txt and /etc/fstab accordingly so that your system will boot with the new PARTUUID.

It's just a unique identifier for Linux to identify the partition. It doesn't matter whether which PARTUUID is used. But they have to be unique and you must not have another partition with the same PARTUUID connected to your Raspberry.

> (Sorry to disturb you with my personal configuration problems)

You're welcome

Blackkout, 2023-04-26:
Hi @framps !

I changed nothing (don't update PARTUUID). And when I check my backup on my nas, I see strange result. Sometimes all folders of the root parrtition are backup. Sometimes 2 or 4, ....
The last 4 backups seem to be complete (one backup per week) : 
[image]
The 2nd of April : 
[image]

Blackkout, 2023-04-26:
Here raspiBackupInstallUI.log when seem to be complete : 
[raspiBackupInstallUI.log](https://github.com/framps/raspiBackup/files/11336629/raspiBackupInstallUI.log)
When it's failed :
[raspiBackupInstallUI_failed.log](https://github.com/framps/raspiBackup/files/11336620/raspiBackupInstallUI_failed.log)

framps, 2023-04-29:
Thank you very much for the installer logs. But they don't help to locate the root cause of the missing directories. Please provide the backup log.

Blackkout, 2023-05-15:
Hi @framps !

Hmm I have a new problem... my raspbian enter in read-only mode...
I think that my sd card is broken...
Can I launch raspibackup on this raspberry (read-only mode) in order to backup (backup store on my NAS) only the boot partition on a new sd card ?
Which command I need to use if it's possible ?

If it's compromised, I think I will operate a fresh install =/

framps, 2023-05-22:
You have to have a RW system.

Please open a new issue whenever you have a new issue and don't reuse an old issue :wink: