framps/raspiBackup #656: Invalid input: Backup path does not exist when using remote FTP mount

Shanghai-maglev, 2023-04-25:
Hi,

I have mounted a remote FTP server using curlftpfs. The remote FTP server is mounted onto a directory located in /mnt

The mount is working fine, no issues with transferring files, ls etc. The raspberry pi does not complain about the mounting.

When using the quick start CLI wizard, I get an error message when attempting to select my mount point as the backup path.

See picture attached.
[image]

Shanghai-maglev, 2023-04-25:
I have manually added the backup path in the .config file. I attempted to run the backup manually and the backup failed. I have attached the log file.

Note, I do not have any issues manually creating mkdir and writing to the /mnt/StarLAN-VM01 remote FTP directory.

[log.txt](https://github.com/framps/raspiBackup/files/11327522/log.txt)

Shanghai-maglev, 2023-04-25:
I fixed this issue with the following:

append "-o allow_other,uid=<insertyourusersid!>,uid=<insertyourusersgroupid!>" onto the curlftpsfs command when mounting the remote FTP location. This allows curlftps to mount the remote FTP location as the user, not root which fixed the permissions issue.

I have a new error message though so let's see how that one goes.

Shanghai-maglev, 2023-04-25:
holy sh*t it's actually working. thanks for your effort and time @framps ! awesome software.

closed 2023-04-25 by Shanghai-maglev

framps, 2023-04-26:
@Shanghai-maglev Congratulations you successfully managed to configure raspiBackup to use curlftpfs 👍 

In theory I knew it should work but I never tested it and you now proved it works.

The installer is primarily targeted for raspiBackup users who are not familiar with Linux and frankly is tested for standard environments only.  curlftpfs is not standard and thus requires a manual  update of raspiBackup config file which you managed perfectly 👍