framps/raspiBackup #661: Restore with "--resizeRootFS-" fails

bregtaeler, 2023-05-30:
Hi framps,

started to try your backup-tool.
Did a tar backup (std. boot + 16GB root partition) of an usb ssd.
Afterwords restorecheck with standard restore to 23GB MC, was successfull (boot from MC ok).
Re-done the restore with "--resizeRootFS-" to same MC worked also, but root fs was still bigger than orginal 16G, what I don't want. therefore I deleted the 2 partitions on MC with fdisk and re-done restore with "--resizeRootFS-"
but restore stops with failure "??? RBK0185E: Rootpartitionserstellung endet fehlerhaft mit RC 1."

pi@rpi-4-2:~ $ sudo raspiBackup.sh --resizeRootFS- -d /dev/mmcblk0 /backup/rpi-4-2/rpi-4-2-tar-backup-20230528-213432/
--- RBK0009I: rpi-4-2: raspiBackup.sh V0.6.8 - 2023-04-15 (f600458) Tue 30 May 22:16:47 CEST 2023 gestartet.
!!! RBK0065W: Gerät /dev/mmcblk0 wird repartitioniert und die gesamten Daten werden gelöscht.
--- RBK0146I: Keine Partitionstabelle auf /dev/mmcblk0 gefunden.
--- RBK0066I: Gerät /dev/mmcblk0 wird überschrieben mit der gesicherten Boot- und Rootpartition.
--- RBK0069I: Bootpartition /dev/mmcblk0p1 wird formatiert und erhält die zurückgespielte Bootpartition.
--- RBK0070I: Rootpartition /dev/mmcblk0p2 wird formatiert und erhält die zurückgespielte Rootpartition.
--- RBK0038I: Bist Du sicher? j/N
j
--- RBK0050I: Backup wird von /backup/rpi-4-2/rpi-4-2-tar-backup-20230528-213432 zurückgespielt.
--- RBK0053I: Erste Partition (Bootpartition) wird auf /dev/mmcblk0p1 zurückgespielt.
??? RBK0185E: Rootpartitionserstellung endet fehlerhaft mit RC 1.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
??? RBK0077E: Restore wurde fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0010I: rpi-4-2: raspiBackup.sh V0.6.8 - 2023-04-15 (f600458) Tue 30 May 22:17:05 CEST 2023 beendet mit Returncode 117.
--- RBK0026I: Debug Logdatei wurde in /home/pi/raspiBackup.logr gesichert.

tried it severlal times, but always the same problem.
Any ideas ?
can you pls help ?

thanks in advance & best regards,
Martin

[raspiBackup.zip](https://github.com/framps/raspiBackup/files/11605235/raspiBackup.zip)

framps, 2023-05-31:
In the debug log I find
```
20230530-221705 MSG 5793:                  --- RBK0054I: Zweite Partition (Rootpartition) /dev/mmcblk0p2 wird formatiert. 
mke2fs 1.46.2 (28-Feb-2021)
The file /dev/mmcblk0p2 does not exist and no size was specified.
```
I checked the code and detected there is an issue when --resizeRootFS option is turned off. I have to fix this.

If it's important to restore the image right now you can use option `-1` (dash one) as a workaround. Otherwise please wait until I created a fix for you and you can verify the fix.

framps, 2023-05-31:
I just created a fix for you :wink: 

Please download the fix with `curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/raspiBackupDownloadFromGit.sh | bash -s -- master_661 raspiBackup.sh` and invoke the fix with `./raspiBackup.sh ...` (Note the leading period!) to verify the fix.

framps, 2023-06-01:
Thank you very much for your verification. I will merge the fix in master today.

closed 2023-06-01 by framps