framps/raspiBackup #671: Backup breaks up with Returncode 101

BMWfan, 2023-08-22:
Hello,

i made today a fresh install of raspiBackup.

Whenever i start a backup manually by typing 'sudo raspiBackup.sh' it breaks up after some minutes with this information:
`--- RBK0009I: raspberrypi: raspiBackup.sh V0.6.8 - 2023-08-18 (cbfe5e8) Di 22. Aug 10:48:49 CEST 2023 gestartet.
--- RBK0151I: Backuppfad /mnt/ftpshare/backups mit Partitionstyp fuse wird benutzt.
--- RBK0044I: Backup der Bootpartition wird in /mnt/ftpshare/backups/raspberrypi/raspberrypi-tgz-backup-20230822-104847/raspberrypi-backup.img erstellt.
??? RBK0001E: Unerwarteter Programmfehler trat auf. (cbfe5e8), Zeile: 5084, Fehler: Mount of boot partition backup file failed with rc 1.
??? RBK0148E: @@@@@@@@@@@@@@@@@@@@ Stacktrace @@@@@@@@@@@@@@@@@@@@
  File "/usr/local/bin/raspiBackup.sh", line 9479, in main
    doit # no return
  File "/usr/local/bin/raspiBackup.sh", line 6237, in doit
                doitBackup
  File "/usr/local/bin/raspiBackup.sh", line 7049, in doitBackup
                backup
  File "/usr/local/bin/raspiBackup.sh", line 6071, in backup
                                        $BACKUPTYPE_TAR|$BACKUPTYPE_TGZ) backupTar
  File "/usr/local/bin/raspiBackup.sh", line 5322, in backupTar
                bootPartitionBackup
  File "/usr/local/bin/raspiBackup.sh", line 5084, in bootPartitionBackup
                                                assertionFailed $LINENO "Mount of boot partition backup file failed with rc $rc"
  File "/usr/local/bin/raspiBackup.sh", line 2435, in assertionFailed
        logStack
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0043I: Unvollständiges Backup in /mnt/ftpshare/backups/raspberrypi/raspberrypi-tgz-backup-20230822-104847 wird gelöscht. Das kann etwas dauern. Bitte Geduld.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0010I: raspberrypi: raspiBackup.sh V0.6.8 - 2023-08-18 (cbfe5e8) Di 22. Aug 10:49:54 CEST 2023 beendet mit Returncode 101.
--- RBK0026I: Debug Logdatei wurde in /root/raspiBackup.log gesichert.`

What could be here the issue?

Greetings

Dany

framps, 2023-08-22:
Could you please provide the debug log `/root/raspiBackup.log` for further analysis?

BMWfan, 2023-08-22:
> Could you please provide the debug log `/root/raspiBackup.log` for further analysis?

Clear, hope its enough `20230822-150301 DBG 2075:  --- Logfiles used: /tmp/raspiBackup.log and /tmp/raspiBackup.msg
20230822-150301 DBG 2086:  --- raspiBackup.sh 0.6.8, 2023-08-18/19:26:33 - cbfe5e8
20230822-150301 DBG 2089:  --- +================================================================================================================================================+
20230822-150301 DBG 2090:  --- | ===>  Viele sensitive Informationen werden in dieser Logdatei maskiert. Vor dem Verteilen des Logs sollte es trotzdem ueberprueft werden  <=== |
20230822-150301 DBG 2091:  --- +================================================================================================================================================+
20230822-150301 DBG 1917:  --> trapWithArg cleanupStartup SIGINT SIGTERM EXIT
20230822-150301 DBG 1922:  <-- trapWithArg
20230822-150301 DBG 3708:  --> readConfigParameters
20230822-150301 DBG 3723:      --- Read config /usr/local/etc/raspiBackup.conf : 0.1.7
20230822-150301 DBG 3723:      --- VERSION_CONFIG="0.1.7"
20230822-150301 DBG 3723:      --- DEFAULT_BACKUPPATH="/mnt/ftpshare/backups"
20230822-150301 DBG 3723:      --- DEFAULT_BACKUPTYPE="tar"
20230822-150301 DBG 3723:      --- DEFAULT_ZIP_BACKUP="1"
20230822-150301 DBG 3723:      --- DEFAULT_DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0
20230822-150301 DBG 3723:      --- DEFAULT_KEEPBACKUPS="30"
20230822-150301 DBG 3723:      --- DEFAULT_KEEPBACKUPS_DD=0
20230822-150301 DBG 3723:      --- DEFAULT_KEEPBACKUPS_DDZ=0
20230822-150301 DBG 3723:      --- DEFAULT_KEEPBACKUPS_TAR=0
20230822-150301 DBG 3723:      --- DEFAULT_KEEPBACKUPS_TGZ=0
20230822-150301 DBG 3723:      --- DEFAULT_KEEPBACKUPS_RSYNC=0
20230822-150301 DBG 3723:      --- DEFAULT_SMART_RECYCLE="1"
20230822-150301 DBG 3723:      --- DEFAULT_SMART_RECYCLE_DRYRUN="0"
20230822-150301 DBG 3723:      --- DEFAULT_SMART_RECYCLE_OPTIONS="7 4 12 1"
20230822-150301 DBG 3723:      --- DEFAULT_STOPSERVICES="systemctl stop cron && systemctl stop containerd && systemctl stop docker"
20230822-150301 DBG 3723:      --- DEFAULT_STARTSERVICES="systemctl start docker && systemctl start containerd && systemctl start cron"
20230822-150301 DBG 3723:      --- DEFAULT_BEFORE_STOPSERVICES=""
20230822-150301 DBG 3723:      --- DEFAULT_AFTER_STARTSERVICES=""
20230822-150301 DBG 3723:      --- DEFAULT_EMAIL=""
20230822-150301 DBG 3723:      --- DEFAULT_SENDER_EMAIL=""
20230822-150301 DBG 3723:      --- DEFAULT_EMAIL_PARMS=""
20230822-150301 DBG 3723:      --- DEFAULT_MAIL_PROGRAM="mail"
20230822-150301 DBG 3723:      --- DEFAULT_TELEGRAM_TOKEN=""
20230822-150301 DBG 3723:      --- DEFAULT_TELEGRAM_CHATID=""
20230822-150301 DBG 3723:      --- DEFAULT_TELEGRAM_NOTIFICATIONS="F"
20230822-150301 DBG 3723:      --- DEFAULT_PUSHOVER_TOKEN=""
20230822-150301 DBG 3723:      --- DEFAULT_PUSHOVER_USER=""
20230822-150301 DBG 3723:      --- DEFAULT_PUSHOVER_NOTIFICATIONS="F"
20230822-150301 DBG 3723:      --- DEFAULT_PUSHOVER_SOUND_SUCCESS=""
20230822-150301 DBG 3723:      --- DEFAULT_PUSHOVER_SOUND_FAILURE=""
20230822-150301 DBG 3723:      --- DEFAULT_PUSHOVER_PRIORITY_SUCCESS="0"
20230822-150301 DBG 3723:      --- DEFAULT_PUSHOVER_PRIORITY_FAILURE="1"
20230822-150301 DBG 3723:      --- DEFAULT_SLACK_WEBHOOK_URL=""
20230822-150301 DBG 3723:      --- DEFAULT_SLACK_NOTIFICATIONS=""
20230822-150301 DBG 3723:      --- DEFAULT_NOTIFY_START=0
20230822-150301 DBG 3723:      --- DEFAULT_MAIL_ON_ERROR_ONLY=0
20230822-150301 DBG 3723:      --- DEFAULT_MSG_LEVEL="0"
20230822-150301 DBG 3723:      --- DEFAULT_COLORING="CM"
20230822-150301 DBG 3723:      --- DEFAULT_EMAIL_COLORING="SUBJECT"
20230822-150301 DBG 3723:      --- DEFAULT_COLOR_CODES=("#FF8000 33" "#FF0000 31")
20230822-150301 DBG 3723:      --- DEFAULT_RESTORE_DEVICE=""
20230822-150301 DBG 3723:      --- DEFAULT_VERBOSE=0
20230822-150301 DBG 3723:      --- DEFAULT_SKIPLOCALCHECK=0
20230822-150301 DBG 3723:      --- DEFAULT_DD_BLOCKSIZE=1M
20230822-150301 DBG 3723:      --- DEFAULT_DD_PARMS=""
20230822-150301 DBG 3723:      --- DEFAULT_EXCLUDE_LIST=""
20230822-150301 DBG 3723:      --- DEFAULT_NOTIFY_UPDATE=1
20230822-150301 DBG 3723:      --- DEFAULT_EXTENSIONS=""
20230822-150301 DBG 3723:      --- DEFAULT_RESTORE_EXTENSIONS=""
20230822-150301 DBG 3723:      --- DEFAULT_LANGUAGE="DE"
20230822-150301 DBG 3723:      --- DEFAULT_DEPLOYMENT_HOSTS=""
20230822-150301 DBG 3723:      --- DEFAULT_YES_NO_RESTORE_DEVICE=""
20230822-150301 DBG 3723:      --- DEFAULT_RESIZE_ROOTFS=1
20230822-150301 DBG 3723:      --- DEFAULT_LINK_BOOTPARTITIONFILES=0
20230822-150301 DBG 3723:      --- DEFAULT_TAR_BOOT_PARTITION_ENABLED=0
20230822-150301 DBG 3723:      --- DEFAULT_SKIP_DEPRECATED=""
20230822-150301 DBG 3723:      --- DEFAULT_CHECK_FOR_BAD_BLOCKS=0
20230822-150301 DBG 3723:      --- DEFAULT_TIMESTAMPS=0
20230822-150301 DBG 3723:      --- DEFAULT_SYSTEMSTATUS=0
[... 751 lines cut ...]
20230822-150412 DBG 3970:                          <-- colorAnnotation
--- RBK0010I: @HOSTNAME@: raspiBackup.sh V0.6.8 - 2023-08-18 (cbfe5e8) Di 22. Aug 15:04:11 CEST 2023 beendet mit Returncode 101.
20230822-150412 DBG 4266:                          --> sendEMail
20230822-150412 DBG 4388:                          <-- sendEMail
20230822-150412 DBG 4487:                          --- Masquerading some mount options
20230822-150412 DBG 4517:                          --- Masquerading home directory name
[... cut ...]
20230822-150413 DBG 2162:                                  *** cat /tmp/raspiBackup.logf
20230822-150413 DBG 2162:
20230822-150413 DBG 3946:                              --> colorAnnotation 1
20230822-150413 DBG 3970:                              <-- colorAnnotation
20230822-150413 DBG 2177:                          <-- logFinish
20230822-150413 DBG 4624:                          --> unLockMe
20230822-150413 DBG 4626:                          <-- unLockMe
20230822-150413 DBG 4759:                      <-- cleanup
`

framps, 2023-08-22:
Thank you very much. Actually you can dragndrop the debug log into github as a file and don't have to copy the contents. That way line breaks will not valish and the debug log is much easier to read :wink: 

But I found the important message which explains the root cause of the error:

```failed to set up loop device: Die Operation wird nicht unterstützt losetup: /dev/loop1: detach failed: Kein passendes Gerät bzw. keine passende Adresse gefunden ```

For some reasons you don't have a loop device even you use bullseye. I updated the code and now raspiBackup skips the boot img check if the loop device setup fails.

Please call `sudo raspiBackup -U -S` to download the updated code. Then start raspiBackup again :wink:

BMWfan, 2023-08-22:
thanks for you explanation and help, but seems to be it is nevertheless not working:
`--- RBK0009I: raspberrypi: raspiBackup.sh V0.6.9-beta - 2023-08-20 (dd9e478) Di 22. Aug 16:25:09 CEST 2023 gestartet.
!!! RBK0165W: =========> HINWEIS <=========
!!! RBK0165W: Dieses ist eine Betaversion welche nicht in Produktion benutzt werden sollte.
!!! RBK0165W: =========> HINWEIS <=========
--- RBK0151I: Backuppfad /mnt/ftpshare/backups mit Partitionstyp fuse wird benutzt.
Warning: Stopping docker.service, but it can still be activated by:
  docker.socket
--- RBK0044I: Backup der Bootpartition wird in /mnt/ftpshare/backups/raspberrypi/raspberrypi-tgz-backup-20230822-162508/raspberrypi-backup.img erstellt.
??? RBK0001E: Unerwarteter Programmfehler trat auf. (dd9e478), Zeile: 5237, Fehler: Mount of boot partition backup file failed with rc 1.
??? RBK0148E: @@@@@@@@@@@@@@@@@@@@ Stacktrace @@@@@@@@@@@@@@@@@@@@
  File "/usr/local/bin/raspiBackup.sh", line 9652, in main
    doit # no return
  File "/usr/local/bin/raspiBackup.sh", line 6427, in doit
                doitBackup
  File "/usr/local/bin/raspiBackup.sh", line 7246, in doitBackup
                backup
  File "/usr/local/bin/raspiBackup.sh", line 6261, in backup
                                        $BACKUPTYPE_TAR|$BACKUPTYPE_TGZ) backupTar
  File "/usr/local/bin/raspiBackup.sh", line 5476, in backupTar
                bootPartitionBackup
  File "/usr/local/bin/raspiBackup.sh", line 5237, in bootPartitionBackup
                                                        assertionFailed $LINENO "Mount of boot partition backup file failed with rc $rc"
  File "/usr/local/bin/raspiBackup.sh", line 2521, in assertionFailed
        logStack
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0043I: Unvollständiges Backup in /mnt/ftpshare/backups/raspberrypi/raspberrypi-tgz-backup-20230822-162508 wird gelöscht. Das kann etwas dauern. Bitte Geduld.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0010I: raspberrypi: raspiBackup.sh V0.6.9-beta - 2023-08-20 (dd9e478) Di 22. Aug 16:26:51 CEST 2023 beendet mit Returncode 101.
--- RBK0026I: Debug Logdatei wurde in /root/raspiBackup.log gesichert.
`

framps, 2023-08-22:
Ouch :cry: You didn't grab the lastest master code but the beta code. My fault . And this code doesn't include the fix yet.

Please use ` curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/raspiBackupDownloadFromGit.sh | bash -s -- master` to download the fix. Please note you have to call raspiBackup with `sudo ./raspiBackup.sh` (note the leading period).

Later on if raspiBackup works for you you have to execute `sudo mv ./raspiBackup.sh /usr/local/bin` to use the fixed version from now on.

BMWfan, 2023-08-24:
@framps sorry for the late reply. Had an issue on my raspi what i had to fix first.
The backup seems to be working. Thanks. 

Can you add docker.socket as an enhancement to the list of "Zu stoppende und startende Services"or include it to the docker service itself? My containers are still up whenever the backup runs even tough i select docker and containerd as a to stopping service.

framps, 2023-08-24:
Great it works now :+1: 

I don't use docker. Do I get you right you suggest to have the installer to suggest to stop `docker.socket` in addition to `docker` and `containerd` ?

framps, 2023-08-28:
```
DEFAULT_STOPSERVICES="systemctl stop cron && systemctl stop containerd && systemctl stop docker && systemctl stop docker.socket"
```
That's what you have now in your config?

framps, 2023-08-29:
I have docker now up and running on RaspbianOS (Bullseye) and when I stop `containerd` docker is stopped and I also don't see any `docker.socket` :thinking: 
```
pi@raspberrypi-bullseye-desktop-docker:~ $ ps -ef | egrep "docker|container"
avahi      503     1  0 18:37 ?        00:00:00 avahi-daemon: running [raspberrypi-bullseye-desktop-docker.local]
root       647     1  0 18:37 ?        00:00:00 /usr/bin/containerd
root       766     1  0 18:37 ?        00:00:02 /usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.sock
root      1334   766  0 18:37 ?        00:00:00 /usr/bin/docker-proxy -proto tcp -host-ip 0.0.0.0 -host-port 9000 -container-ip 172.17.0.2 -container-port 9000
root      1341   766  0 18:37 ?        00:00:00 /usr/bin/docker-proxy -proto tcp -host-ip :: -host-port 9000 -container-ip 172.17.0.2 -container-port 9000
root      1378   766  0 18:37 ?        00:00:00 /usr/bin/docker-proxy -proto tcp -host-ip 0.0.0.0 -host-port 8181 -container-ip 172.18.0.3 -container-port 80
root      1384   766  0 18:37 ?        00:00:00 /usr/bin/docker-proxy -proto tcp -host-ip :: -host-port 8181 -container-ip 172.18.0.3 -container-port 80
root      1432     1  0 18:37 ?        00:00:00 /usr/bin/containerd-shim-runc-v2 -namespace moby -id 0297d490b252fce8fb720dbb4b5556fb427bc27dc176e83ec9af3abca86ad6cd -address /run/containerd/containerd.sock
root      1433     1  0 18:37 ?        00:00:00 /usr/bin/containerd-shim-runc-v2 -namespace moby -id 3b806d219aa96af122994da935ae338144107cf1663634e6518e8df45ff2cc4b -address /run/containerd/containerd.sock
root      1488     1  0 18:37 ?        00:00:00 /usr/bin/containerd-shim-runc-v2 -namespace moby -id 3efe9b5284dc47cadab8ff8afdf18ec9ac0008bff2b8ec6e9639fef3c2fe9c9b -address /run/containerd/containerd.sock
pi@raspberrypi-bullseye-desktop-docker:~ $ sudo systemctl stop containerd
pi@raspberrypi-bullseye-desktop-docker:~ $ ps -ef | egrep "docker|container"
avahi      503     1  0 18:37 ?        00:00:00 avahi-daemon: running [raspberrypi-bullseye-desktop-docker.local]
pi        2609  2138  0 18:43 pts/0    00:00:00 grep -E --color=auto docker|container
```

BMWfan, 2023-09-02:
@framps if i stop docker by using `systemctl stop docker` if get the following message:
`Warning: Stopping docker.service, but it can still be activated by:
  docker.socket`

closed 2023-10-12 by framps