framps/raspiBackup #673: AutoFS / RO file systems

NilsRo, 2023-08-31:
Hi,

I am using raspiBackup on an Raspi1 that is running as an IoT device with an RO file system...so why I like to backup it...the log files are stored in an tmpfs and also I do some changes remotly switching the file system to RW temporally. 

Actually I have to issues:

1. raspiBackup tries to create the backup directory instead of accessing it which brings in issues with autoFS.  
2. The log file is written to /root (or the user executing raspiBackup but this folrders are RO. It would be nice if the output could be set to e.g. /var/log by the user.

mkdir: cannot create directory ‘/nfs/backup’: Permission denied
/usr/local/bin/raspiBackup.sh: line 2053: /root/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup.sh: line 2053: /root/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup.sh: line 2053: /root/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup.sh: line 2053: /root/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup.sh: line 2053: /root/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup.sh: line 2053: /root/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup.sh: line 2053: /root/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup.sh: line 2158: /root/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup.sh: line 2159: /root/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup.sh: line 2302: /root/raspiBackup.msg: Read-only file system
/usr/local/bin/raspiBackup.sh: line 2053: /root/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup.sh: line 2167: /root/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup.sh: line 2053: /root/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup.sh: line 2053: /root/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup.sh: line 2053: /root/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup.sh: line 2053: /root/raspiBackup.log: Read-only file system 

Ich hoffe raspiBackup kommt auch bald für Debian. ;-)

Viele Grüße,
Nils

framps, 2023-08-31:
```
1.) raspiBackup tries to create the backup directory instead of accessing it which brings in issues with autoFS.
```
raspiBackup has to be able to create directories. I'm not familiar with autoFS but maybe there is some way to exclude the backup directory from autoFS handling ? 

```
2.) The log file is written to /root (or the user executing raspiBackup but this folrders are RO. It would be nice if the output could be set to e.g. /var/log by the user.
```

The log file is created in /tmp and finally moved into the  backup directory if the backup succeeds. If the backup fails it's indeed moved into the callers home directory. I understand that when you initially start to use raspiBackup there is a given probability you have some config or environment issues and raspiBackup fails and the log file cannot be saved. Either setup your system RW until you managed to get raspiBackup  not to fail or temporarily create a home directory in a RW partition.

```
Ich hoffe raspiBackup kommt auch bald für Debian. ;-)
```
Not sure I get you point. RaspbianOS is Debian.

framps, 2023-09-10:
Re 1) I need a debug log in order to locate the root cause of the issue. Maybe it's just a small change in raspiBackup.

Re 2)   I didn't get your point initially ... There exists option `-L` (See [here](https://www.linux-tips-and-tricks.de/en/invocation-and-options#parm_L) for the doc) which allows to change the target for the log. Please check :-)

Re 3) Please read [this page](https://www.linux-tips-and-tricks.de/en/raspibackupcategorye/609-supported-hard-and-software). Note: raspiBackup requires the two partitions used by a Raspberry and will not run on any other system which doesn't have the /boot and /root partition.

If you use a Debian on a Raspberry or Raspberry compatible HW there is a high chance raspiBackup will work. If it fails I'm willing to check the log whether it's possible to fix the issue with an acceptable effort. But in any case I will not execute any regression tests on this environment.

NilsRo, 2023-09-12:
Hi,

1) Prepared everything for testing, restarted to set  DEFAULT_LOG_OUTPUT andd it worrked. Do not know if there was some problems with autofs but it worrked for my interactive session like shown in the video above. Anyway, I will watch the cron backups and we will see....

2) I do not know why but I overseen the list. I found some articles and go through all parameters available in the config but this one is missing in the standard so I thought there is not any available. Perhaps you should link the option list on the installation page to cover idiots like me. :-)

3) 👍

framps, 2023-09-12:
1. I'll close this issue. If it pops up again create a new issue
2. I created an issue to dokument option L more prominent

closed 2023-09-12 by framps