framps/raspiBackup #674: Raspibackup fails restore

augea, 2023-09-01:
I'm trying to restore a backup but it fails to restore since partition cannot be created.

**Setup:**
- Raspberrypi 4 4GB RAM 64bit Rasbian OS Bullseye
- USB Boot from SSD 119 GB
- Backup created on a SD-Card (256GB) located in raspberry SD-slot.

**Restore started on a separte Raspberrypi 3** Rasbian OS Buster 32Bit - all updates installed
SSD /dev/sda (not mounted)
SD-Card /dev/sdb (mounted EXT4)

Part of log-file:
`20230901-122502 DBG 5700:                  --- sourceSDSize: 128035676160 - targetSDSize: 128035676160
Checking that no-one is using this disk right now ... OK

Disk /dev/sda: 119,2 GiB, 128035676160 bytes, 250069680 sectors
Disk model:                 
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
**>>> line 5: unsupported command**

New situation:
Leaving.

20230901-122503 DBG 3950:                  --> colorAnnotation 1
20230901-122503 DBG 3974:                  <-- colorAnnotation 
20230901-122503 MSG 5747:                  ?**?? RBK0111E: Fehler beim Erstellen der Partitionen. RC 1 - sfdisk error.** 
20230901-122503 DBG 2454:                  --> exitError 112
20230901-122503 DBG 2461:                  <-- exitError 112
20230901-122503 DBG 4635:                  --> cleanup 
20230901-122503 DBG 3950:                      --> colorAnnotation 1
20230901-122503 DBG 3974:                      <-- colorAnnotation 
20230901-122503 MSG 4646:                      --- RBK0033I: Bitte warten bis aufgeräumt wurde. 
20230901-122503 DBG 4779:                      --> cleanupRestore 
20230901-122503 DBG 4783:                          --- Got trap EXIT
20230901-122503 DBG 4784:                          --- rc: 112
20230901-122503 DBG 2202:                          --> callExtensions post
20230901-122503 DBG 2246:                          <-- callExtensions 
20230901-122503 DBG 3632:                          --> isMounted /tmp/raspiBackup
20230901-122503 DBG 3634:                                  *** cat /proc/mounts
20230901-122503 DBG 3634:                                      /dev/root / ext4 rw,noatime 0 0
20230901-122503 DBG 3634:                                      devtmpfs /dev devtmpfs rw,relatime,size=413012k,nr_inodes=103253,mode=755 0 0`
[raspiBackup.zip](https://github.com/framps/raspiBackup/files/12499801/raspiBackup.zip)

Is this a problem of my setup or a software problem?

framps, 2023-09-01:
It's a know issue: sfdisk changed unfortunately the output format from Buster to Bullseye. An additional line with `sector-size` is added when you create a backup with Bullseye. This additional line creates the error you report when you try to restore the backup with Buster or earlier. 

In next release 0.6.9 raspiBackup will detect this mismatch and delete the additional line.

But for now you have to either use a Bullseye OS to restore the backup or temporarily delete the additional line `sector-size` in the `.sfdisk` file in the backup directory :wink:

augea, 2023-09-01:
Restore was now successful. You saved my day! :smiley:

closed 2023-09-01 by augea