framps/raspiBackup #688: As the fdisk command is missing on raspi, an error message is displayed but no message is sent via Telegram, despite correct configuration

akzagain, 2023-10-15:
Hi,

I am using raspiBackup with DEFAULT_TELEGRAM_NOTIFICATIONS="FM". 

Recently I discovered that backups were missing. Manually running the script gave me an error message. RBK0193E: Missing required command 'fdisk sfdisk' and the advice to execute "????". RBK0194E: Required packages are missing. Install them with 'sudo apt-get install util-linux'." 

Two things:
1. Why was an error message never sent? Seems a failure, right?
2.  You need to run "sudo apt-get install fdisk" to solve the problem.

Cheers!

framps, 2023-10-15:
Would you please attach the debug log?

akzagain, 2023-10-15:
20231015-092235 DBG 2075:  --- Logfiles used: /tmp/raspiBackup.log and /tmp/raspiBackup.msg
20231015-092235 DBG 2086:  --- raspiBackup.sh 0.6.8, 2023-08-22/17:50:31 - 7c4a713
20231015-092235 DBG 2089:  --- +================================================================================================================================================+
20231015-092235 DBG 2090:  --- | ===> A lot of sensitive information is masqueraded in this log file. Nevertheless please check the log carefully before you distribute it <=== |
20231015-092235 DBG 2091:  --- +================================================================================================================================================+
20231015-092235 DBG 1917:  --> trapWithArg cleanupStartup SIGINT SIGTERM EXIT
20231015-092235 DBG 1922:  <-- trapWithArg
20231015-092235 DBG 3712:  --> readConfigParameters
20231015-092235 DBG 3727:      --- Read config /usr/local/etc/raspiBackup.conf : 0.1.7
20231015-092235 DBG 3727:      --- VERSION_CONFIG="0.1.7"
20231015-092235 DBG 3727:      --- DEFAULT_BACKUPPATH="/media/usb0/backup"
20231015-092235 DBG 3727:      --- DEFAULT_BACKUPTYPE="dd"
20231015-092235 DBG 3727:      --- DEFAULT_ZIP_BACKUP=0
20231015-092235 DBG 3727:      --- DEFAULT_DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0
20231015-092235 DBG 3727:      --- DEFAULT_KEEPBACKUPS=3
20231015-092235 DBG 3727:      --- DEFAULT_KEEPBACKUPS_DD=0
20231015-092235 DBG 3727:      --- DEFAULT_KEEPBACKUPS_DDZ=0
20231015-092235 DBG 3727:      --- DEFAULT_KEEPBACKUPS_TAR=0
20231015-092235 DBG 3727:      --- DEFAULT_KEEPBACKUPS_TGZ=0
20231015-092235 DBG 3727:      --- DEFAULT_KEEPBACKUPS_RSYNC=0
20231015-092235 DBG 3727:      --- DEFAULT_SMART_RECYCLE=0
20231015-092235 DBG 3727:      --- DEFAULT_SMART_RECYCLE_DRYRUN=1
20231015-092235 DBG 3727:      --- DEFAULT_SMART_RECYCLE_OPTIONS="7 4 12 1"
20231015-092235 DBG 3727:      --- DEFAULT_STOPSERVICES=""
20231015-092235 DBG 3727:      --- DEFAULT_STARTSERVICES=""
20231015-092235 DBG 3727:      --- DEFAULT_BEFORE_STOPSERVICES=""
20231015-092235 DBG 3727:      --- DEFAULT_AFTER_STARTSERVICES=""
20231015-092235 DBG 3727:      --- DEFAULT_EMAIL=""
20231015-092235 DBG 3727:      --- DEFAULT_SENDER_EMAIL=""
20231015-092235 DBG 3727:      --- DEFAULT_EMAIL_PARMS=""
20231015-092235 DBG 3727:      --- DEFAULT_MAIL_PROGRAM="mail"
20231015-092235 DBG 3727:      --- DEFAULT_TELEGRAM_TOKEN="2@@@@s(45)"
20231015-092235 DBG 3727:      --- DEFAULT_TELEGRAM_CHATID="1@@@@@@@3"
20231015-092235 DBG 3727:      --- DEFAULT_TELEGRAM_NOTIFICATIONS="FM"
20231015-092235 DBG 3727:      --- DEFAULT_PUSHOVER_TOKEN=""
20231015-092235 DBG 3727:      --- DEFAULT_PUSHOVER_USER=""
20231015-092235 DBG 3727:      --- DEFAULT_PUSHOVER_NOTIFICATIONS="F"
20231015-092235 DBG 3727:      --- DEFAULT_PUSHOVER_SOUND_SUCCESS=""
20231015-092235 DBG 3727:      --- DEFAULT_PUSHOVER_SOUND_FAILURE=""
20231015-092235 DBG 3727:      --- DEFAULT_PUSHOVER_PRIORITY_SUCCESS="0"
20231015-092235 DBG 3727:      --- DEFAULT_PUSHOVER_PRIORITY_FAILURE="1"
20231015-092235 DBG 3727:      --- DEFAULT_SLACK_WEBHOOK_URL=""
20231015-092235 DBG 3727:      --- DEFAULT_SLACK_NOTIFICATIONS=""
20231015-092235 DBG 3727:      --- DEFAULT_NOTIFY_START=0
20231015-092235 DBG 3727:      --- DEFAULT_MAIL_ON_ERROR_ONLY=0
20231015-092235 DBG 3727:      --- DEFAULT_MSG_LEVEL=0
20231015-092235 DBG 3727:      --- DEFAULT_COLORING="CM"
20231015-092235 DBG 3727:      --- DEFAULT_COLOR_CODES=("#FF8000 33" "#FF0000 31")
20231015-092235 DBG 3727:      --- DEFAULT_EMAIL_COLORING="SUBJECT"
20231015-092235 DBG 3727:      --- DEFAULT_RESTORE_DEVICE=""
20231015-092235 DBG 3727:      --- DEFAULT_VERBOSE=0
20231015-092235 DBG 3727:      --- DEFAULT_SKIPLOCALCHECK=0
20231015-092235 DBG 3727:      --- DEFAULT_DD_BLOCKSIZE=1MB
20231015-092235 DBG 3727:      --- DEFAULT_DD_PARMS=""
20231015-092235 DBG 3727:      --- DEFAULT_EXCLUDE_LIST=""
20231015-092235 DBG 3727:      --- DEFAULT_NOTIFY_UPDATE=1
20231015-092235 DBG 3727:      --- DEFAULT_EXTENSIONS=""
20231015-092235 DBG 3727:      --- DEFAULT_RESTORE_EXTENSIONS=""
20231015-092235 DBG 3727:      --- DEFAULT_LANGUAGE=""
20231015-092235 DBG 3727:      --- DEFAULT_DEPLOYMENT_HOSTS=""
20231015-092235 DBG 3727:      --- DEFAULT_YES_NO_RESTORE_DEVICE="loop"
20231015-092235 DBG 3727:      --- DEFAULT_RESIZE_ROOTFS=1
20231015-092235 DBG 3727:      --- DEFAULT_LINK_BOOTPARTITIONFILES=0
20231015-092235 DBG 3727:      --- DEFAULT_TAR_BOOT_PARTITION_ENABLED=0
20231015-092235 DBG 3727:      --- DEFAULT_SKIP_DEPRECATED=""
20231015-092235 DBG 3727:      --- DEFAULT_CHECK_FOR_BAD_BLOCKS=0
20231015-092235 DBG 3727:      --- DEFAULT_TIMESTAMPS=0
20231015-092235 DBG 3727:      --- DEFAULT_SYSTEMSTATUS=0
20231015-092235 DBG 3727:      --- DEFAULT_REBOOT_SYSTEM=0
20231015-092235 DBG 3727:      --- DEFAULT_RESTORE_REMINDER_INTERVAL=6
[... 123 lines cut ...]
20231015-092238 DBG 3974:      <-- colorAnnotation
??? RBK0194E: Missing required packages. Install them with 'sudo apt-get install util-linux util-linux'.
20231015-092238 DBG 2454:      --> exitError 120
20231015-092238 DBG 2461:      <-- exitError 120
20231015-092238 DBG 4592:      --> cleanupStartup
20231015-092238 DBG 4924:          --> cleanupTempFiles
20231015-092238 DBG 4931:          <-- cleanupTempFiles
20231015-092238 DBG 4491:          --- Masquerading some mount options
20231015-092239 DBG 4498:          --- Masquerading telegram token
20231015-092239 DBG 4503:          --- Masquerading telegram chatid
[... cut ...]
20231015-092239 DBG 2150:              --- Logfiles used: /var/log/raspiBackup/raspiBackup.log.log and /tmp/raspiBackup.msg
20231015-092239 DBG 2155:              --- Logfiles used: /var/log/raspiBackup/raspiBackup.log.log and /var/log/raspiBackup/raspiBackup.log.msg
20231015-092239 DBG 2162:                  *** cat /tmp/raspiBackup.logf
20231015-092239 DBG 2162:
20231015-092240 DBG 3950:              --> colorAnnotation 1
20231015-092240 DBG 3974:              <-- colorAnnotation
20231015-092240 DBG 2177:          <-- logFinish
20231015-092240 DBG 4611:      <-- cleanupStartup

framps, 2023-10-15:
Please *attach* the file and don't copy the contents because that's difficult to read :-)

akzagain, 2023-10-15:
Like this? 
[raspiBackup.log](https://github.com/framps/raspiBackup/files/12909173/raspiBackup.log)

framps, 2023-10-15:
Perfect :-)

framps, 2023-10-23:
> Why was an error message never sent? Seems a failure, right?

Yes. I have to create a fix.

> You need to run "sudo apt-get install fdisk" to solve the problem

Yes. I have to fix the message.

framps, 2023-11-01:
@akzagain Please verify the fix :wink:

akzagain, 2023-11-02:
Will do this weekend. Thanks for the heads-up.

framps, 2023-11-11:
> But the advice have twice fdisk. The command should be: sudo apt-get install fdisk

I'm sorry. I don't get your point. Would you please elaborate on this a bit more?

akzagain, 2023-11-12:
When the scripts detects the missing fdisk command, it advises to install it. That advice tells that the command to execute is "sudo apt-get install fdisk fdisk" instead of "sudo apt-get install fdisk".

framps, 2023-11-12:
Got it now. Thank you

framps, 2023-11-12:
Merged into master

framps, 2023-11-20:
Fix undone in 0.6.9 because option -U et al caused faiures.

closed 2024-01-21 by framps