framps/raspiBackup #692: Restored system of Bookworm misses files

framps, 2023-10-24:
Backup of Bookwork succeeds and restored system boots. But the contents of /boot is not identical of the original system.

Original system:
```
pi@raspberrypi-bookworm32-lite:~ $ ls -la /boot
total 71118
drwxr-xr-x  3 root root     4096 Oct 10 01:25 .
drwxr-xr-x 19 root root     4096 Oct 23 15:29 ..
-rw-r--r--  1 root root       83 Oct  5 12:33 System.map-6.1.0-rpi4-rpi-v6
-rw-r--r--  1 root root       83 Oct  5 12:33 System.map-6.1.0-rpi4-rpi-v7
-rw-r--r--  1 root root       83 Oct  5 12:33 System.map-6.1.0-rpi4-rpi-v7l
-rw-r--r--  1 root root       83 Oct  5 12:33 System.map-6.1.0-rpi4-rpi-v8
lrwxrwxrwx  1 root root       20 Oct 10 01:07 cmdline.txt -> firmware/cmdline.txt
-rw-r--r--  1 root root   199654 Oct  5 12:33 config-6.1.0-rpi4-rpi-v6
-rw-r--r--  1 root root   204942 Oct  5 12:33 config-6.1.0-rpi4-rpi-v7
-rw-r--r--  1 root root   223049 Oct  5 12:33 config-6.1.0-rpi4-rpi-v7l
-rw-r--r--  1 root root   230343 Oct  5 12:33 config-6.1.0-rpi4-rpi-v8
lrwxrwxrwx  1 root root       19 Oct 10 01:07 config.txt -> firmware/config.txt
drwxr-xr-x  3 root root     6144 Jan  1  1970 firmware
-rw-r--r--  1 root root 10360913 Oct 10 01:25 initrd.img-6.1.0-rpi4-rpi-v6
-rw-r--r--  1 root root 10475428 Oct 10 01:25 initrd.img-6.1.0-rpi4-rpi-v7
-rw-r--r--  1 root root 10677629 Oct 10 01:24 initrd.img-6.1.0-rpi4-rpi-v7l
-rw-r--r--  1 root root 10587679 Oct 10 01:24 initrd.img-6.1.0-rpi4-rpi-v8
lrwxrwxrwx  1 root root       17 Oct 10 01:06 overlays -> firmware/overlays
-rw-r--r--  1 root root  6664272 Oct  5 12:33 vmlinuz-6.1.0-rpi4-rpi-v6
-rw-r--r--  1 root root  6990632 Oct  5 12:33 vmlinuz-6.1.0-rpi4-rpi-v7
-rw-r--r--  1 root root  7429200 Oct  5 12:33 vmlinuz-6.1.0-rpi4-rpi-v7l
-rw-r--r--  1 root root  8726865 Oct  5 12:33 vmlinuz-6.1.0-rpi4-rpi-v8
```

framps, 2023-10-24:
Work in progress in https://github.com/framps/raspiBackup/tree/master_bookworm

Root cause for the missing new /boot files in rsync backup is the exclusion of /boot in rsync. Fixed in https://github.com/framps/raspiBackup/commit/b1d3fed67d388c2c6b6f9ec365b6e8cb4f1f61bd.

framps, 2023-10-24:
As of now rsync and tar backup incl option -B and restore are fixed and tested on bookworm for an SD card. Regression test on Bullseye is open.

Next backup and restore for an HD only system has to be worked on.

framps, 2023-10-24:
rsync and tar backup incl option -B and restore are fixed and tested on bookworm for an USB only system successfully.

framps, 2023-10-24:
rsync and tar backup/restore was successfully tested with bullseye on an SD card. USB tests are outstanding.

framps, 2023-10-25:
USB tests on bullseye finished successfully. 

`-P` tests on bookworm and bullseye finished successfully.

framps, 2023-10-25:
Release 0.6.8.1 will have the fix for Bookworm

WillScott73, 2023-10-25:
Would this explain why I'm getting an sfdisk error when I've been trying to restore a bookworm backup?

framps, 2023-10-25:
No. Not I'm aware of. Which sfdisk error do you get? Did you use a bookworm os to restore the backup? See [this issue](https://github.com/framps/raspiBackup/issues/659) which may be the root cause and there will be a workaround in next release.

WillScott73, 2023-10-25:
No, I was using buster, as it was all I had available on another SD card, but I figured out from this post https://forum-raspberrypi.de/forum/thread/48527-raspibackup-hilfe-beim-mounten-von-netzwerkordner-und-email-einrichtung/?pageNo=3 that it might need Bookworm to do the restore.

So I installed Bookwork, and when I tried to install raspiBackup it told me that Bookworm isn't supported, directed me to this issue, and I posted my question.

Fortunately my pi is only running PiHole, so it shouldn't take too long to start from scratch.

framps, 2023-10-25:
> So I installed Bookwork, and when I tried to install raspiBackup it told me that Bookworm isn't supported,

It's not supported **right now** until a fix is available :wink: . I'm busy working on it. 

> Fortunately my pi is only running PiHole, so it shouldn't take too long to start from scratch.

Glad to hear this. It's a nasty issue because the restored os boots successfully and there is no indication there are some files in /boot missing.

framps, 2023-10-25:
Fix published in release v0.6.8.1

closed 2023-10-25 by framps

berndhose, 2023-10-25:
I have successfully updated to version 0.6.8.1, but it is still not working when switch --unsupportedEnvironment is specified.
OS is DietPi Bookworm (stripped down Raspbian Bookworm ) on Rasperyy PI Model B

fdzaebel, 2023-10-25:
Thank you very much for the fix. Would be great if "curl -L https://raspibackup.linux-tips-and-tricks.de/install | sudo bash" would include this fix.

Again...thanks a lot! :)

wodkab, 2023-10-26:
> I have updated to version 0.6.8.1, but it is still not working when switch --unsupportedEnvironment is specified. Still telling that Bookworm is not supported. OS is DietPi Bookworm (stripped down Raspbian Bookworm ) on Raspberry PI Model B

Same for me with Raspberry Pi OS 12 (bookworm) on a Pi 4, updated to 0.6.8.1 (from 0.6.8.0) I get a message "Bookworm not supported" and a link to THIS issue (692).

framps, 2023-10-26:
I forgot to remove the bookworm guards I added in latest 0.6.8 :cry: . They are now gone.

Unfortunately you will not be able to upgrade 0.6.8.1 with this guard to the fixed version with `sudo raspibackup -U -S`. You have to delete the red lines you find in [this update](https://github.com/framps/raspiBackup/commit/b93ce60e662a78b41db0d42f6e713676b0b67478#) first and then execute the previous update command. 

I'm sorry for the inconvenience.

framps, 2023-10-26:
> BTW: With 0.6.8.0 I already did a restore (exchanging SD-Card) and it worked.

Yes. It works. But there are files missing in the restored /boot directory :unamused: 

That's a nasty issue and I think I will add an os test in next release which rejects to call raspiBackup on an OS which I haven't tested raspiBackup on.

berndhose, 2023-10-26:
> I forgot to remove the bookworm guards I added in latest 0.6.8 😢 . They are now gone.
> 
> Unfortunately you will not be able to upgrade 0.6.8.1 with this guard to the fixed version with `sudo raspibackup -U -S`. You have to delete the red lines you find in [this update](https://github.com/framps/raspiBackup/commit/b93ce60e662a78b41db0d42f6e713676b0b67478#) first and then execute the previous update command.
> 
> You also can execute
> 
> ```
> curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/raspiBackupDownloadFromGit.sh | bash -s -- master
> sudo chown root:root ./raspiBackup.sh
> sudo mv ./raspiBackup.sh /usr/local/bin
> ```
> 
> to update current installed raspiBackup.sh with latest code from git.
> 
> Sorry for the inconvenience.

Thank you !