framps/raspiBackup #694: Install tool not working via remote terminal (SSH)

fdzaebel, 2023-10-27:
Hello,

I tried to install raspiBackup via SSH connection from three different clients (Mac terminal, Windows terminal, ShellFish) but was not able to control the GUI using the keys (arrow, space, enter etc.). Also the GUI is broken.

[image]

Could you please advice me how to install raspiBackup using an SSH connection? I try to do this on the bookworm release.

Thank you very much.

framps, 2023-10-27:
Please make sure you configured your client to use UTF-8.

fdzaebel, 2023-10-27:
As far as I can see the terminal is using UTF-8. Here is the output from the locale command:

$ locale
LANG=de_DE.UTF-8
LANGUAGE=
LC_CTYPE="de_DE.UTF-8"
LC_NUMERIC="de_DE.UTF-8"
LC_TIME="de_DE.UTF-8"
LC_COLLATE="de_DE.UTF-8"
LC_MONETARY="de_DE.UTF-8"
LC_MESSAGES="de_DE.UTF-8"
LC_PAPER="de_DE.UTF-8"
LC_NAME="de_DE.UTF-8"
LC_ADDRESS="de_DE.UTF-8"
LC_TELEPHONE="de_DE.UTF-8"
LC_MEASUREMENT="de_DE.UTF-8"
LC_IDENTIFICATION="de_DE.UTF-8"
LC_ALL=

framps, 2023-10-27:
Hm ... that's nasty. I never had this issue before. Which OS do you use ? Please show the contents of /etc/os_release

fdzaebel, 2023-10-27:
Yes, I was also surprised. "rasps-config" e.g. is working as expected.

Here is the os_release output:

PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

framps, 2023-10-27:
> Yes, I was also surprised. "rasps-config" e.g. is working as expected.

The raspiBackupInstaller uses the same UI tool :thinking: 

As a workaround you can use `curl -o install -L https://raspibackup.linux-tips-and-tricks.de/install; sudo bash ./install -i` which will install raspiBackup without the UI. But then you have to configure all Options manualls in /usr/local/etc/raspiBackup.conf. In addition /etc/cron.d/raspiBackup has to be configured.

I think it's somehow related to some locale setting. Do you use the lite version or desktop version?

fdzaebel, 2023-10-27:
Thanks for the tipp.

No, I use this regular version:

[image]

framps, 2023-10-27:
Well, i didn't test the installer on the 64 bit desktop os. I tested only on 32 bit lite. But it worked on all variations on bullseye.

framps, 2023-10-27:
I installed Bookworm64 desktop and was able to install raspiBackup when I connected with ssh from a Linux system.

Than I started Windows10 on VMware on my Linux box and connected via ssh from the commandline. Installation of raspiBackup is possible.

I'm sorry - I cannot reproduce your issue and thus cannot help you :cry: . Either you find the root cause or you install raspiBackup without the installer and configure everything manually in the config files.

NewRavenGer, 2023-11-05:
Not sure if I can bring anything new to the table, but I have exactly the same issue with a 64 bit installation of bookworm when trying to install raspiBackup through SSH with Windows 11. After that, I tried to connect through ssh from my (android-)phone with ConnectBot, but the first screen got messed up the same way as under windows and I, was again, not be able to do any inputs - so this doesn't seem to be an isolated issue with the terminal ssh client on windows. 

Note: I was able to install raspiBackup from Windows 11 trhough SSH in the past on my Pi when it still ran Bullseye, so if I had to take an (un-)educated guess, I would think it has more to do with a change in the SSH implementation on the Pi with bookworm than with what SSH client you use.

framps, 2023-11-05:
> Not sure if I can bring anything new to the table,

Of course you do :+1: 

Given the fact you also face this issue it seems to be an issue :cry: .

Unfortunately I don't run any Windows11 and cannot reproduce the issue. It's good to know it worked with Bullseye and it works with raspi-config on Bookworm. Actually the raspiBackup installer uses the same toolset as raspi-config. So I can check for any changes in raspi-config and hopefully will detect something which they changed and will update the installer. But given the fact I cannot verify whether any change in the installer fixes the issue I need you to verify the fix. Do you want to support me on this?

fdzaebel, 2023-11-05:
Hi framps,

I now found the solution for me:

For the installation of raspiBackup I found two different install commands. On the GitHub page I found:
curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/installation/install.sh | sudo bash

This one caused the problems for me.

On https://www.linux-tips-and-tricks.de/de/raspberry/493-raspibackup-schnellstart-installiere-und-konfiguriere-raspibackup-in-5-minuten-um-ein-backup-zu-erstellen I found:
curl -o install -L https://raspibackup.linux-tips-and-tricks.de/install; sudo bash ./install

This one worked without any problems.

framps, 2023-11-05:
Thank you very much for the heads-up. I use the command from the website all the time. Frankly I have no clue why the other command creates this issue. But I'm now able to reproduce the issue partionally: When I use the command from github I cannot move the cursor.

I'll investigate on this now ...

framps, 2023-11-05:
I updated the installation command on github. @NewRavenGer Would you please verify the command `curl -o install -L https://raspibackup.linux-tips-and-tricks.de/install; sudo bash ./install` also fixes the issue for you?

NewRavenGer, 2023-11-05:
Yes, indeed I can confirm that the other install command fixes the issue completely (on Windows and also on Android). Thank you :)

framps, 2023-11-05:
Perfect. Thank you for your verification :smile: 

I frankly have no clue why the different installation command causes an issue on bookworm :cry: A lot of things changed on boockworm and I also had to update raspiBackup and publish 0.6.8.1 in order to create a complete backup for bookworm.

@fdzaebel Glad you found out it's caused by the installation command. I would never have detected this because I expect everybody to install raspiBackup according the instructions from my website.

closed 2023-11-05 by framps