framps/raspiBackup #695: Error restoring backup

rffuller, 2023-10-29:
I am having a problem restoring a backup. Every time it fails when trying to mount the root partition. Log attached...
[raspiBackup.logr.log](https://github.com/framps/raspiBackup/files/13196959/raspiBackup.logr.log)

framps, 2023-10-29:
Mount fails with following error message:
```
mount: /tmp/raspiBackup: wrong fs type, bad option, bad superblock on /dev/sda2, missing codepage or helper program, or other error.
```
/dev/sda2 was created and formatted
```
20231028-112141 MSG 5800:                  --- RBK0054I: Formating second partition (root partition) /dev/sda2. 
mke2fs 1.46.2 (28-Feb-2021)
Creating filesystem with 16316672 4k blocks and 4079616 inodes
Filesystem UUID: b098ff66-ebe3-4880-9c99-38f083551c1d
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208, 
	4096000, 7962624, 11239424

Allocating group tables:   0/498       done                            
Writing inode tables:   0/498       done                            
Creating journal (65536 blocks): done
Writing superblocks and filesystem accounting information:   0/498       done

20231028-112149 DBG 5389:                  --> waitForPartitionDefsChanged 
20231028-112152 DBG 5392:                      --- --- partprobe ---
/dev/sda: msdos partitions 1 2
/dev/mmcblk0: msdos partitions 1 2
```
/dev/sda
```
20231028-112114 DBG 5598:                              Disk /dev/sda: 62.5 GiB, 67109912576 bytes, 131074048 sectors
20231028-112114 DBG 5598:                              Disk model: MassStorageClass
20231028-112114 DBG 5598:                              Units: sectors of 1 * 512 = 512 bytes
20231028-112114 DBG 5598:                              Sector size (logical/physical): 512 bytes / 512 bytes
20231028-112114 DBG 5598:                              I/O size (minimum/optimal): 512 bytes / 512 bytes
20231028-112114 DBG 5598:                              Disklabel type: dos
20231028-112114 DBG 5598:                              Disk identifier: 0x00000000
20231028-112114 DBG 5598:                              
20231028-112114 DBG 5598:                              Device     Boot Start       End   Sectors  Size Id Type
20231028-112114 DBG 5598:                              /dev/sda1          32 131074047 131074016 62.5G  c W95 FAT32 (LBA)
```

The message is not very specific - unfortunately
wrong fs type: fs type is OK
bad option: options are OK
bad superblock on /dev/sda2: may be, device bad?
missing codepage or helper program: No idea what may be missing
or other error: ...

I suggest to execute following tests:
1) Check with `blkid` whether the creation and formatting of the target device succeeded (according the log it should)
2) Try to mount /dev/sda2 manually on /mnt 
3) Use a different restore device

rffuller, 2023-10-29:
Hi

Thanks for the reply.

I know, the message is not very specific. I already tried to format and mount the partition manually and I also ran the mkfs.ext4 manually with the -c switch. No errors.

I have also tried with another sd card. Same error. Both are brand new 64gb cards.

Tried with 16gb USB drive. It worked 🤡

Any more ideas?

Regards
Richard

framps, 2023-10-29:
> Tried with 16gb USB drive. It worked 🤡

That's strange :anguished: 

I have no 64GB sd card and cannot try to reproduce your issue. Did you try a smaller sd card? Maybe your sd card reader cannot handle sd cards of 64GB? 

Did you verify the sd card was partitioned and formatted correctly by raspiBackup? Please show the output of `blkid

rffuller, 2023-10-29:
Hi

Yes, the card is partitioned correctly. Output from lsblk is

***@***.***:~ $ blkid /dev/sdb2
/dev/sdb2: UUID="b47ecf0b-75b9-4c7c-80a9-97d66d98cfef" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="8ba0d6c8-02"

Interestingly, I just tried to restore onto a 32gb card and it worked. Looks like 64gb does not play nicely with pi’s, at least not mine ☹

So it looks like the problem is with the pi software rather than your backup script. Maybe you should investigate yourself with 64gb card and post an advisory if it also does not work for you.

Anyway, thanks for your help.

Regards
Richard

framps, 2023-10-30:
> So it looks like the problem is with the pi software rather than your backup script. Maybe you should investigate yourself with 64gb card and post an advisory if it also does not work for you.

I found people reporting  in the net they are not able to use a 64GB SD card on a Raspberry. It's not clear to me whether it's an RaspbianOS issue or SD card reader issue. But as far as I understand the SD card reader at least has to support SDXC in order to access SD cards >32GB. If your SD card reader is old you may order a new one and check it supports SDXC. 

Please close the issue if you're satisfied so far.

closed 2023-10-30 by rffuller

framps, 2023-11-10:
@rffuller I just orderd some 64GB SD cards and will try to reproduce your issue :wink:

rffuller, 2023-11-10:
I can confirm that the problem is not the SD card reader as I tried a different one that is confirmed to support SDXC and it did not work. Looking forward to seeing your results :)

framps, 2023-11-14:
@rffuller I just received an EVOSelect 64GB SD card from Samsung (See [here] for details(https://www.amazon.de/dp/B09D3N56DD?ref=ppx_yo2ov_dt_b_product_details&language=en_GB&currency=EUR&th=1)).

I was able to restore a Bookworm Image with raspiBackup on my RPi 3 and the restored system boots successfully on an RPi 1B. The SD card has less than 64GB but more than 32GB. 

Maybe the SD card capabilities helps to identify the root cause for your SD card to fail.

```
pi@raspberrypi-bookworm32-lite:~ $ mmc csd read /sys/block/mmcblk0/device
type: 'SD'
card classes: 10 switch, 8 application specific, 7 lock card, 5 erase, 4 block write, 2 block read, 0 basic, 
capacity: 59.69Gbyte (64088965120 bytes, 125173760 sectors, 512 bytes each)
pi@raspberrypi-bookworm32-lite:~ $ mmc scr read /sys/block/mmcblk0/device
type: 'SD'
version: SD 3.0x
bus widths: 4bit, 1bit, 
pi@raspberrypi-bookworm32-lite:~ $ mmc cid read /sys/block/mmcblk0/device
type: 'SD'
manufacturer: 'Transcend/Samsung' 'SM'
product: 'GC1S5' 3.0
serial: 0x428e6398
manufacturing date: 2023 apr
```