framps/raspiBackup #696: Reminder writes could not be disabled when using a read-only filesystem

NilsRo, 2023-10-29:
Hi,

I tried to disable the "reminder" as it tries to write to a file that is not possible in my read-only setup to avoid writes to the SD Card. I set the parameters like this but it does not stop trying.

DEFAULT_RESTORE_REMINDER_INTERVAL=0
DEFAULT_RESTORE_REMINDER_REPEAT=0

In the FAQ and parameter list I did not find how to disable the reminder. Is that actually possible?

Error: /usr/local/bin/raspiBackup.sh: line 7982: /var/lib/raspiBackup/restore.reminder: Read-only file system

[raspiBackup.log](https://github.com/framps/raspiBackup/files/13199078/raspiBackup.log)

Cheers,
Nils

framps, 2023-10-29:
The restore reminder just reminds you to execute a restore from time to time. I checked your debug log but I don't find any message which puzzles you and you want to get rid of :thinking: 

> as it tries to write to a file that is not possible in my read-only setup to avoid writes to the SD Card.

You can use option `-u` or config option `DEFAULT_EXCLUDE_LIST` to exclude any files from the backup. Not sure whether this will solve your issue because I frankly don't get you issue :cry: Please give more details about your issue.

framps, 2023-10-29:
> Error: /usr/local/bin/raspiBackup.sh: line 7982: /var/lib/raspiBackup/restore.reminder: Read-only file system

I just detected I missed following error message :cry: 

I use raspiBackup on systems which don't have RO filesystems configured. Looks like I have to persist the reminder file in some other directory which is not set to RO. Did you configure the RO filesystem with raspi-config? Then I can reproduce your issue and change the location of the reminder file accordingly

NilsRo, 2023-10-29:
Set it up with another guide but in my opinion /tmp is mapped to tmpfs everytime. Perhaps it would be the easiest to disable reminder somehow via settings.

framps, 2023-10-30:
> Or put it into the backup directory as it should be writeable.

That will work. But I changed the code an now the reminder stuff is just skipped. Please test the fix with 
```
curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/raspiBackupDownloadFromGit.sh | bash -s -- master_696
```

Anyhow I will execute tests with a RO system and update the code if needed in next release.

NilsRo, 2023-11-02:
Running without an issue. The simplest solution for users...

framps, 2023-11-02:
Fixed with https://github.com/framps/raspiBackup/commit/05b3e21b27af9fd3a22d2b23b5efeebf9015cb23 and just merged into master

closed 2023-11-02 by framps