framps/raspiBackup #698: Logfile cannot be stored in read-only filesystem if mount is not successful.

NilsRo, 2023-10-31:
Hi,

Accidently I shut down my NAS which resulted into a hanging raspiBackup process. No error notification is send. raspiBackup -n it stops with an error so I will watch the behavior. It is perhaps related to #696 

But what I still can see: the logfile is not written...it tries to put it to the read-only homefolder. I cannot put the whole homefolder to tmpfs...is it possible to move the location in that case to /tmp or /var/log as fallback?

root@ht3(rw):~# pstree
systemd─┬─ModemManager───2*[{ModemManager}]
        ├─agetty
        ├─automount───3*[{automount}]
        ├─avahi-daemon───avahi-daemon
        ├─cron───cron───sh───raspiBackup.sh───2*[raspiBackup.sh───tee]
        ├─dbus-daemon
        ├─dhcpcd
		
		
As also the home directory is write protected it cannot write to /home/nils as it tries with raspiBackup -n with root.
At best it should fall back to /var/log as this is write enabled for other process by default.

A logfile with write-enabled filesystem:
[raspiBackup.log](https://github.com/framps/raspiBackup/files/13220477/raspiBackup.log)

root@ht3(rw):~# raspiBackup -n
--- RBK0009I: ht3: raspiBackup.sh V0.6.8 - 2023-10-30 (d4856db) Sat 13 May 12:38:35 CEST 2023 gestartet.
??? RBK0258E: Dynamischer mount von /backup bekommt Fehler 32
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0010I: ht3: raspiBackup.sh V0.6.8 - 2023-10-30 (d4856db) Sat 13 May 12:38:40 CEST 2023 beendet mit Returncode 131.
--- RBK0026I: Debug Logdatei wurde in /home/nils/raspiBackup.log gesichert.

root@ht3(ro):~# raspiBackup -n
--- RBK0009I: ht3: raspiBackup.sh V0.6.8 - 2023-10-30 (d4856db) Tue 31 Oct 08:15:51 CET 2023 gestartet.
??? RBK0258E: Dynamischer mount von /backup bekommt Fehler 32
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0010I: ht3: raspiBackup.sh V0.6.8 - 2023-10-30 (d4856db) Tue 31 Oct 08:15:55 CET 2023 beendet mit Returncode 131.
/usr/local/bin/raspiBackup: line 2053: /home/nils/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup: line 2053: /home/nils/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup: line 2053: /home/nils/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup: line 2053: /home/nils/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup: line 2053: /home/nils/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup: line 2053: /home/nils/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup: line 2053: /home/nils/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup: line 2158: /home/nils/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup: line 2159: /home/nils/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup: line 2053: /home/nils/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup: line 2053: /home/nils/raspiBackup.log: Read-only file system
--- RBK0026I: Debug Logdatei wurde in /home/nils/raspiBackup.log gesichert.
/usr/local/bin/raspiBackup: line 2302: /home/nils/raspiBackup.msg: Read-only file system
/usr/local/bin/raspiBackup: line 2167: /home/nils/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup: line 2053: /home/nils/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup: line 2053: /home/nils/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup: line 2053: /home/nils/raspiBackup.log: Read-only file system
/usr/local/bin/raspiBackup: line 2053: /home/nils/raspiBackup.log: Read-only file system

Cheers,
Nils

framps, 2023-10-31:
> It is perhaps related to #696

No. That's a different story. Please verify the fix I created for you.

I suggest to check option `-L` or `DEFAULT_LOG_OUTPUT` :wink:

NilsRo, 2023-11-01:
It's set to "backup" to use the backup nfs folder but it is not available because my NAS is down. Now it falls back to a different folder (/home/nils) that is write protected. So perhaps the settings for using the backup folder and a specific default folder have to be separated. 

BTW. this night the backup runs into an error (as expected) and did not hang. So we can ignore this single issue.

framps, 2023-11-01:
> It's set to "backup" to use the backup nfs folder but it is not available because my NAS is down.

`backup` is the default and `~` is the directory if the backup fails. If you use `varlog` the log is written in /var/log. But I see your point. I'll add another option in next release which allows to define the location of the log file if the backup fails. Default will be `~`.

framps, 2023-11-01:
> BTW. this night the backup runs into an error (as expected) and did not hang. So we can ignore this single issue.

I created an enhancement issue #700 and close this issue now.

closed 2023-11-01 by framps