framps/raspiBackup #704: Cannot start  sudo raspiBackupInstallUI.sh after an installation

mcmellow, 2023-11-11:
**Note**
Issues are closed automatically after 2 weeks if there is no activity any more

After what seems a good installation and configuration of raspiBackup. I have an odd error on the commandline. See lines below.
`pi@mcmellow-pi:~ $ sudo /usr/local/bin/raspiBackupInstallUI.sh
/usr/local/bin/raspiBackupInstallUI.sh: 2: /usr/local/bin/raspiBackupInstallUI.sh: Syntax error: newline unexpected
pi@mcmellow-pi:~ $
`
Here is a copy of my raspiBackupInstallUI.sh
[raspiBackupInstallUI.sh.txt](https://github.com/framps/raspiBackup/files/13326275/raspiBackupInstallUI.sh.txt)

And the install.log
[raspiBackupInstallUI.log](https://github.com/framps/raspiBackup/files/13326286/raspiBackupInstallUI.log)

Greetings
McMelloW

framps, 2023-11-11:
raspiBackupInstallUI.sh is not correct. It should start with 
```
#!/bin/bash
#######################################################################################################################
#
# Script to download, install, configure and uninstall raspiBackup.sh using windows.
# Commandline installation is also possible. Use option -h to get a list of all options.
#
# Visit http://www.linux-tips-and-tricks.de/raspiBackup for latest code and other details
#
# Credits to following people for their translation work
#         FI - teemue
#         ZH - GoogleBeEvil
#         FR - mgrafr
#
#######################################################################################################################
```
Looks like there is/was an issue when raspiBackup was installed. The debug log unfortunately is not the one of the initial installation.

Would you please delete raspiBackup and then start the installation again and the provide the debug log? In order to delete raspiBackup just execute 
```
curl https://raspibackup.linux-tips-and-tricks.de/install | sudo bash -s -- -u
```
Then invoke the installation curl call again 
```
curl -o install -L https://raspibackup.linux-tips-and-tricks.de/install; sudo bash ./install
```
and install raspiBackup and finish the installer. Then attach the debug log.

mcmellow, 2023-11-11:
Hi Framps,

The un-install and the re-install went OK.  In the debug.log is no adding made since 17:30 this afternoon. No use in adding this log I guess.
`sudo raspiBackupInstallUI.sh`  works OK now

The installation log is added. 
[raspiBackupInstallUI.log](https://github.com/framps/raspiBackup/files/13326435/raspiBackupInstallUI.log)

I think it is OK now. I have to wait for new SD-cards, coming on Monday. So I hop to do some backup testing.

Thanks for you fast and good support.

framps, 2023-11-11:
> I think it is OK now.

Yes. According the log
```
MSG: RBI0002I: Downloading raspiBackup.sh...
20231111-210738: DBG: 1614 -                 -> downloadURL raspiBackup.sh
20231111-210738: DBG: 1617 -                 <- downloadURL https://www.linux-tips-and-tricks.de/raspiBackup/raspiBackup.sh
20231111-210738: DBG: 1832 -                 -> downloadFile URL: https://www.linux-tips-and-tricks.de/raspiBackup/raspiBackup.sh, file: /tmp/raspiBackup.sh
20231111-210738: DBG: 1838 -                    httpCode: 200 RC: 0
20231111-210738: DBG: 1851 -                 <- downloadFile 0
MSG: RBI0015I: Created /usr/local/bin/raspiBackup.sh.
MSG: RBI0015I: Created /usr/local/bin/raspiBackupInstallUI.sh.
20231111-210738: DBG: 1791 -              <- code_download_execute 
20231111-210739: DBG: 1889 -              -> config_download_execute 
MSG: RBI0002I: Downloading raspiBackup.conf...
20231111-210739: DBG: 1614 -                 -> downloadURL raspiBackup_en.conf
20231111-210739: DBG: 1617 -                 <- downloadURL https://www.linux-tips-and-tricks.de/raspiBackup/raspiBackup_en.conf
20231111-210739: DBG: 1832 -                 -> downloadFile URL: https://www.linux-tips-and-tricks.de/raspiBackup/raspiBackup_en.conf, file: /usr/local/etc/raspiBackup.conf
20231111-210740: DBG: 1838 -                    httpCode: 200 RC: 0
20231111-210740: DBG: 1851 -                 <- downloadFile 0
```
the download finished with HTTP 200 and creation of raspiBackup and the installer also succeeded. Maybe my website provider had some temporary outage and the error handling in the installer failed.

closed 2023-11-11 by framps