framps/raspiBackup #708: Smart Backup deletes all but 1 of the daily backups

vithusel, 2023-11-25:
Debug Log - 
[raspiBackup.log](https://github.com/framps/raspiBackup/files/13465344/raspiBackup.log)

I've set my smart backup to Daily:24 Weekly:4 Monthly:12 Yearly:3

The backup runs hourly via Cron. 

the OS is Ubuntu 22.04 server 64Bit Arm64

For some reason. After every backup it deletes everything older then the most recent one. (for that day) 

--- RBK0218I: Applying smart backup strategy. Daily:24 Weekly:4 Monthly:12 Yearly:3.
--- RBK0217I: 2 backups will be smart recycled. 3 backups will be kept. Please be patient.
--- RBK0221I: Smart backup strategy deletes /backup/ARM0000001/ARM0000001-rsync-backup-20231125-212729.
--- RBK0221I: Smart backup strategy deletes /backup/ARM0000001/ARM0000001-rsync-backup-20231125-214132.
--- RBK0017I: Backup finished successfully.

I know having 24 daily backups is probably a bit over the top. But this pi is running a application that collects live data which cannot be recreated. 

Im not sure if i've configured something wrong. Setting to a lower value (7 4 12 1) makes no difference. 

Below is my config 

#####################################################
#
# Default configuration file for raspiBackup.sh
#
# (C) 2013-2022 framp at linux-tips-and-tricks dot de
#
# For more details about options visit
# http://www.linux-tips-and-tricks.de/de/raspibackup/#parameter
#
# $Date: 2023-11-19 11:36:37 +0100$
# $Sha1: 5a99704$
#
#####################################################

# do not change following line
VERSION_CONFIG="0.1.7"

#####################################################
#
# Note:
#
# 0 turns an option off
# 1 turns an option on
#
#####################################################

# path to store the backupfile
DEFAULT_BACKUPPATH="/backup"

# type of backup: dd, tar or rsync
DEFAULT_BACKUPTYPE="rsync"

# zip tar or dd backup
DEFAULT_ZIP_BACKUP="0"

# dd backup will save space used by partitions only
DEFAULT_DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0

# how many backups to keep (1-n). Will be ignored if Smartrecycle is used.
DEFAULT_KEEPBACKUPS="52"
# how many backups to keep of the specific backup type. If zero DEFAULT_KEEPBACKUPS is used
DEFAULT_KEEPBACKUPS_DD=0
DEFAULT_KEEPBACKUPS_DDZ=0
DEFAULT_KEEPBACKUPS_TAR=0
DEFAULT_KEEPBACKUPS_TGZ=0
DEFAULT_KEEPBACKUPS_RSYNC=0

# Smart recycle
DEFAULT_SMART_RECYCLE="1"
# Smart recycle dryrun
DEFAULT_SMART_RECYCLE_DRYRUN="0"
# Smart recycle parameters (daily, weekly, monthly and yearly)
DEFAULT_SMART_RECYCLE_OPTIONS="24 4 12 3"

# commands to stop services before backup separated by &&
DEFAULT_STOPSERVICES="systemctl stop frp && systemctl stop cron && systemctl stop apache2 && systemctl stop apache-htcacheclean && systemctl stop containerd && systemctl stop snapd"
# commands to start services after backup separated by &&
DEFAULT_STARTSERVICES="systemctl start snapd && systemctl start containerd && systemctl start apache-htcacheclean && systemctl start apache2 && systemctl start cron && systemctl start frp"

# commands to execute before services are stopped separated by &&
DEFAULT_BEFORE_STOPSERVICES=""
# commands to execute when services were started again separated by &&
DEFAULT_AFTER_STARTSERVICES=""

# email to send completion status
DEFAULT_EMAIL=""

# Sender emailadress used with ssmtp and msmtp
DEFAULT_SENDER_EMAIL=""

# Additional parameters for email program (optional)
DEFAULT_EMAIL_PARMS=""

# mailprogram
DEFAULT_MAIL_PROGRAM="ssmtp"

# Telegram token
DEFAULT_TELEGRAM_TOKEN=""
# Telegram target chatid
DEFAULT_TELEGRAM_CHATID=""
# Telegram notifications to send. S(uccess), F(ailure), M(essages as file), m(essages as text)
DEFAULT_TELEGRAM_NOTIFICATIONS="F"

# Pushover token
DEFAULT_PUSHOVER_TOKEN=""
# Pushover user
DEFAULT_PUSHOVER_USER=""
# Pushover notifications to send. S(uccess), F(ailure), M(essages)
DEFAULT_PUSHOVER_NOTIFICATIONS="F"
# Pushover sound for success
DEFAULT_PUSHOVER_SOUND_SUCCESS=""
# Pushover sound for failure
DEFAULT_PUSHOVER_SOUND_FAILURE=""
# Pushover priority for success
DEFAULT_PUSHOVER_PRIORITY_SUCCESS="0"
# Pushover priority for failure
DEFAULT_PUSHOVER_PRIORITY_FAILURE="1"

# Slack webhook URL
DEFAULT_SLACK_WEBHOOK_URL=""
# Slack notifications to send. S(uccess), F(ailure), M(essages)
DEFAULT_SLACK_NOTIFICATIONS=""

# Send eMail and/or Telegram notification when backup starts
DEFAULT_NOTIFY_START=0

# Send email only in case of errors. Use with care !
DEFAULT_MAIL_ON_ERROR_ONLY=0

# msg level (0 = minimal, 1 = detailed)
DEFAULT_MSG_LEVEL="1"

# Colorize console output (C) and/or email (M)
DEFAULT_COLORING="CM"

# HTML color and VT100 color for warning and error, yellow red
DEFAULT_COLOR_CODES=("#FF8000 33" "#FF0000 31")

# eMail coloring (SUBJECT or OPTION)
DEFAULT_EMAIL_COLORING="SUBJECT"

# restore device
DEFAULT_RESTORE_DEVICE=""

# Default verbose log
DEFAULT_VERBOSE=0

# skip check for remote mount of backup path
DEFAULT_SKIPLOCALCHECK=0

# blocksize used for dd
DEFAULT_DD_BLOCKSIZE=1M
# addition parms used for dd
DEFAULT_DD_PARMS=""

# exclude list
# NOTE: The statement has to be a valid exclude statement for the used backuptool
# Prefix all excludes with --exclude. Example "--exclude /home/pi/.ssh --exclude /root/.ssh"
DEFAULT_EXCLUDE_LIST=""

# notify in email if there is an updated script version available
DEFAULT_NOTIFY_UPDATE=1

# backup extensions to call
DEFAULT_EXTENSIONS=""

# restore extensions to call
DEFAULT_RESTORE_EXTENSIONS=""

# language (DE, EN, FI, ZH or FR)
DEFAULT_LANGUAGE="EN"

# hosts which will get the updated backup script with parm -y - non pwd access with keys has to be enabled
# Example: "root@raspberrypi root@fhem root@openhab root@magicmirror"
DEFAULT_DEPLOYMENT_HOSTS=""

# Don't ask for restore confirmation for this device. Use with care !
DEFAULT_YES_NO_RESTORE_DEVICE=""

# Resize root filesystem during restore
DEFAULT_RESIZE_ROOTFS=1

# Use hardlinks for partitionbootfiles
DEFAULT_LINK_BOOTPARTITIONFILES=0

# save boot partition with tar
DEFAULT_TAR_BOOT_PARTITION_ENABLED=0

# Versions not to generated deprecated messages, separated with commas
DEFAULT_SKIP_DEPRECATED=""

# Check for bad blocks when formating restore device (Will take a long time)
DEFAULT_CHECK_FOR_BAD_BLOCKS=0

# add timestamps in front of messages
DEFAULT_TIMESTAMPS=0

# add system status in debug log
DEFAULT_SYSTEMSTATUS=0

# reboot system at end of backup
DEFAULT_REBOOT_SYSTEM=0

# Restoretest reminder interval (unit: months)
DEFAULT_RESTORE_REMINDER_INTERVAL=6
# Number of times to remind to test restore
DEFAULT_RESTORE_REMINDER_REPEAT=3

# Options for partitionoriented mode
DEFAULT_PARTITIONBASED_BACKUP="0"
DEFAULT_PARTITIONS_TO_BACKUP="1 2"

# Name of backup partition to dynamically mount (e.g. /dev/sda1 or /backup), should be empty to disable dynamic mount
DEFAULT_DYNAMIC_MOUNT=""

framps, 2023-11-26:
> I know having 24 daily backups is probably a bit over the top. But this pi is running a application that collects live data which cannot be recreated.

That's a misunderstanding from you side:
```
# Smart recycle parameters (daily, weekly, monthly and yearly)
DEFAULT_SMART_RECYCLE_OPTIONS="24 4 12 3"
```
The daily option defines the number of daily backups to keep - not how many backups per day. With 24 you will get one backup per day for the last 24 days. If you use 7 you get daily backups of the last 7 days.

> But this pi is running a application that collects live data which cannot be recreated.

I see your point but why do you want to backup the whole system all the time? Just save the important data every hour. This will be much faster and consume less disk space for the backup.

vithusel, 2023-11-27:
I may have to look into this. It was simply ease of effort. I have a lot of storage on my backup server and backup time was about 3 minutes. 

Thanks for coming back to me :)

closed 2023-11-27 by vithusel

framps, 2023-11-27:
If you use the other backup recycle method and use `KEEP_BACKUPS` you can keep hourly backups. But then you have to take care of them yourself to keep backups following the GFS principle. 

raspiBackup smart recycle is not designed to manage backups on a less than daily base. If you create more than one backup per day the latest backup will be kept and all other backups will be recycled. That's what you saw when you created multiple backups per day :wink: